bluekitchen/btstack #196: Cmake build system

squonk11, 2019-03-11:
I am trying to use the btstack in spp profile mode with an ESP32. Unfortunately the GNU make system using MSYS32 under Windows 10 currently works extremely slow. The Cmake build system of ESP-IDF is much faster.
So, my question: are there any plans to also support the Cmake build system in future also?

mringwal, 2019-03-12:
Yupp. Someone submitted a partial pull request for this, and there's a first version on the https://github.com/bluekitchen/btstack/tree/esp32-cmake branch. It works for Classic examples, but misses the ability to generate the .h from .gatt files used for examples with GATT Server.

Please let me know if that works for you.

squonk11, 2019-03-12:
Thank you for the hint - it now works perfectly for me. I currently don't need to generate .h from .gatt files because I only use SPP profile.

mringwal, 2019-03-12:
What's the difference in compile time between the old Makefile and CMake? When using CMake, what is generated? On Mac, CMake generates a regular Makefile.

squonk11, 2019-03-13:
In the past the speed difference in compilation between cmake and GNU make was roughly a factor of two. Recently something (I don't know what) changed and the GNU make takes much longer and the compile time increases continuously over time after starting the MSYS window. After a fresh start of the MSYS window a complete compile of one of the examples takes roughly 10 minutes with GNU make under MSYS and 1,5 minutes with cmake und Windows natively. 
But as already mentioned the compile time under MSYS using GNU make increases over time and reaches easily 30 minutes after a few hours of working.
Due to this GNU make build system under MSYS is not really an option if you are in a development phase.

mringwal, 2019-03-14:
Thanks. One more question: as far as I understand CMake uses the CMakeLists.txt to generate a project file for another build system. On Mac, it can create a regular Makefile or a Xcode project. What does it create on MINGW? Or does it do all the building itself?

xnlcasad, 2019-03-22:
CMake is a build system generator. Depending on the available so-called "cmake generators" in your environment (Mac, Linux, Windows, etc) CMake can generate build system scripts for GNU Make, Ninja, XCode, Visual Studio and some others.

CMake will never build by itself.

mringwal, 2019-03-25:
@xnlcasad Thanks, that was my understanding. That's why I've asked what generator is used in the Windows environment (although I suspect that it's also GNU make).

jschoch, 2019-08-13:
this seems to be having issues with micro-ecc

`  Failed to resolve component 'micro-ecc'.`

mringwal, 2019-08-13:
Good point. Espressif moved their micro-ecc into their bootloader. This was fixed for the Makefile but not for the CMake.

Adding BTstack's 3rd-party is easy, the includes need to be added as private includes (see latest Makefile changess)

jschoch, 2019-08-13:
I had to add both of these lines to port/esp32/CMakeLists.txt to get an example to build
```
set(PRIV_INCLUDE_DIRS "3rd-party/micro-ecc")
set(PRIV_REQUIRES "micro-ecc")
```

mringwal, 2019-08-16:
Thanks for the update. Could you post your full CMakeLists.txt please? I've tried with esp-idf master banch and get

[ 68%] Building C object esp-idf/btstack/CMakeFiles/__idf_btstack.dir/src/btstack_crypto.c.obj
/Projects/esp32/esp-idf/components/btstack/src/btstack_crypto.c:64:18: fatal error: uECC.h: No such file or directory

although I've copied 3rd-party/micro-ecc into esp-idf/components/btstack/3rd-party

jschoch, 2019-08-16:
here it is, i think you also need to have the dir listed in srcdirs but I have also in includedirs.  I also had to add the dir to the integration script

Full CMakeFiles.txt

```
set(COMPONENT_ADD_INCLUDEDIRS "3rd-party/bluedroid/decoder/include"
                              "3rd-party/bluedroid/encoder/include"
                              "3rd-party/hxcmod-player"
                              "3rd-party/hxcmod-player/mods"
                              "3rd-party/md5"
                              "3rd-party/yxml"
                              "3rd-party/micro-ecc"
                              "src/classic"
                              "src/ble/gatt-service"
                              "src/ble"
                              "src/classic"
                              "src"
                              "platform/embedded"
                              "platform/freertos"
                              "include")

set(COMPONENT_SRCDIRS "3rd-party/bluedroid/decoder/srce"
                      "3rd-party/bluedroid/encoder/srce"
                      "3rd-party/hxcmod-player"
                      "3rd-party/hxcmod-player/mods"
                      "3rd-party/md5"
                      "3rd-party/micro-ecc"
                      "src/ble/gatt-service"
                      "src/ble"
                      "src/classic"
                      "src/"
                      "platform/freertos"
                      ".")

set(PRIV_INCLUDE_DIRS "3rd-party/micro-ecc")
set(PRIV_REQUIRES "micro-ecc")

set(COMPONENT_REQUIRES "nvs_flash" "bt")
register_component()

```

full diff

```
diff --git a/port/esp32/components/btstack/CMakeLists.txt b/port/esp32/components/btstack/CMakeLists.txt
index e984084..6ad912d 100644
--- a/port/esp32/components/btstack/CMakeLists.txt
+++ b/port/esp32/components/btstack/CMakeLists.txt
@@ -4,6 +4,7 @@ set(COMPONENT_ADD_INCLUDEDIRS "3rd-party/bluedroid/decoder/include"
                               "3rd-party/hxcmod-player/mods"
                               "3rd-party/md5"
                               "3rd-party/yxml"
+                             "3rd-party/micro-ecc"
                               "src/classic"
                               "src/ble/gatt-service"
                               "src/ble"
@@ -18,6 +19,7 @@ set(COMPONENT_SRCDIRS "3rd-party/bluedroid/decoder/srce"
                       "3rd-party/hxcmod-player"
                       "3rd-party/hxcmod-player/mods"
                       "3rd-party/md5"
+                     "3rd-party/micro-ecc"
                       "src/ble/gatt-service"
                       "src/ble"
                       "src/classic"
@@ -25,6 +27,9 @@ set(COMPONENT_SRCDIRS "3rd-party/bluedroid/decoder/srce"
                       "platform/freertos"
                       ".")

-set(COMPONENT_REQUIRES "micro-ecc" "nvs_flash" "bt")
+set(PRIV_INCLUDE_DIRS "3rd-party/micro-ecc")
+set(PRIV_REQUIRES "micro-ecc")
+
+set(COMPONENT_REQUIRES "nvs_flash" "bt")
 register_component()

diff --git a/port/esp32/integrate_btstack.py b/port/esp32/integrate_btstack.py
index 9b0759f..bb7c495 100755
--- a/port/esp32/integrate_btstack.py
+++ b/port/esp32/integrate_btstack.py
@@ -41,6 +41,7 @@ dirs_to_copy = [
 '3rd-party/hxcmod-player',
 '3rd-party/md5',
 '3rd-party/yxml',
+'3rd-party/micro-ecc',
 'platform/freertos',
 'platform/embedded',
 'tool'
```

mringwal, 2021-07-08:
Cmake works with current versions of esp-idf and btstack.

closed 2021-07-08 by mringwal