bluekitchen/btstack #198: Unable to compile "own" examples with lean Makefile

Webunity, 2019-03-13:
**First of all, great work on the stack! This is both an issue report and the fix of it**

I  have cloned the repository on my raspberry pi in `/opt/btstack/`. In order to compile for my situation, i had to go to `/opt/btstack/port/libusb` and execute `make`, which successfully compiles. The samples work, so then i decided to extend one sample as a spike.

### The issue:
As starting point for my spike, i copied `/opt/btstack/examples/hog_keyboard_demo.*` to `/opt/test/hog_test.*`. This needs a `Makefile` off course, and to keep my Makefile as lean as possible (and benefit from future btstack updates to your own Makefiles) i attempted to include the Makefile from btstack.

However, the following line in my own Makefile horribly fails
```
include /opt/btstack/port/libusb/Makefile
```

This is caused by the following line in `/opt/btstack/port/libusb/Makefile`:
```
include ${BTSTACK_ROOT}/example/Makefile.inc
```

Since `BTSTACK_ROOT` [is set to](https://github.com/bluekitchen/btstack/blob/master/port/libusb/Makefile#L2) `../..` at the top of the Makefile, the line
```
include ${BTSTACK_ROOT}/example/Makefile.inc
```

Translates into:
```
include ../../example/Makefile.inc
```
That off course fails if you are not in the `/opt/btstack/port/libusb` directory.

### The fix:
In order to fix that, i replaced:
```
BTSTACK_ROOT = ../..
```

(in `/opt/btstack/port/libusb/Makefile`) with the following two lines:
```
CWD:=$(shell dirname $(realpath $(lastword $(MAKEFILE_LIST))))
BTSTACK_ROOT = $(CWD)/../..
```

Which then takes the actual path of the Makefile as a starting point. All keeps compiling, and then i can slim down my own Makefile to this, which also magically works!

```
PORT_DIR = /opt/btstack/port/libusb
include ${PORT_DIR}/Makefile

# Add the 'port' directory for searching main.o etc.
VPATH += ${PORT_DIR}
CFLAGS += -I${PORT_DIR}

# Create our target
TEST = test

test: hog_test.h ${CORE_OBJ} ${COMMON_OBJ} ${ATT_OBJ} ${GATT_SERVER_OBJ} ${SM_OBJ} battery_service_server.o device_information_service_server.o hids_device.o btstack_ring_buffer.o hog_test.c
        ${CC} $(filter-out hog_test.h,$^) ${CFLAGS} ${LDFLAGS} -o $@ && \
        rm *.h
```

Hope this helps and can be adjusted in the makefile!

Discussion: https://groups.google.com/d/msg/btstack-dev/KdVO6qUYmxg/NbGa1wOcCAAJ

mringwal, 2019-04-08:
Hi! Thanks for the suggestion. On first sight, I'd rather avoid the use of shell commands in the Makefile to get the absolute path. Would it help if BTSTACK_ROOT is only set if not already defined? By that, you could set it in your Makefile.

Webunity, 2019-04-09:
Yes, that would help

Webunity, 2019-04-14:
@mringwal if you could 'patch' this i can release my open source BLE HID driver.

mringwal, 2019-04-15:
See commit. Feel free to announce your HID driver on the BTstack mailing list.

commit f67eb7c2 (2019-04-15): port: allow to defined BTSTACK_ROOT and include exising Makefiles. Fixes #198
closed 2019-04-15 by mringwal