bluekitchen/btstack #199: Issue in hids_device.c ?

Webunity, 2019-03-14:
```
// ATT Client Read Callback for Dynamic Data
...
    if (att_handle == instance->hid_report_input_client_configuration_handle){
        return att_read_callback_handle_little_endian_16(instance->hid_report_input_client_configuration_value, offset, buffer, buffer_size);
    }

    if (att_handle == instance->hid_report_input_client_configuration_handle){
        return att_read_callback_handle_little_endian_16(instance->hid_report_output_client_configuration_value, offset, buffer, buffer_size);
    }

    if (att_handle == instance->hid_report_input_client_configuration_handle){
        return att_read_callback_handle_little_endian_16(instance->hid_report_feature_client_configuration_value, offset, buffer, buffer_size);
    }
...
```

The 'if' checks all point to the same handle, should this read:
```
    if (att_handle == instance->hid_report_input_client_configuration_handle){
        return att_read_callback_handle_little_endian_16(instance->hid_report_input_client_configuration_value, offset, buffer, buffer_size);
    }

    if (att_handle == instance->hid_report_output_client_configuration_value){
        return att_read_callback_handle_little_endian_16(instance->hid_report_output_client_configuration_value, offset, buffer, buffer_size);
    }

    if (att_handle == instance->hid_report_feature_client_configuration_value){
        return att_read_callback_handle_little_endian_16(instance->hid_report_feature_client_configuration_value, offset, buffer, buffer_size);
    }
```

e.g. look at the last 2 'ifs' ??

mringwal, 2019-03-14:
that's a very good observation. on the first glance, you're totally right. I'll fix soon...

Webunity, 2019-03-14:
@mringwal added it to my pull request, i think i did the right fix 👍

mringwal, 2019-03-18:
Thanks, fixed in 826b57ef.

closed 2019-03-18 by mringwal