bluekitchen/btstack #202: hid examples don't work on esp32

fluffymadness, 2019-03-24:
I tried to flash and execute the hid examples for the esp32.
Both, the mouse and keyboard example pair for me, but the connection gets closed after a few seconds. I tried using windows 10 and my android phone.
When I try to connect to the esp32 after pairing with my phone, it says connecting, stays there for a second and disconnects afterwards. Same for windows 10. It stays connected for 10~15 secs after pairing, then disconnects.
I already search around a few hours and tried various things, but no solution so far. Hopefully someone can point me in the right direction. Thanks.
Here is the log:

`[00:00:00.757] LOG -- main.c.251: transport: configure SCO over HCI, result 0x0000
[00:00:00.758] LOG -- hci.c.4063: BTSTACK_EVENT_STATE 1
BTstack: execute run loop
[00:00:00.762] LOG -- hci.c.1975: Manufacturer: 0x0060
[00:00:00.767] LOG -- hci.c.1889: local name:
[00:00:00.770] LOG -- hci.c.1648: Received local name, need baud change 0
[00:00:00.777] LOG -- hci.c.1986: Local supported commands summary 0x3f
[00:00:00.784] LOG -- hci.c.1937: Local Address, Status: 0x00: Addr: 3C:71:BF:AB:19:4A
[00:00:00.791] LOG -- hci.c.1906: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4
[00:00:00.802] LOG -- hci.c.1964: Packet types cc18, eSCO 1
[00:00:00.807] LOG -- hci.c.1967: BR/EDR support 1, LE support 1
[00:00:00.816] LOG -- hci.c.4181: BTSTACK_EVENT_DISCOVERABLE_ENABLED 1
[00:00:00.820] LOG -- hci.c.1917: hci_le_read_buffer_size: size 251, count 10
[00:00:00.826] LOG -- hci.c.1756: Supported commands 30
[00:00:00.831] LOG -- hci.c.1924: hci_le_read_maximum_data_length: tx octets 251, tx time 2120 us
[00:00:00.840] LOG -- hci.c.1930: hci_le_read_white_list_size: size 12
[00:00:00.846] LOG -- hci.c.1512: hci_init_done -> HCI_STATE_WORKING
[00:00:00.852] LOG -- hci.c.4063: BTSTACK_EVENT_STATE 2
BTstack: up and running.
[00:00:20.721] LOG -- hci.c.2076: Connection_incoming: A8:7D:12:01:BE:F1, type 1
[00:00:20.721] LOG -- hci.c.186: create_connection_for_addr A8:7D:12:01:BE:F1, type ff
[00:00:20.725] LOG -- hci.c.3438: sending hci_accept_connection_request, remote eSCO 0
[00:00:20.912] LOG -- hci.c.2100: Connection_complete (status=0) A8:7D:12:01:BE:F1
[00:00:20.913] LOG -- hci.c.2113: New connection: handle 128, A8:7D:12:01:BE:F1
[00:00:20.916] LOG -- hci.c.4132: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[00:00:20.923] LOG -- hci.c.2162: HCI_EVENT_READ_REMOTE_SUPPORTED_FEATURES_COMPLETE, bonding flags 6, eSCO 1
[00:00:20.932] LOG -- l2cap.c.1864: channel 0x3ffd1368, local_cid 0x0040
[00:00:20.938] LOG -- hci.c.4266: gap_request_security_level requested level 0, planned level 0, current level 0
[00:00:20.948] LOG -- hci.c.4163: hci_emit_security_level 0 for handle 80
[00:00:20.954] LOG -- l2cap.c.2224: l2cap - security level update for handle 0x0080
[00:00:20.962] LOG -- l2cap.c.2234: channel 0x3ffd1368, cid 0040 - state 5: actual 0 >= required 0?
[00:00:20.971] LOG -- l2cap.c.939: L2CAP_EVENT_INCOMING_CONNECTION addr A8:7D:12:01:BE:F1 handle 0x80 psm 0x1 local_cid 0x40 remote_cid 0x40
[00:00:20.983] LOG -- l2cap.c.2345: L2CAP_ACCEPT_CONNECTION local_cid 0x40
[00:00:20.991] LOG -- l2cap.c.1100: l2cap_stop_rtx for local cid 0x40
[00:00:20.996] LOG -- l2cap.c.1109: l2cap_start_rtx for local cid 0x40
[00:00:21.002] LOG -- l2cap.c.2550: L2CAP signaling handler code 4, state 11
[00:00:21.009] LOG -- l2cap.c.2401: Remote MTU 1024
[00:00:21.013] LOG -- l2cap.c.2550: L2CAP signaling handler code 5, state 11
[00:00:21.020] LOG -- l2cap.c.1100: l2cap_stop_rtx for local cid 0x40
[00:00:21.026] LOG -- l2cap.c.2481: l2cap_signaling_handle_configure_response
[00:00:21.034] LOG -- l2cap.c.905: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr A8:7D:12:01:BE:F1 handle 0x80 psm 0x1 local_cid 0x40 remote_cid 0x40 local_mtu 1691, remote_mtu 1024, flush_timeout 0
[00:00:21.097] LOG -- hci.c.3491: IO Capability Request received, stack bondable 1, io cap 3
[00:00:21.098] LOG -- hci.c.4241: gap_mitm_protection_required_for_security_level 0
[00:00:21.380] LOG -- hid_mouse_demo.c.238: SSP User Confirmation Request with numeric value '182715'

[00:00:21.380] LOG -- hid_mouse_demo.c.239: SSP User Confirmation Auto accept

[00:00:21.853] LOG -- hci.c.388: gap_store_link_key_for_bd_addr: A8:7D:12:01:BE:F1, type 4
[00:00:21.853] LOG -- btstack_tlv_esp32.c.54: tag 42544c00 -> 42544C00
[00:00:21.856] LOG -- btstack_tlv_esp32.c.68: read tag 42544C00
[00:00:21.862] LOG -- btstack_tlv_esp32.c.54: tag 42544c01 -> 42544C01
[00:00:21.868] LOG -- btstack_tlv_esp32.c.68: read tag 42544C01
[00:00:21.873] LOG -- btstack_tlv_esp32.c.54: tag 42544c02 -> 42544C02
[00:00:21.880] LOG -- btstack_tlv_esp32.c.68: read tag 42544C02
[00:00:21.885] LOG -- btstack_tlv_esp32.c.54: tag 42544c03 -> 42544C03
[00:00:21.892] LOG -- btstack_tlv_esp32.c.68: read tag 42544C03
[00:00:21.897] LOG -- btstack_tlv_esp32.c.54: tag 42544c04 -> 42544C04
[00:00:21.904] LOG -- btstack_tlv_esp32.c.68: read tag 42544C04
[00:00:21.909] LOG -- btstack_tlv_esp32.c.54: tag 42544c05 -> 42544C05
[00:00:21.915] LOG -- btstack_tlv_esp32.c.68: read tag 42544C05
[00:00:21.921] LOG -- btstack_tlv_esp32.c.54: tag 42544c06 -> 42544C06
[00:00:21.927] LOG -- btstack_tlv_esp32.c.68: read tag 42544C06
[00:00:21.933] LOG -- btstack_tlv_esp32.c.54: tag 42544c07 -> 42544C07
[00:00:21.939] LOG -- btstack_tlv_esp32.c.68: read tag 42544C07
[... 52 lines cut ...]
[00:00:27.357] LOG -- hci.c.4266: gap_request_security_level requested level 2, planned level 0, current level 2
[00:00:27.362] LOG -- hci.c.4163: hci_emit_security_level 2 for handle 80
[00:00:27.368] LOG -- l2cap.c.2224: l2cap - security level update for handle 0x0080
[00:00:27.376] LOG -- l2cap.c.2234: channel 0x3ffd13d4, cid 0043 - state 5: actual 2 >= required 2?
[00:00:27.384] LOG -- l2cap.c.939: L2CAP_EVENT_INCOMING_CONNECTION addr A8:7D:12:01:BE:F1 handle 0x80 psm 0x13 local_cid 0x43 remote_cid 0x43
[00:00:27.397] LOG -- l2cap.c.2364: L2CAP_DECLINE_CONNECTION local_cid 0x43
[00:00:27.404] LOG -- l2cap.c.1100: l2cap_stop_rtx for local cid 0x43
[00:00:27.410] LOG -- hid_device.c.667: L2CAP_EVENT_INCOMING_CONNECTION, decline connection for A8:7D:12:01:BE:F1
[00:00:27.420] LOG -- l2cap.c.2234: channel 0x3ffd1368, cid 0041 - state 12: actual 2 >= required 2?
[00:00:32.411] LOG -- l2cap.c.1864: channel 0x3ffd13d4, local_cid 0x0044
[00:00:32.412] LOG -- hci.c.4266: gap_request_security_level requested level 2, planned level 0, current level 2
[00:00:32.417] LOG -- hci.c.4163: hci_emit_security_level 2 for handle 80
[00:00:32.423] LOG -- l2cap.c.2224: l2cap - security level update for handle 0x0080
[00:00:32.431] LOG -- l2cap.c.2234: channel 0x3ffd13d4, cid 0044 - state 5: actual 2 >= required 2?
[00:00:32.440] LOG -- l2cap.c.939: L2CAP_EVENT_INCOMING_CONNECTION addr A8:7D:12:01:BE:F1 handle 0x80 psm 0x13 local_cid 0x44 remote_cid 0x44
[00:00:32.452] LOG -- l2cap.c.2364: L2CAP_DECLINE_CONNECTION local_cid 0x44
[00:00:32.459] LOG -- l2cap.c.1100: l2cap_stop_rtx for local cid 0x44
[00:00:32.465] LOG -- hid_device.c.667: L2CAP_EVENT_INCOMING_CONNECTION, decline connection for A8:7D:12:01:BE:F1
[00:00:32.475] LOG -- l2cap.c.2234: channel 0x3ffd1368, cid 0041 - state 12: actual 2 >= required 2?
[00:00:32.484] LOG -- l2cap.c.2550: L2CAP signaling handler code 6, state 12
[00:00:32.491] LOG -- l2cap.c.933: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x41
HID Disconnected
[00:00:32.499] LOG -- l2cap.c.1100: l2cap_stop_rtx for local cid 0x41
[00:00:35.531] LOG -- hci.c.923: Connection closed: handle 0x80, A8:7D:12:01:BE:F1
[00:00:35.531] LOG -- hci.c.4132: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0`

fluffymadness, 2019-03-25:
I enabled more advanced logging functions. Here is the new log of the keyboard hid demo with the disconnects.
```
Toolchain path: /home/chainer/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /home/chainer/esp/esp-idf/requirements.txt are satisfied.
MONITOR
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5236
load:0x40078000,len:10876
load:0x40080400,len:6360
entry 0x4008070c
[0;32mI (170) cpu_start: Pro cpu up.[0m
[0;32mI (171) cpu_start: Application information:[0m
[0;32mI (171) cpu_start: Project name:     hid_keyboard_demo[0m
[0;32mI (174) cpu_start: App version:      ad7cba71-dirty[0m
[0;32mI (179) cpu_start: Compile time:     Mar 25 2019 14:58:51[0m
[0;32mI (185) cpu_start: ELF file SHA256:  6b0eb06f6e706c8f...[0m
[0;32mI (191) cpu_start: ESP-IDF:          v4.0-dev-141-g106dc0590[0m
[0;32mI (198) cpu_start: Single core mode[0m
[0;32mI (202) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (209) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (215) heap_init: At 3FFB7468 len 00000B98 (2 KiB): DRAM[0m
[0;32mI (221) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (227) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM[0m
[0;32mI (233) heap_init: At 3FFC82A0 len 00017D60 (95 KiB): DRAM[0m
[0;32mI (240) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM[0m
[0;32mI (246) heap_init: At 40078000 len 00008000 (32 KiB): IRAM[0m
[0;32mI (252) heap_init: At 40090850 len 0000F7B0 (61 KiB): IRAM[0m
[0;32mI (258) cpu_start: Pro cpu start user code[0m
[0;32mI (52) esp_core_dump_common: Init core dump to UART[0m
[0;31mE (53) esp_core_dump_common: No core dump partition found![0m
[0;32mI (53) cpu_start: Starting scheduler on PRO CPU.[0m
BTstack: setup
[00:00:00.061] LOG -- btstack_run_loop_freertos.c.260: run loop task 0x3ffbc0c4
[00:00:00.068] LOG -- btstack_run_loop_freertos.c.263: run loop init, queue item size 8
[00:00:00.075] LOG -- main.c.272: transport_register_packet_handler
[00:00:00.086] LOG -- btstack_tlv.c.48: TLV Instance 0x3f4079a4
[00:00:00.087] LOG -- l2cap.c.3512: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
HID service record size: 247
Device ID SDP service record size: 64
[00:00:00.100] LOG -- l2cap.c.3512: L2CAP_REGISTER_SERVICE psm 0x13 mtu 100
[00:00:00.106] LOG -- l2cap.c.3512: L2CAP_REGISTER_SERVICE psm 0x11 mtu 100
[00:00:00.113] LOG -- hci.c.2931: hci_power_control: 1, current mode 0
[00:00:00.119] LOG -- main.c.204: transport_init
[00:00:00.124] LOG -- main.c.220: transport_open
[0;32mI (128) BTDM_INIT: BT controller compile version [80351af][0m
[0;32mI (135) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (224) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0[0m
[00:00:00.528] LOG -- main.c.251: transport: configure SCO over HCI, result 0x0000
[00:00:00.529] LOG -- hci.c.4063: BTSTACK_EVENT_STATE 1
[00:00:00.530] EVT <= 60 01 01 
[00:00:00.533] LOG -- hci.c.1203: hci_initializing_run: substate 0, can send 1
[00:00:00.540] CMD => 03 0C 00 
BTstack: execute run loop
[00:00:00.545] LOG -- btstack_run_loop_freertos.c.180: RL: execute
[00:00:00.551] EVT <= 6E 00 
[00:00:00.554] EVT <= 0E 04 05 03 0C 00 
[00:00:00.557] LOG -- hci.c.1528: Command complete for expected opcode 0c03 at substate 1
[00:00:00.565] LOG -- hci.c.1203: hci_initializing_run: substate 2, can send 1
[00:00:00.572] CMD => 01 10 00 
[00:00:00.575] EVT <= 0E 0C 05 01 10 00 08 0E 03 08 60 00 0E 03 
[00:00:00.580] LOG -- hci.c.1975: Manufacturer: 0x0060
[00:00:00.585] LOG -- hci.c.1528: Command complete for expected opcode 1001 at substate 3
[... 552 lines cut ...]
[00:00:19.149] LOG -- hci.c.679: hci_send_acl_packet_fragments loop before send (more fragments 0)
[00:00:19.157] ACL => 80 00 0C 00 08 00 01 00 07 7B 04 00 41 00 41 00 
[00:00:19.164] LOG -- hci.c.698: hci_send_acl_packet_fragments loop after send (more fragments 0)
[00:00:19.172] LOG -- hci.c.707: hci_send_acl_packet_fragments loop over
[00:00:19.179] LOG -- l2cap.c.933: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x41
[00:00:19.185] EVT <= 71 02 41 00 
HID Disconnected
[00:00:19.190] LOG -- l2cap.c.1100: l2cap_stop_rtx for local cid 0x41
[00:00:19.196] EVT <= 13 05 01 80 00 01 00 
[00:00:19.200] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4294967295
[00:00:19.208] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 4294967295
[00:00:19.215] EVT <= 6E 00 
[00:00:19.218] CMD => 35 0C 05 01 80 00 02 00 
[00:00:19.222] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4294967295
[00:00:19.229] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 4294967295
[00:00:19.237] EVT <= 6E 00 
[00:00:19.240] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4294967295
[00:00:19.247] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 4294967295
[00:00:22.264] EVT <= 05 04 00 80 00 13 
[00:00:22.265] LOG -- hci.c.923: Connection closed: handle 0x80, A8:7D:12:01:BE:F1
[00:00:22.265] LOG -- hci.c.4132: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
[00:00:22.271] EVT <= 61 01 00 
[00:00:22.274] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4294967295
[00:00:22.281] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 4294967295
```

mringwal, 2019-03-25:
thanks for the extended version. You can disable ENABLE_LOG_DEBUG, it's really a last resort and adds a lot of noise. Could you convert the text log into PacketLogger format with btstack/tool/create_packet_log.py (and post it here)? It can then be opened & analyzed with Wireshark.

fluffymadness, 2019-03-25:
Thanks for your quick response. The log is the pairing request from my windows 10 computer to the esp32 hid keyboard example. It says connected in the windows 10 bluetooth settings after pairing.
This state stays for about 10secs then it jumps right to "device paired" loosing the connection.
[hci_dump_keyboard.zip](https://github.com/bluekitchen/btstack/files/3004155/hci_dump_keyboard.zip)

fluffymadness, 2019-03-25:
Hope the conversion worked. I just redirected the output of "make monitor" > test.log
and made the conversion with the python utility.

mringwal, 2019-03-25:
The conversion worked. BTstack rejects the incoming HID Interrupt connection, which is a bug. Are you on the master or the develop branch? If on master, please switch to develop and try again. (although there might be another issue, need to double check that).

fluffymadness, 2019-03-25:
I'm on the master branch. I'll try switching to dev.

fluffymadness, 2019-03-25:
it goes further now....Says Hid connected, start typing..but it stacktraces

[00:00:20.882] LOG -- l2cap.c.905: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr A8:7D:12:01:BE:F1 handle 0x80 psm 0x13 local_cid 0x42 remote_cid 0x42 local_mtu 100, remote_mtu 672, flush_timeout 0
HID Connected, please start typing...
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400fd1bf  PS      : 0x00060a30  A0      : 0x800fd76e  A1      : 0x3ffbbde0
0x400fd1bf: hid_device_request_can_send_now_event at /home/chainer/esp/esp-idf/components/btstack/src/classic/hid_device.c:996

A2      : 0x00000041  A3      : 0x3ffd144c  A4      : 0x3ffc1724  A5      : 0x00000080
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffbbdc0
A10     : 0x00000000  A11     : 0x00060823  A12     : 0x00060820  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x02070903  SAR     : 0x00000010  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000010  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

ELF file SHA256: 3f9a8fee8dbb2eaa71b87831a27f849344033cc28c5420ff2ad384f241ea6d47

Backtrace: 0x400fd1bf:0x3ffbbde0 0x400fd76b:0x3ffbbe00 0x4010a33a:0x3ffbbe70 0x400fffad:0x3ffbbe90 0x400fffdd:0x3ffbbeb0 0x400e3101:0x3ffbbed0 0x400e6cd9:0x3ffbbef0 0x400e6e66:0x3ffbbf30 0x400e29e4:0x3ffbbf50 0x400e2a35:0x3ffbbf80 0x400e7ba1:0x3ffbbfa0 0x400e766d:0x3ffbbfd0 0x400e2b6c:0x3ffbbff0 0x400d17f4:0x3ffbc010 0x4008d22d:0x3ffbc030
0x400fd1bf: hid_device_request_can_send_now_event at /home/chainer/esp/esp-idf/components/btstack/src/classic/hid_device.c:996

mringwal, 2019-03-25:
ok, that might be the other issue... I'll try to have a look this week and post and post and update here.

mringwal, 2019-03-28:
Hm. I've tried hid_keyboard_demo both on Mac (libusb) and ESP32. It worked fine.
Could you use the code from current develop without modifications (in the current code, there's no line 996  "hid_device_request_can_send_now_event at /home/chainer/esp/esp-idf/components/btstack/src/classic/hid_device.c:996"). If it still/again crashes, please post the .pklg again.

fluffymadness, 2019-03-29:
Steps that I did:

- Pulled latest develop branch.
- Changed folder to /port/esp32/
- Running ./integrate_btstack.py
- Then I went to /port/esp32/examples/hid_keyboard_demo/
- make flash
- make monitor
- Then I paired bluetooth via windows 10

Here is the log: 
[newlog.zip](https://github.com/bluekitchen/btstack/files/3022396/newlog.zip)

Did you build the example with the latest commit of the esp-idf or are you on an older maybe more stable version?

As for this line, /home/chainer/esp/esp-idf/components/btstack/src/classic/hid_device.c:996....I'm wondering myself, because it is, as you said, not in the code.
I checked the file in this directory and line 996 is: l2cap_disconnect(hid_device->control_cid, 0); // reason isn't used

Thanks again for helping me.

mringwal, 2019-03-30:
Thanks for the retry and the log. In the new log, there's no hint of a null pointer access. Up to the crash, everything looks fine to me.

I did use the current master branch, 5beb2802 (from Sun March 17).

Could you try to connect from another device, e.g. a smartphone?
(don't have a better idea at the moment..)

fluffymadness, 2019-03-31:
Using another device did trick.
Connecting from my phone works. Keyboard example is doing fine then.
Connecting via Windows 10 instant stack-traces everything...doesn't matter if I use the internal bluetooth of my notebook or an usb bluetooth stick. That's interresting.

mringwal, 2019-04-02:
Interesting! Maybe it's a NULL pointer after all.

Thanks for helping with testing and sorry I couldn't test with Windows myself yet. I hope to get to that this week somehow.

fluffymadness, 2019-04-02:
I'm happy to help. Hopefully the bug is somehow reproducible. Another thing I've noticed is when doing an sdp tool query on linux the connection never terminates after I get the records of the bluetooth device (keyboard example). It goes on until I manually kill sdptool.
When I tested this with a dualshock 4 and a bluetooth audio receiver, the connection terminated after 5-10 seconds when I got the sdp records.

mringwal, 2019-04-11:
Hi! I've found the bug, it really was BTstack's fault. After the last change, incoming HID requests (e.g. the one from Windows) directly caused a NULL pointer access. Please update develop, try again, and let me know.

For the Linux SDP tool: could you enable HCI logging again and post a .pklg file.

fluffymadness, 2019-04-13:
Hi. Thanks for fixing the bug. I'll try to get to testing this week.

www378660084, 2019-04-14:
"pull request 207" fixs the bug

mringwal, 2019-04-15:
@www378660084 see comment on your pull request. Code on develop has the fix.

fluffymadness, 2019-04-16:
Windows 10 is working fine now, but I have another problem when trying to re-build the hid keyboard example. I made a make clean before the make flash, but it didn't help.
Maybe because of this commit? https://github.com/bluekitchen/btstack/commit/a71df68ad195a359a83f8b5163e365003a7bd0d8

`CC build/bootloader_support/src/secure_boot_signatures.o
In file included from /home/chainer/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:22:0:
/home/chainer/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c: In function 'esp_secure_boot_verify_signature_block':
/home/chainer/esp/esp-idf/components/btstack/3rd-party/micro-ecc/uECC.h:42:24: error: called object is not a function or function pointer
 #define uECC_secp256r1 3
                        ^
/home/chainer/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:93:33: note: in expansion of macro 'uECC_secp256r1'
                                 uECC_secp256r1());
                                 ^
/home/chainer/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:38:20: warning: passing argument 3 of 'uECC_verify' makes pointer from integer without a cast [-Wint-conversion]
 #define DIGEST_LEN 32
                    ^
/home/chainer/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:91:33: note: in expansion of macro 'DIGEST_LEN'
                                 DIGEST_LEN,
                                 ^
In file included from /home/chainer/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:22:0:
/home/chainer/esp/esp-idf/components/btstack/3rd-party/micro-ecc/uECC.h:236:5: note: expected 'const uint8_t * {aka const unsigned char *}' but argument is of type 'int'
 int uECC_verify(const uint8_t public_key[uECC_BYTES*2],
     ^
/home/chainer/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:89:16: error: too many arguments to function 'uECC_verify'
     is_valid = uECC_verify(signature_verification_key_start,
                ^
In file included from /home/chainer/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:22:0:
/home/chainer/esp/esp-idf/components/btstack/3rd-party/micro-ecc/uECC.h:236:5: note: declared here
 int uECC_verify(const uint8_t public_key[uECC_BYTES*2],
     ^
make[1]: *** [/home/chainer/esp/esp-idf/make/component_wrapper.mk:290: src/secure_boot_signatures.o] Error 1
make: *** [/home/chainer/esp/esp-idf/make/project.mk:578: component-bootloader_support-build] Error 2`

mringwal, 2019-04-16:
That's most likely related to the a71df68. What version (git commit) of esp-idf do you use? 
I did use the latest version from their master when making this commit.

fluffymadness, 2019-04-16:
I'm on the latest dev branch commit.

fluffymadness, 2019-04-16:
ah ok...will update esp-idf then and try again.

fluffymadness, 2019-04-16:
Thx, works fine now. Here is the non-terminating sdptool request log for the keyboard example.
[sdptool.zip](https://github.com/bluekitchen/btstack/files/3085015/sdptool.zip)

fluffymadness, 2019-04-16:
I made a small pull request for a working hid-gamepad example. Your last fix got it to work in windows 10 too.

mringwal, 2019-04-16:
Thanks for the sdptool log. In it, we see bluez creating an SDP connection and requesting the SDP record for handle X, with X = 0x10000, 0x10001, 0x10002, ...

The SDP Record Handle is a 32-bit number, asking one by one starting at 0x10000 is pretty odd.
BTstack responds for 0x10001 - which is the HID SDP record and 'invalid handle' otherwise.

Usually, an SDP client ask for 'give me the record handles for all records that match X', or, 'give me the following properties for all records that macht X'. I'll close the ticket: HID is working, no idea about the SDP query.

closed 2019-04-16 by mringwal