bluekitchen/btstack #204: Sending a scan response leaks memory

peet1993, 2019-04-03:
Hi,
Could be that I'm just using it wrong, but I have problems with the scan response function.
I'm using scan responses to inform scanning clients about the name of the device. For that, I use
```
const uint8_t scan_resp_data[] = {
    0x09, BLUETOOTH_DATA_TYPE_COMPLETE_LOCAL_NAME, 'F', 'A', 'K', 'E', ' ', 'R', 'C', 'U'
};
gap_scan_response_set_data(sizeof(scan_resp_data), scan_resp_data);
```
This creates a malformed scan response with extra data. I narrowed it down to the command assembler `hci_cmd_create_from_template` in hci_cmd.c. The format symbol A gets replaced by copying 31 bytes into the buffer, but the source only contains "0x090x09FAKE RCU" (10 Bytes, data length + type + data). The remaining bytes get read from adjacent memory which could be anything. It looks like it doesn't get padded correctly, which works fine for advertisments (confirmed in the hci_dump.pklg).
The resulting packet looks like this:
[image]

mringwal, 2019-04-03:
Thanks for pointing this out. I think it's not actually wrong - the core spec says "only the significant part of the Advertising_Data should be transmitted in the advertising packets, as defined in [Vol 3] Part C, Section 11." - but it's bad form nevertheless to send random data to the Controller.

I need to think about this a bit more. The problem is that va_arg only provides a pointer but not length. For the used commands (set scan response data, set adv data), the param before contains the length, but that's not particularly elegant.

mringwal, 2019-04-03:
quick fix. if you provide 31 bytes as adv data, it's correct. I now also remember, that BTstack gap implementation is using the commands instead of the user, so there should be a way to zero the remaining bytes. Let me check.

mringwal, 2019-04-03:
ah! got it. BTstack does the padding, but passes the wrong array. Fix see https://github.com/bluekitchen/btstack/commit/214bfd60e6dfaee7b5cce3ac964d5a1297e758a9

Thanks again!

closed 2019-04-03 by mringwal

peet1993, 2019-04-03:
Wow, thanks for the quick fix, impressive!

mringwal, 2019-04-04:
Thanks! It was almost implemented correctly, just a copy&paste error.