bluekitchen/btstack #208: le_data_channel_server hangs

AnagramEngineering, 2019-04-15:
I'm trying to connect to the example program

_le_data_channel_server_ 

using the ios app

https://github.com/bluekitchen/CBL2CAPChannel-Demo

Unfortunately when trying to connect the server side running on raspberry pi shows 

_Connected, requesting conn param update for handle 0x0040_

The function 

_didConnect_ 

on the ios side is never invoked.

AnagramEngineering, 2019-04-15:
I had to comment out 

 gap_request_connection_parameter_update(le_data_channel_connection.connection_handle, 12, 12, 0, 0x0048);

in le_data_channel_server.c to make it work.

closed 2019-04-15 by AnagramEngineering

mringwal, 2019-04-16:
Hi! I just tried on an iPhone 5 SE with the sources as is and it works. Could you post the /tmp/hci_dump.pklg file for the failing setup?

mringwal, 2019-04-16:
Btw. what iOS version did you try?

AnagramEngineering, 2019-04-17:
I'm using IOS 12.0
iPhone 6 and a Raspberry PI model 3 B V1.2

It's reproducible on my side, if commented out it works, if it's there it fails.

AnagramEngineering, 2019-04-17:
Hi Matthias, 

The hci_dump output looks a bit strange (at least when viewing with vim/cat etc.) 

You can find it [here](http://a-ds-1.anagram.at/hci_dump.pckglog).

I was also a bit sursprised about the speed, i reached ~1KB/sec, but would have expected a higher troughput. If you run it (with the interval change commented in) what speed do you reach? 

Great project btw!

mringwal, 2019-04-17:
Hi. You can open the  hci_dump.pklg with Wireshark. It looks much better than.

Speed: according to the log, BTstack uses the H5 transport which doesn't work well on POSIX systems. There's a new h5-new branch that fixes that, but it doesn't contain the Raspi port.

On iPhone 5 SE (not the fastest), I get 13-15 kB/s to BTstack running on my Mac with an USB Bluetooth Dongle. The iPhone is the limiting factor. The LE Streamer example between two CSR USB dongles sends 50 kB/s using GATT and LE Data Channels should be a tiny bit faster due to reduced overhead.

--

In the log, there's no further event after sending an HCI LE Encrypt Command. I've already reported that here: https://community.cypress.com/thread/8424

I guess it would make sense to calculate AES128 in software on an Raspi.

How this is related to the connection parameter change is unclear, but I'd say it's a bug of the 4343.

--

Please use a different Raspberry Pi model were H4 is used, or use an external USB dongle with the libusb port to get higher/full speed.