bluekitchen/btstack #210: a2dp_sink channel close automatic on esp32

akashravva, 2019-04-16:
I was trying to connect the esp32 to phone using classic Bluetooth where phone acts as a source and esp32 as a sink. And when the connection is established the channel that's allocated for the stream is getting suspended automatically after 10 seconds.

mringwal, 2019-04-16:
What OS do you have on your phone?
Did you start to play music?

Please enable packet logger in esp-idf/components/btstack/main.c, try again, and convert the output to .pklg using btstack/tool/create_packet_log.py, then post here.

akashravva, 2019-04-17:
Android Pie 
Yes i started playing music.

And i think i have found the reason after referring to Manual,
Below might be the explanation. Please correct if I'm wrong.

Explanation from the Manual as follows:

3.1. Single threaded design. BTstack does not use or require multi-threading.
It uses a single run loop to handle data sources and timers. Data sources represent
communication interfaces like an UART or an USB driver. Timers are used
by BTstack to implement various Bluetooth-related timeouts. For example, to
disconnect a Bluetooth baseband channel without an active L2CAP channel after
20 seconds.

akashravva, 2019-04-17:
I have a query on connectivity related to Bluetooth.
I need to route the audio data from phone (acting as a source) to BTStack Implemented ESP32 (acting as a sink) ..........and the same audio data has to be transferred to Headphones(acting as a sink) from ESP32(acting as a source) simultaneously with out any delay.

So basically ESP32's BLUETOOTH has to change the roles from SOURCE TO SINK continuously to route the data between Phone to headphones. 

For this I used Audio Duplex example .....And i am not knowing how to setup the source of audio data when esp is acting as a sink and ......how to setup sink when esp is acting as a source!!!

mringwal, 2019-04-17:
The single threaded design is not a problem with music playback. It worked for me and others on ESP32 before. Please post packet log as described earlier.

Please ask general questions on the BTstack mailing list.

akashravva, 2019-04-17:
The log is attached.
[sink_log.txt](https://github.com/bluekitchen/btstack/files/3089172/sink_log.txt)

I was asking you about the [audio_duplex demo] in the BTStack on how to get the sink and source instance. 
It says Audio source not setup 
and the sink setup is success.

Thanks @mringwal for looking into the issue.

mringwal, 2019-04-17:
Please enable packet logger in esp-idf/components/btstack/main.c, try again, and convert the output to .pklg using btstack/tool/create_packet_log.py, then post here.

--

Audio duplex is mainly for testing the audio driver. Sound input is not implemented for the ESP32 (there isn't a built-in microphone e.g.)

akashravva, 2019-04-17:
[Log.zip](https://github.com/bluekitchen/btstack/files/3089725/Log.zip)

Thanks for the info on Audio_Duplex.

akashravva, 2019-04-20:
Hi @mringwal Did you have a look at the logs?
The same problem still persists,
I can connect to the phone from the command interface and play a song in the phone but, it is getting disconnected exactly after 20 seconds.
But the vice-versa works without any disconnection.(i.e initiating connection from phone to ESP)

mringwal, 2019-04-20:
In the log, there's a LMP Disconnect - indicating a problem with radio connection between the  devices. Up to that point, everything looks as expected.
(The disconnect seems to be roughly 30 seconds after the connect according to the  log).

No idea at the moment. You could try with other devices (computers, smartphones...)

With full HCI logging, its probably not possible to listen to  music as the log printing is slower than the data. For testing, it's ok to  check if the connection stays alive first.

mringwal, 2019-09-06:
Closing due to missing feedback.

closed 2019-09-06 by mringwal