bluekitchen/btstack #211: CSR8811 A12 Dual Mode

apmorton, 2019-04-17:
simultaneous dual mode operation does not work correctly on the A12 revision out of the box.
It does work out of the box on CSR8811 A08 chips.

The A12 revision has a new boolean pskey which defaults to false.
Programmer name:  `BLE_SLAVE_ALLOW_BREDR`
Friendly name: `Allow BLE slaves to partake in BR/EDR activities.`

Help info on this pskey from pstool:
```
This PSKey controls internal link manager behaviour. Do not change
it except on specific advice from CSR.
This PSKey exists to disable specific BLE and  BR/EDR role
combinations that are prohibited by the Generic Access Profile.
Enabling this PSKey will allow the prohibited role combinations,
which will potentially allow BLE links to clash with BR/EDR links
with unpredictable results and possible loss of throughput.
```

Enabling this pskey appears to work and the A12 chip behaves like the A08 does: allowing a BLE slave device to also make BR/EDR connections.

I cannot comment on the warnings from CSR in the above help information - they likely turned this behavior off by default for a reason in the newer chip revision.  All I can say is that enabling it fixes the issues I was having.

I added the following code in the csr chipset `init_script` after setting `HCI_NOP_DISABLE` (although the order is not really important so long as its before the warm reset)
```c
// 0x0121: Set BLE_SLAVE_ALLOW_BREDR
0x00, 0xFC, 0x13, 0xc2, 0x02, 0x00, 0x09, 0x00, 0x01, 0x00, 0x03, 0x70, 0x00, 0x00, 0x21, 0x01, 0x01, 0x00, 0x08, 0x00, 0x01, 0x00,
```

mringwal, 2019-04-17:
Thanks for the detailed report and suggestions. Does the A12 has build 10520? The PSKEY list for 10520 contains the BLE_SLAVE_ALLOW_BREDR.

What combination of BLE and BR did you try that didn't work with the key set to false?

From the PSKEY name, it sounds like BLE Slave is not supported when using BR/EDR at the same time. I don't remember any restrictions between BLE and BR/EDR in the Bluetooth Spec.

I'd like to reproduce your issue and then add the PSKEY to the CSR list inside an #ifdef  ENABLE_CSR_... block.

apmorton, 2019-04-17:
Can check on build number later.

- Enable any BR profile (register any l2cap psm - sdp for example)
- configure ble slave to advertise

During hci initialization run the `le set advertise enable` hci command will fail with an error code `0x0C` (command disallowed).

Note, the Bluetooth stack will not print any error - it seems the command complete events are completely ignored.

mringwal, 2019-04-18:
Thanks for the details. It would fit the name of PSKEY.

I've checked and only have access to 2 different CSR8811A08 (one from Microchip BT805 Bluetooth Audio Devkit, and the Laird DVK-BT830), but no A12. 

Do you know a CSR8811A12 devkit or a module which is hand solderable?

Yes, BTstack does not check the result. Asides from printing an error, I don't see how this should be handled. Reporting an error wouldn't hurt though. (and there are a few during HCI Init which could be avoided, but there wasn't any harm either).

apmorton, 2019-04-18:
Having an error printed on all hci command completion events with a failure status code would definitely be helpful, even if the rest of the hci state machine doesn't do any error handling with it.

Tracking down this issue was a bit of a pain. Other than a single 0x00 changing to 0x0c when hci dumping was enabled there was no differences to indicate something went wrong (besides the obvious "my ble device doesn't show up").

Unfortunately I don't know of any A12 devkits or modules at all. I am using the WLCSP package on a custom pcb that is assembled by machine. I honestly doubt anyone makes them. According to qualcomm the A08 revision is still in full production, and they have no plans to stop producing them, so people just haven't bothered updating their designs.

For what it's worth, the A08 and A12 are pin compatible, so if you have the tools available to you it's not _impossible_ to swap the chip on one of your modules.

If you are unable to source a board with the A12 on it I may be able to send you something if that helps.

mringwal, 2019-04-18:
Thanks for the details. Pro Tip: if you enable HCI dump and open it in Apple's PacketLogger, events with an error code are shown in red, so it's easy to spot them. I don't think Wireshark does that by default, but I guess it could be configured to do so as well.

Right now, I have a development setup from a customer that runs Android and has an A12, so I could try on that one (it's just not really setup for a quick test). I'll probably update the documentation and add the PSKEY conditionally, so you don't need to modify the csr chipset driver code.

apmorton, 2019-04-18:
The other thing to note with the A12 is that Qualcomm has actually released a patchset to fix some software defects. Of course you must be under NDA to access this information.

Including the patchset in your stack may be legally questionable - but I'm not a lawyer.

The patchset is just a series of pskeys to write before warm reset.

Maybe it could be useful to just have a way to include a bunch of pskeys for the csr chipset code to write during startup?

The full hci wrapped bccmd protocol required to set a pskey is pretty wasteful space wise as well - might be nice to have the chipset driver construct the full hci command from a byte array of pskey index, size, payload segments?

apmorton, 2019-04-22:
>Does the A12 has build 10520?

Just confirmed, yes.
```gem (Hardware ID 0x2429) firmware version 10520.```

mringwal, 2019-04-23:
Thanks for the confirmation and the additional details. Of course, we cannot release a patch unless it's has been made publicly available.

Good point about providing the ability to include a custom set of HCI commands, or as you've suggested even a list of PSKEYs.

Totally unrelated CSR8811 question: did you try using HFP Wide-band speech? I'm trying to figure out which option: PCM/I2S, SCO over HCI, or, using the DSP on the Controller itself,  is the least painful...

apmorton, 2019-04-26:
Have never played with HFP at all.

The only audio stuff I've done over Bluetooth is talking to a device that embeds SBC frames at the end of it's HID reports - definitely not a common scenario I don't think.

I've looked through all the DSP documentation on the CSR8811 though, and if I had to guess I would say that is the *most* painful of the options.

The DSP is only available in the A12, at least as far as I can tell, so that limits it's usefulness pretty significantly for most people. According to our Qualcomm reps the A08 is still by far the most popular in North America.

I've also never actually found any code that makes use of the DSP on the 8811, so getting it to do anything would probably be a whole lot of trial and error while you work through the docs.

At least the other two options are somewhat standardized.

mringwal, 2019-04-30:
Thanks for your view on HFP. 

I've created a small python tool that converts a .psr file (e.g, the patch for the A12) into a sequence of HCI commands (wasting about 32 bytes per PSKEY) and provided a .psr of the current btstack default config. Could you try if you can use the A12 patch with this?