bluekitchen/btstack #212: How could TEST HFP  with [ Bluetooth Music BOX--has mic and speaker]

mailtoxiong, 2019-04-23:
How could TEST HFP  with [ Bluetooth Music BOX--has mic and speaker]

--taken few days with the open source code , pray for someone's helps .

mringwal, 2019-04-24:
Looks like you're running BTstack on Windows. What Windows version do you use? SCO over HCI, which is needed for HFP, requires Windows 8.1 or higher. See warning in your last screenshot.

mailtoxiong, 2019-04-24:
@mringwal , thanks for your reply .
I knew the reminder in log,and i am using win 7 sp1 version system . If have no more choice  for using HFP in win 7 ,such as add dll  or sth else ?  Thanks a lot !

mringwal, 2019-04-24:
Well, BTstack is very flexible. I've used WinUSB on Windows and it support Isochronous USB Endpoints from Windows 8.1. I guess it would be possible to write a Windows kernel driver and use that instead with older versions. I don't know much about Windows, though.

mailtoxiong, 2019-04-26:
Hello @mringwal .Thank you so much for patiently answers. 
After used btstach in Windows 8 (previous is win 7 sp1,at last changed to Win 8.1 sytem ).
Now ,i am already moved the source code to embeded sytem (like STM32,as below shown)
[image]
My question is :
  a. When i tried  to send SINE SCO data  :
      #define SCO_DEMO_MODE      SCO_DEMO_MODE_SINE
      
      In sco_demo_util.c ,
      void sco_demo_send(hci_con_handle_t sco_handle)
{
....

       // test silence

# if 0
       //  original open source code
       memset(sco_packet+3, 0, sco_payload_length);
# else
       // modified code  ,  2019.04.29
       //scotest_Array   ,sine data
        memset(sco_packet+3, 0x6A, sco_payload_length);
     //  memcpy(sco_packet+3, scotest_Array, sco_payload_length);
# endif

....
}

     Means ,i added some code for send all data like 0x6a..
      Always  shown the error logs like:
       
[image]
    
        
       Could you help me or give me some advice.Anyway ,thank you so much .

mringwal, 2019-04-26:
What Bluetooth Controller (chip) do you use? Is it connected via UART or USB? (also closing as there's no issue on Windows 8.1). Please post on the developer mailing list with regular questions.

closed 2019-04-26 by mringwal

mailtoxiong, 2019-04-28:
What Bluetooth Controller (chip) do you use? 
-----RTK8761 ,Realtek 
Is it connected via UART or USB? (also closing as there's no issue on Windows 8.1). 
-----yes ,via this for  firmware download (seems no issue on Windows 8.1,But not works on Embeded PCB  system )
Please post on the developer mailing list with regular questions.
-----How could i mail the developer list ?

mringwal, 2019-04-28:
I have no experience with the RTK8761 - didn't see much documentation about it.

It seems to be an USB device. If it works with Windows 8.1 but not with embedded, your embedded driver probably is not working correctly.

Mailing list: http://groups.google.com/group/btstack-dev

mailtoxiong, 2019-04-30:
Actually , maybe you know google link and google search  in locked by China ,so,can not link in this Mailing list website .
Mailing list: http://groups.google.com/group/btstack-dev

If HAS any other website for instead and i could do mailing .

Thanx a lot for ever took more interrupts for you .

mringwal, 2019-05-06:
Sorry. The mailing list is hosted by Google. It looks like it's necessary to create a "Google Account" for your email address to join - which is not possible for you then.