bluekitchen/btstack #213: ESP-IDF component micro-cc moved

pedrominatel, 2019-04-23:
The component micro-ecc was moved from ESP-IDF folder ESP_IDF/components/ to ESP_IDF/components/bootloader/subproject/components

ESP-IDF version: 4.0dev (master branch)

To solve this I've created a symbolic as turn around solution.

pedrominatel, 2019-04-23:
Error message:

`make: *** No rule to make target 'component-micro-ecc-build', needed by 'component-btstack-build'.  Stop.`

mringwal, 2019-04-24:
hi Pedro. yupp, that's correct. I've updated the Makefile on the BTstack develop branch to use the micro-ecc from BTstack. Now, develop doesn't compile with ESP-IDF 3.2...

I guess the python integrate script could check what's needed if a general solution would be to complicated to support 3.2 and 4.x+

mringwal, 2019-04-26:
hi. I've found a way to have BTstack only use it's internal copy, see 5ec47f0f9 (on develop branch)

closed 2019-04-26 by mringwal

pedrominatel, 2019-04-30:
Thanks @mringwal for supporting this issue!

Matheusmbgm, 2019-05-24:
Hi, someone have a update on this issue? I had a project with esp idf v3.2 using btstack and i had to update my esp idf to V4.0 (i guess thats the version number, its the newest version) so i could use OTA Security. After i did the update i have the same problem related in this topic. I tried to update btstack lib but i still have this problem.