bluekitchen/btstack #214: BTstack port for WICED platform for CYW20706 WICED

makubica, 2019-05-04:
Do you have a port or plan to support the CYW20706 WICED BT chips?
From all what I read it is not clear if it can works with wiced and this cypress chip. I think I have read that you only support the broadcom but on the following site you said btstack is ported.
http://bluekitchen-gmbh.com/btstack/ports/existing_ports/#sec:wiced-h5Port

Thanks
Manfred

mringwal, 2019-05-04:
Hi Manfred. Do you want to run BTstack on the CYW20706 itself? 

I don't know if the HCI interface is supported on the 20706 and didn't dig deeper to find out yet. (When I got the dev kit, I was trying to see if it could be used via HCI, which is also unclear, but I was suggested to use the CYW20704 instead).
Please have a look at the ESP32 for an alternative all-in-one solution.

If you have a main MCU (e.g. some ARM Cortex M) already, you could use BTstack with the CYW20704 (same family, just with HCI UART interface). 

Best
 Matthias

makubica, 2019-05-04:
Hi Matthias,
I have taken a look in the wicked port code and it seems that it could work. The 20706 has an USB uart and the board with wicked studio has two com port one "HCI Peripheral" and one  "peripheral uart".
I need the function of this cypress board (classic mode and other) but I have seen that the ti CC256 chip has nearly the features of this board. What do you think? For me it looks that it is easier in the beginning with the ti and USB under mingw64 and later if I have developed the functions to switch to the cypress and to embed it in the real mcu.
One question:
How is the btstack splitted between host and client. Is it possible to bring the btstack main code to Bt chip and let only run part on the main mcu or is this impossible?
In the wicked port it seems  that the btstack is under libraries and therefore on the cypress chip but in your blog you have posted a connection between the ti chip and stm32f4 and here the btstack runs on the stm32.  

Thanks
Manfred

mringwal, 2019-05-04:
Quick follow up. I think the HCI UART isn't a full HCI on the CYW20706 - there's a 'WICED HCI' which looks similar to regular HCI, but the commands are different and it is used as an IPC for higher layer protocols..  If you want to use it as a regular Bluetooth Controller via HCI, check out the CYW20704 (see https://www.lairdconnect.com/wireless-modules/bluetooth-modules/bluetooth-5-modules/bt86x-series)

It's best to start on a desktop system, either with an USB Bluetooth dongle or a dev kit connected via USB-Serial. There, the hci_dump.pklg file can be viewed in Wireshark.

CYW20704 is newer than CC2564, but this is only related to BLE features. CC2564 is currently required for HFP Wide-band speech, though.

BTstack generally is on the Host MCU and uses the standard HCI UART (H4/H5) protocols to communicate with the Controller. If the stack would be on the Bluetooth Controller, you'd need another IPC.

makubica, 2019-05-05:
But the advantage of Wiced HCI is that it runs on the chip. This makes it flexible if you have e.g. not so much memory and cpu power on the Host MCU.  You said generally it runs on the host. What must be done that btstack runs on the Controller?
The point is that at the moment I can not decide which chip we take. I need a flexible library to switch each time in the development process to an other chip if the project needs this and at the end the price of the chip is also a very relevant thing.

Thanks for the Laird link. This is an very interesting chip. It can easily be mounted on our Host MCU.

I read that the btstack HID is not complete. What is missing?

makubica, 2019-05-05:
Do you know how many simultaneous connections the CYW20704 can hold in BT classic? 
It would be great if you could set this information on your web site in the chips table.

mringwal, 2019-05-06:
BTstack uses the Host Controller Interface to communicate with the Bluetooth Controller, and it is defined in src/hci_transport.h. On ESP32 and on devices running the Zephyr BLE Link Layer, a virtual HCI interface is used. In WICED/20706, I don't see an include file for hci, only some symbols that are promising in bthci.o...

HID Device is fully implemented and tested against PTS, but it hasn't bee included in an official Bluetooth SIG listing yet.

As far as I know, all relevant Bluetooth Controllers that support Bluetooth Classic support up to 7 ACl connection in Master role, but I didn't do any tests on that. If you create the connection, the Controller is in Master role. Most also support 1-2 additional connections in Slave mode.

closed 2019-05-06 by mringwal