bluekitchen/btstack #215: In test HFP ,How could changed the music (data) in

mailtoxiong, 2019-05-05:
[image]

#define SCO_DEMO_MODE_SINE		 0
#define SCO_DEMO_MODE_ASCII		 1
#define SCO_DEMO_MODE_COUNTER	 2
#define SCO_DEMO_MODE_55         3
#define SCO_DEMO_MODE_00         4
#define SCO_DEMO_MODE_MICROPHONE 5

Dear Friend @mringwal ,
if i wanna test music data , which  SCO_DEMO mode ,should be selected ?
 Means want to play must in HFP mode ,instead of real-time call ,just send some music data .

Thanks a lot !

Your sincerely 
 Kerwin

mringwal, 2019-05-05:
hi. Well, you can send music in sco_demo_send as well. You need to provide PCM 16 bit data in little endian at 8 kHz for CVSD or 16 kHz for mSBC. Check out the mod_player, I think it can also render the music in 8/16 kHz.

mailtoxiong, 2019-05-06:
Yes , i noticed the  folder: modes --> file:  nao-deceased_by_disease.c  
         mod_data[] data could be changed and filled with [.mod ] music , i tried and it works well in A2DP.

[image]

       Well , i will try it in HFP mode and feed back the result .

mailtoxiong, 2019-05-07:
i am in standing of this mattet ,i saw some latency in audition ,but do not know why .
[image]

mringwal, 2019-05-07:
Sending SCO audio is very tedious, as the data needs to be sent at the exact time.
It works well with TI CC2564 and UART, where there's an explicit SCO flow control and a some packets can be buffered. It also works well, where USB Isochronous Endpoints work correctly.

What setup is this?

closed 2019-05-07 by mringwal

mailtoxiong, 2019-05-08:
Dear mringwal  ,i used Realtek RTL8761  +STM32,
This problem really consusing me for few days .

In your OK case :TI CC2564 and UART
you also used the Master code ,or any branch code?

So thanks for your replies ,and i need go fowarding to resolve this problem .

mringwal, 2019-05-08:
Ti CC2564(A,B,C) works with master or develop branch.

With other Controllers from Cypress/Broadcom or CSR/Qualcomm, the current version on develop works mainly on an embedded device. However, it requires to send at regular interval (+1/-1 ms) and that can easily be disturbed. I'm afraid with these, I2S/PCM is the only successful approach. 

No experience with RealTek, there are no datasheets available either.

aadnesd, 2019-07-04:
Hi, I'm working with the a2dp_source_demo and want to use one of my own files to listen to BT headset. The nao-decease works fine. I tried to change the mod_data to the hex data of my file (originally wav), but even though I'm able to flash I now just don't hear anything. Any solutions?