bluekitchen/btstack #220: No rule to make target 'component-micro-ecc-build' after update esp-idf.

Matheusmbgm, 2019-05-24:
Hi, someone have a update on this issue? I had a project with esp idf v3.2 using btstack and i had to update my esp idf to V4.0 (i guess thats the version number, its the newest version) so i could use OTA Security. After i did the update i have this problem: "No rule to make target 'component-micro-ecc-build'".. I tried to update btstack lib to the newest version aswell but it didn't work. I did read this topic https://github.com/bluekitchen/btstack/issues/213 but i couldn't make it work.

Sorry if i'm broking any forum rules. I sent a message on tthe topic that i mentioned before, but it was closed and i tought maybe that would not be a polite thing to do. I'm new to forums, sorry.

Waiting for an answer.

pedrominatel, 2019-05-24:
Try to checkout the "develop" branch and make again.

Matheusmbgm, 2019-05-24:
it work out! Thank you very very much!

closed 2019-05-24 by Matheusmbgm