bluekitchen/btstack #221: Stability Issues with libusb & linux

skoperst, 2019-05-26:
I noticed bad stability when starting & stopping SCO under Linux & libusb.
That means the btstack crashes, or the bluetooth device starts to have slowness/degraded audio/crashes.

I tested develop branch (26687d14499e21cced002eae78e784d17656d908), but this happens in master as well.

How to reproduce:
1. Use example/hfp_hf_demo.c with a modern smartphone.
2. Start call, then end call, then start again.. 
3. After a while(can be even 2 calls) either the btstack or the hci device will give an error/ experience degraded performance.

A valgrind run gives some errors:
```
==6793== Invalid read of size 8
==6793==    at 0x10F8C7: usb_process_ds (hci_transport_h2_libusb.c:539)
==6793==    by 0x10F95F: usb_process_ts (hci_transport_h2_libusb.c:566)
==6793==    by 0x112606: btstack_run_loop_posix_execute (btstack_run_loop_posix.c:229)
==6793==    by 0x10D8EC: btstack_run_loop_execute (btstack_run_loop.c:189)
==6793==    by 0x10C0E9: main (main.c:230)
==6793==  Address 0x5c52488 is 152 bytes inside a block of size 244 free'd
==6793==    at 0x4C30D3B: free (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==6793==    by 0x11075A: usb_sco_stop (hci_transport_h2_libusb.c:879)
==6793==    by 0x111C9F: usb_set_sco_config (hci_transport_h2_libusb.c:1395)
==6793==    by 0x117694: hci_shutdown_connection (hci.c:949)
==6793==    by 0x11B76B: event_handler (hci.c:2545)
==6793==    by 0x11BAE7: packet_handler (hci.c:2659)
==6793==    by 0x10F54F: handle_completed_transfer (hci_transport_h2_libusb.c:448)
==6793==    by 0x10F8DD: usb_process_ds (hci_transport_h2_libusb.c:540)
==6793==    by 0x10F95F: usb_process_ts (hci_transport_h2_libusb.c:566)
==6793==    by 0x112606: btstack_run_loop_posix_execute (btstack_run_loop_posix.c:229)
==6793==    by 0x10D8EC: btstack_run_loop_execute (btstack_run_loop.c:189)
==6793==    by 0x10C0E9: main (main.c:230)
==6793==  Block was alloc'd at
==6793==    at 0x4C31B25: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==6793==    by 0x4E44F4E: libusb_alloc_transfer (io.c:1265)
==6793==    by 0x110612: usb_sco_start (hci_transport_h2_libusb.c:855)
==6793==    by 0x111CB3: usb_set_sco_config (hci_transport_h2_libusb.c:1399)
==6793==    by 0x11A95A: event_handler (hci.c:2160)
==6793==    by 0x11BAE7: packet_handler (hci.c:2659)
==6793==    by 0x10F54F: handle_completed_transfer (hci_transport_h2_libusb.c:448)
==6793==    by 0x10F8DD: usb_process_ds (hci_transport_h2_libusb.c:540)
==6793==    by 0x10F95F: usb_process_ts (hci_transport_h2_libusb.c:566)
==6793==    by 0x112606: btstack_run_loop_posix_execute (btstack_run_loop_posix.c:229)
==6793==    by 0x10D8EC: btstack_run_loop_execute (btstack_run_loop.c:189)
==6793==    by 0x10C0E9: main (main.c:230)
==6793== 
==6793== Invalid read of size 1
==6793==    at 0x10F51C: handle_completed_transfer (hci_transport_h2_libusb.c:447)
==6793==    by 0x10F8DD: usb_process_ds (hci_transport_h2_libusb.c:540)
==6793==    by 0x10F95F: usb_process_ts (hci_transport_h2_libusb.c:566)
==6793==    by 0x112606: btstack_run_loop_posix_execute (btstack_run_loop_posix.c:229)
==6793==    by 0x10D8EC: btstack_run_loop_execute (btstack_run_loop.c:189)
==6793==    by 0x10C0E9: main (main.c:230)
==6793==  Address 0x5c52469 is 121 bytes inside a block of size 244 free'd
==6793==    at 0x4C30D3B: free (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==6793==    by 0x11075A: usb_sco_stop (hci_transport_h2_libusb.c:879)
==6793==    by 0x111C9F: usb_set_sco_config (hci_transport_h2_libusb.c:1395)
==6793==    by 0x117694: hci_shutdown_connection (hci.c:949)
==6793==    by 0x11B76B: event_handler (hci.c:2545)
==6793==    by 0x11BAE7: packet_handler (hci.c:2659)
==6793==    by 0x10F54F: handle_completed_transfer (hci_transport_h2_libusb.c:448)
==6793==    by 0x10F8DD: usb_process_ds (hci_transport_h2_libusb.c:540)
==6793==    by 0x10F95F: usb_process_ts (hci_transport_h2_libusb.c:566)
==6793==    by 0x112606: btstack_run_loop_posix_execute (btstack_run_loop_posix.c:229)
==6793==    by 0x10D8EC: btstack_run_loop_execute (btstack_run_loop.c:189)
==6793==    by 0x10C0E9: main (main.c:230)
==6793==  Block was alloc'd at
==6793==    at 0x4C31B25: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==6793==    by 0x4E44F4E: libusb_alloc_transfer (io.c:1265)
==6793==    by 0x110612: usb_sco_start (hci_transport_h2_libusb.c:855)
==6793==    by 0x111CB3: usb_set_sco_config (hci_transport_h2_libusb.c:1399)
[... 244 lines cut ...]
==6793==    by 0x10D8EC: btstack_run_loop_execute (btstack_run_loop.c:189)
==6793==    by 0x10C0E9: main (main.c:230)
```

After some debugging I got to the problem in hci_transport_h2_libusb.c:
```
    // Handle any packet in the order that they were received
    while (handle_packet) {
        // log_info("handle packet %p, endpoint %x, status %x", handle_packet, handle_packet->endpoint, handle_packet->status);
        void * next = handle_packet->user_data;
        handle_completed_transfer(handle_packet);
        // handle case where libusb_close might be called by hci packet handler        
        if (libusb_state != LIB_USB_TRANSFERS_ALLOCATED) return;

        // Move to next in the list of packets to handle
        if (next) {
            handle_packet = (struct libusb_transfer*)next;
        } else {
            handle_packet = NULL;
        }
    }

```
In this code, handle_completed_transfer sometimes lead to end of call, because it eventually passes the event into packet-handler, which calls 'usb_sco_stop'.As you can see in the valgrind log.
Which means when `handle_packet = (struct libusb_transfer*)next is called, next will point to a memory which is **already freed** in handle_completed_transfer(handle_packet).

mringwal, 2019-05-27:
Hi! Thanks for reporting and even pointing out the actual bug. Your analysis seem right on.

To fix it, I guess the SCO packets / transfers in the 'ready' list need to be remove as well, to avoid the use-after-free. Will have a look at it (a bit tricky with the while loop ready to iterate over them).

skoperst, 2019-05-27:
Yes, I thought about this solution as well, but could not think of a way to overcome the loop losing elements in the list while iterating.

I think storing SCO in a separate list might work, also doing the cleanup in usb_process_ds(aynchronously), but that will not keep the current advantage of 'instant'(synchronous) cleanup which is useful (and possible). Then keep track of the situation in a state machine such as this one:

`typedef enum {
    LIBUSB_SCO_DISABLED = 0,
    LIBUSB_SCO_ENABLE_REQUESTED = 1,
    LIBUSB_SCO_ENABLE_ONGOING = 2,
    LIBUSB_SCO_ENABLED = 3,
    LIBUSB_SCO_DISABLE_REQUESTED = 4,
    LIBUSB_SCO_DISABLE_ONGOING = 5,
} libusb_sco_state_t;`

Alternatively, to do it synchronously(instantly), use the user_data(or a flag array) to store the state of each transfer without putting into 'handle_packet' list at all, and keeping it synchronous at all times. Then its possible to cleanup in one place instantly. However there is slight performance impact, as in such case every time you'll need to search the transfer in the array(though its a very small array).
I think this option is better as instant cleanup is good to have.

mringwal, 2019-06-12:
hi. I've had a look at the libusb transfer handling. The crash you've got results from holding a pointer to the next transfer while calling the btstack packet handler. This can be fixed easily: https://github.com/bluekitchen/btstack/commit/06c6f94c1fff62a521583b46ca0ff4c22b060f8c

Now, we get the SCO closed and usb_sco_stop gets called as a result. It tries to stop all sco in transfers that are active - all completed ones that are in the 'handle_packet' queue are not active. Then it waits for all to complete. So, even if there are a few complete SCO transfers to process, the only issue is that the SCO data is sent to BTstack _after_ the SCO closed. So, that looks fine to me. Do you think I'm missing something here?

Could you try if the issue got fixed with the small update?

skoperst, 2019-08-08:
I tried, it seems to fix the issue.

closed 2019-08-12 by mringwal