bluekitchen/btstack #224: HELP!!! How to use HFP to send audio data from i2s microphone in the esp32 board?

pxxgogo, 2019-06-03:
We found that the hfp_hf_demo in this project only supported to receive data from external devices. But we are wondering how to send audio data to other devices. We tried modifying the  btstack_audio_esp32.c by adding functions about esp32_source. But they are all not executed. Is there anything else we should notice, like changing some macros? Could you give us relevant   examples. We have tried it too many days... And the ddl of our project is this Thursday...

We add the following codes in the btstack_audio_esp32.c.
```
static void (*recording_callback)(int16_t * buffer, uint16_t num_samples);
static void unfill_buffer(void){
    size_t bytes_read;
    uint8_t buffer[DMA_BUFFER_SAMPLES * BYTES_PER_SAMPLE_STEREO];
    i2s_read(i2s_num_in, buffer, DMA_BUFFER_SAMPLES * bytes_per_sample, &bytes_read, portMAX_DELAY);
    (*recording_callback)((int16_t*)buffer, DMA_BUFFER_SAMPLES);
    log_info("unfill buffer");
}
static void driver_timer_handler(btstack_timer_source_t * ts){

    // playback buffer ready to fill
    if (playback_callback){
        // read from i2s event queue
        i2s_event_t i2s_event;
        if( xQueueReceive( i2s_event_queue, &i2s_event, 0) ){
            // fill buffer when DMA buffer was sent            
            if (i2s_event.type == I2S_EVENT_TX_DONE){
                fill_buffer();
            }
        }
    }
    if (recording_callback){
        // read from i2s event queue
        log_info("recording callback");
        i2s_event_t i2s_event;
        if( xQueueReceive( i2s_event_queue, &i2s_event, 0) ){
            // fill buffer when DMA buffer was sent            
            if (i2s_event.type == I2S_EVENT_RX_DONE){
                unfill_buffer();
            }
        }
    }

    // re-set timer
    btstack_run_loop_set_timer(ts, DRIVER_POLL_INTERVAL_MS);
    btstack_run_loop_add_timer(ts);
}
static int btstack_audio_esp32_source_init(
    uint8_t channels,
    uint32_t samplerate, 
    
    void (*recording)(const int16_t * buffer, uint16_t num_samples)
){
    recording_callback = recording;
    num_channels       = channels;
    bytes_per_sample   = channels * 4;  // 16-bit
    i2s_config_t config = 
    {
        .mode                 = I2S_MODE_MASTER | I2S_MODE_RX, 
        .sample_rate          = samplerate,
        .bits_per_sample      = I2S_BITS_PER_SAMPLE_32BIT,
        .channel_format       = I2S_CHANNEL_FMT_ONLY_RIGHT,
        .communication_format = I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB,
        .dma_buf_count        = DMA_BUFFER_COUNT,              // Number of DMA buffers. Max 128.
        .dma_buf_len          = DMA_BUFFER_SAMPLES,            // Size of each DMA buffer in samples. Max 1024.
        .intr_alloc_flags     = ESP_INTR_FLAG_LEVEL1
    };
    i2s_pin_config_t pins = 
    {
        .bck_io_num           = 26,
        .ws_io_num            = 25,
        .data_out_num         = -1,
        .data_in_num          = 4
    };
    log_info("source init?");
    i2s_driver_install(i2s_num_in, &config, DMA_BUFFER_COUNT, &i2s_event_queue);
    i2s_set_pin(i2s_num_in, &pins);
    i2s_zero_dma_buffer(i2s_num_in);

    return 0;
}
//#if 0
static void btstack_audio_esp32_source_start_stream(void){

    if (recording_callback){
        // pre-fill HAL buffers
        log_info("start stream?");
        uint16_t i;
        for (i=0;i<DMA_BUFFER_COUNT;i++){
            unfill_buffer();
        }
    }

    // TODO: setup input

    // start i2s
    i2s_start(i2s_num_in);

    // start timer
    btstack_run_loop_set_timer_handler(&driver_timer, &driver_timer_handler);
    btstack_run_loop_set_timer(&driver_timer, DRIVER_POLL_INTERVAL_MS);
    btstack_run_loop_add_timer(&driver_timer);

    // state
    source_streaming = 1;
}
static void btstack_audio_esp32_source_stop_stream(void){

    if (!source_streaming) return;

    // stop timer
    btstack_run_loop_remove_timer(&driver_timer);

    // stop i2s
     i2s_stop(i2s_num_in);

    // state
    source_streaming = 0;
}

static void btstack_audio_esp32_source_close(void){

    if (source_streaming) {
        btstack_audio_esp32_source_stop_stream();
    }

    // uninstall driver
    i2s_driver_uninstall(i2s_num_in);
}
static const btstack_audio_source_t btstack_audio_source_esp32 = {
    /* int (*init)(..);*/                                       &btstack_audio_esp32_source_init,
    /* void (*start_stream(void));*/                            &btstack_audio_esp32_source_start_stream,
    /* void (*stop_stream)(void)  */                            &btstack_audio_esp32_source_stop_stream,
    /* void (*close)(void); */                                  &btstack_audio_esp32_source_close
};

const btstack_audio_source_t * btstack_audio_esp32_source_get_instance(void){
    return &btstack_audio_source_esp32;
}

```

mringwal, 2019-06-03:
hi. You're lucky: SCO over HCI only works since less than a month. Corrrect: the audio path isn't implemented yet. 

Do you get the sine wave on the remote when SCO_DEMO_MODE  == SCO_DEMO_MODE_SINE in sco_demo_util.c.

On my device, the sine wasn't perfect, and I was waiting for feedback from Espressif.

Please see here: https://github.com/espressif/esp-idf/issues/1118

pxxgogo, 2019-06-03:
THANKS for your reply!!!!! we can receive data if we set the SCO_DEMO_MODE to SCO_DEMO_MODE_SINE. But they seem like constant sounds, not sine. By the way, how can we implement the audio path? where is it? Thanks again!!!

pxxgogo, 2019-06-03:
@mringwal  I add `btstack_audio_source_set_instance(btstack_audio_esp32_source_get_instance());` in the btstack/main.c so that it can register audio_esp32_source in btstack_audio_esp32.c.

pxxgogo, 2019-06-03:
Just now, we succeeded in transferring audio. But package loss is frequent. Is it due to the issue [espressif/esp-idf#1118](https://github.com/espressif/esp-idf/issues/1118)?

mringwal, 2019-06-04:
How frequent? like every 1-2 seconds a glitch sound, or worse?
Yes, in this issue I'm asking when to send packets in order to avoid glitches.

Ideally, there a number of packet buffers and we can fill that then. Then, we get a message for every packet that was sent.

In your setup, you could try to disable SCO-over-HCI and route the SCO data directly to the I2S lines.. In this case, Wide-Band Speech cannot be used. If you try, please let me know if that works - with I2S, there's no problem with flow control as the ESP32 would generate the audio clock.

pxxgogo, 2019-06-04:
Currently, The audio data we get through HFP is noise. We are not sure which part is wrong... I post all modification below.
1. We implemented `btstack_audio_esp32_source` in the file `btstack_audio_esp32.c` 
2. We added `btstack_audio_source_set_instance(btstack_audio_esp32_source_get_instance());` in the file `btstack/main.c` so that it can register audio_esp32_source.

The new btstack_audio_esp32.c is below:
```
/*
 * Copyright (C) 2018 BlueKitchen GmbH
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 *
 * 1. Redistributions of source code must retain the above copyright
 *    notice, this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *    notice, this list of conditions and the following disclaimer in the
 *    documentation and/or other materials provided with the distribution.
 * 3. Neither the name of the copyright holders nor the names of
 *    contributors may be used to endorse or promote products derived
 *    from this software without specific prior written permission.
 * 4. Any redistribution, use, or modification is done solely for
 *    personal benefit and not for any commercial purpose or for
 *    monetary gain.
 *
 * THIS SOFTWARE IS PROVIDED BY BLUEKITCHEN GMBH AND CONTRIBUTORS
 * ``AS IS'' AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 * LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
 * FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL MATTHIAS
 * RINGWALD OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
 * INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING,
 * BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS
 * OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED
 * AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
 * OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF
 * THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF
 * SUCH DAMAGE.
 *
 * Please inquire about commercial licensing options at 
 * contact@bluekitchen-gmbh.com
 *
 */

#define __BTSTACK_FILE__ "btstack_audio_esp32.c"

/*
 *  btstack_audio_esp32.c
 *
 *  Implementation of btstack_audio.h using polling ESP32 I2S driver
 *
 */

#include "btstack_config.h"
#include "btstack_debug.h"
#include "btstack_audio.h"
#include "btstack_run_loop.h"
#include "hal_audio.h"

#include "freertos/FreeRTOS.h"
#include "driver/i2s.h"

#define DRIVER_POLL_INTERVAL_MS          5
#define DMA_BUFFER_COUNT                 2
#define DMA_BUFFER_COUNT_SOURCE                 2
#define DMA_BUFFER_SAMPLES               512
#define DMA_BUFFER_SAMPLES_SOURCE               512
#define BYTES_PER_SAMPLE_STEREO          4

static int num_channels;
static int bytes_per_sample;
[... 245 lines cut ...]
    }

    // uninstall driver
    i2s_driver_uninstall(i2s_num_in);
}

static const btstack_audio_source_t btstack_audio_source_esp32 = {
        /* int (*init)(..);*/                                       &btstack_audio_esp32_source_init,
        /* void (*start_stream(void));*/                            &btstack_audio_esp32_source_start_stream,
        /* void (*stop_stream)(void)  */                            &btstack_audio_esp32_source_stop_stream,
        /* void (*close)(void); */                                  &btstack_audio_esp32_source_close
};

//#endif
const btstack_audio_sink_t *btstack_audio_esp32_sink_get_instance(void) {
    return &btstack_audio_sink_esp32;
}

//#if 0
const btstack_audio_source_t *btstack_audio_esp32_source_get_instance(void) {
    return &btstack_audio_source_esp32;
}
//#endif

```

hiddennasty, 2019-06-04:
@mringwal Hello! I have the same problem with this hfp_hf_demo. The i2s microphone I get only works when the bits_per_sample is set to 32 bits, but the CVSD is a 16-bit format. I'm wondering if the lack of format conversion is the reason why I get the noise instead of the actual sound.

mringwal, 2019-06-04:
hi there. as mentioned, I got stuck in sending a correct sine wave, and didn't go further. The STM32F4 Discovery port implements support for the built-in PDM microphone and is fully working.

After verifying that the sine is sent properly, you can use the audio_duplex example to check if the microphone input (btstack_audio_source) is working well.

@hiddennasty if you get 32 bits, you probably need to manually drop the lower 16 bits before passing the audio samples on to the higher layer.

pxxgogo, 2019-06-04:
Thanks a lot! We tested the Sine wave with the latest esp32 lib, which seems correct. We will test our microphone later as you suggest.

piotrkundu, 2019-10-21:
Actually I'm also looking into sending audio from PDM (or I2S, ESP32 supports both) microphone to an Android or iOS phone in a push to talk setup for voice commands.

mringwal, 2021-07-08:
@pxxgogo did you get it to work? I just re-tested sine wave on esp32 and at least for me it's finally working as expected. The recording path for esp32 isn't fully implemented, but that would be doable now. I guess I'll get the ESP32 audio dev kit now..

cgreening, 2021-07-23:
I have something working base on the code snipped posted above - I need to tidy up my code.

I currently have all the output code commented out and I've removed the I2S event queue code which seemed to be causing me to get gaps in the audio.

cgreening, 2021-07-23:
I've pushed my working code up to here: https://github.com/atomic14/esp32-audio-streaming/tree/main/esp_bluetooth_mic

cgreening, 2021-07-24:
I know have this fully working with microphone and speaker output - https://github.com/atomic14/esp32-hsp-hf

mringwal, 2021-07-24:
Hi @cgreening - thanks for providing your code. I'm getting a 404 when clicking on either of your links. Could you check if the repos are publicly accessible?

cgreening, 2021-07-24:
@mringwal Sorry about that - it's now public.

mringwal, 2021-07-24:
@cgreening thanks! What microphone did you use, resp. on what dev kit does this run? (We ordered a LyraT and a LyraT MSC recently)

cgreening, 2021-07-24:
I'm using a TinyPico - but I can't see why it wouldn't work on a standard ESP32 dev board. I'm using an I2S microphone - ICS-43434 and the Adafruit I2S amplifier - the MAX98537.

I'm pretty sure that any I2S microphone would work (e.g. the INMP441) as it's just using standard I2S setup. Same with the amplifier.

cgreening, 2021-07-24:
I'm using the master branch of the IDF framework - not sure if that makes any difference.

mringwal, 2021-07-26:
Thanks. I think I didn't find an I2S microphone without soldering when I was looking for it. On the LyraT there are 2 mics that are connected via I2S. I think I've also seen some codec chips that need to be configured via i2c. I should be able to have a look at that and then compare to your code.

cgreening, 2021-07-26:
I think the LyraT has a codec chip that needs to be configured as the microphones are connected to that which makes it a bit more complicated to use as you need to configure the codec.

You can get I2S breakout boards that just require header pins soldering on - the INMP441 is pretty popular - you might even be able to find some with pins already attached.

mringwal, 2021-07-26:
Thanks again for the hint. Soldering headers is fine, I just didn't want to design a breakout board first.
I've ordered 3 of those round INMP441 breakout boards on Aliexpress, but might try to configure the codec on the LyraT first (and I would expect to have that code ready to use in the ESP-ADF, which I had issues compiling as I already have the ESP-IDF installed and they don't seem to mix very well - just using the codec code should be fine).

mringwal, 2023-02-21:
The ESP32 port provides an audio driver for both sink and source. Closing this.

closed 2023-02-21 by mringwal