bluekitchen/btstack #225: HID Host Send Output Interrupt to Device

Richard-CRT, 2019-06-03:
Hi,
I'm sorry if this is this the wrong place to ask this.
Does anyone have an example of how the host can send an output interrupt report to a connected device?

I found the source of `hid_device_send_interrupt_message(...)` and tried to use `l2cap_send(...)` and `l2cap_request_can_send_now_event(...)` found there using the `l2cap_hid_interrupt_cid` saved during the connection process. It's possible I have the report wrong, however I'm not getting any sort of indication of success, and I'm wondering if I have the method of sending correct at all.
This is my current function for it:
```
hid_host_send_interrupt_message() {
	if (!l2cap_hid_interrupt_cid) return;

	uint8_t report[] = { 0xA1, 0x03, 0xFF, 0x64, 0x64, 0x64, 0x64, 0xFF, 0x00, 0x05 };
	printf_hexdump(report, 10);
	l2cap_send(l2cap_hid_interrupt_cid, report, 10);
	l2cap_request_can_send_now_event(l2cap_hid_interrupt_cid);
}
```

Thank you to anyone that can confirm/suggest anything!

Richard-CRT, 2019-06-03:
The controller had disabled vibration since the battery was "low" but that code was valid (except the 0xA1 should be 0xA2)

closed 2019-06-03 by Richard-CRT

mringwal, 2019-06-04:
hi. thanks for updating your issue. I think I didn't try output reports with a read device yet, and we also didn't fully test HID Host fully (it basically works though). I'm glad, you've got it working!