bluekitchen/btstack #226: a2dp sink not work on windows 10 1089

PikaJian, 2019-06-04:
windows 10 cannot connect to btstack a2dp sink,
l2cap handle configure response seems failed.

L2CAP signaling handler code 5, state 11
l2cap_stop_rtx for local cid 0x40
l2cap_signaling_handle_configure_response
l2cap cid 64, unknown options
l2cap cid 64, unknown options
l2cap_stop_rtx for local cid 0x40
l2cap_start_rtx for local cid 0x40
L2CAP signaling handler code 5, state 11
l2cap_stop_rtx for local cid 0x40

hci_dump.log is in attatchment

[hci_dump.log](https://github.com/bluekitchen/btstack/files/3250765/hci_dump.log)

mringwal, 2019-06-04:
hi. I didn't see this error yet. Is there anything special about your windows 10? What Bluetooth Controller and Bluetooth Stack does it use? Did you have to install Bluetooth drivers?

BTstack tries to configure ERTM mode and Windows response with "don't understand", although it reported support for ERTM in an earlier message.

[image]

I guess BTstack could try without ERTM in this case, but I like to understand, whats wrong.

PikaJian, 2019-06-18:
I found a bluetooth speaker that win10 can connect, btstack's l2cap configure request's are different to speaker.

### btstack's retransmission and flow control option are always use basic mode (0x0)

[image]

### The speaker use ertm mode (0x3)

[image]

mringwal, 2019-08-19:
hi. we made a few fixes related to ERTM, most notably one where an incoming connection was not accepted  when ERTM was enabled, but not configured.

Could you check with your Windows 10 as well as with that speaker?

mringwal, 2019-10-28:
HI again. We made more changes to ERTM mode and it does not send Retransmission and Flow Control option unless explicitly requested. Please test current develop branch.

closed 2019-10-28 by mringwal