bluekitchen/btstack #227: l2cap_max_le_mtu computation

Lefix2, 2019-06-07:
hi,
I've noticed in l2cap.c that if no custom mtu is given, l2cap_max_mtu is returned, which is HCI_ACL_PAYLOAD_SIZE - L2CAP_HEADER_SIZE.
I was wondering if this value is correct in ble since max MTU is 512!
This result in having MTU > 512 in negotiation (for android stack i've negociated MTUs of 519...)

shouldn't l2cap_max_mtu be MIN(l2cap_max_mtu(),512)?

mringwal, 2019-06-07:
For which protocol do you think the max mtu should be 512? (I recall reading 512 somewhere, but don't remember where)

Lefix2, 2019-06-07:
for gatt protocol, due to characteristic max len of 512?

mringwal, 2019-06-07:
I see. The max attribute value is 512. From that, a GATT Characteristic also only has 512.

For reference: Core 5.1, Vol 3, 3.2.9 "The maximum length of an attribute value shall be 512 octets."

Anyway, this does is still fine for l2cap_max_le_mtu. You could argue that the ATT MTU could be limited to 515 as there's no point in having larger MTUs. It doesn't hurt either - and I've seen people use larger Characteristics (which seems to be supported in iOS + Android ).

Did you have a problem with ATT MTU > 512?

Lefix2, 2019-06-07:
Ok it sound pretty clear!
In fact we already used an MTU > 512 but it wasn't voluntary.
It's very interesting if it's supported natively by IOS and Android.

Yes we encountered a crash on android that bring this question but it's due to a bad "android stack" named FastBLE. The final goal is to use android API directly!

closed 2019-06-07 by Lefix2

mringwal, 2019-06-07:
I didn't personally test longer Characteristics, but I've heard that it worked from someone else. The Bluetooth spec lists this requirement only once. Also, I don't think this isn't documented in the iOS APIs, so, I would try not to rely on it as it might change any time.