bluekitchen/btstack #229: Missing HID Input Reports on ESP32

JSnowden33, 2019-06-09:
Hi,

I'm working with the BTstack develop branch to interface with Joy-Con controllers, which use Bluetooth HID protocol. I've used the "hid_host_demo" to form a connection, and "l2cap_send()" to send output reports since that is not included in the demo. Output reports work fine - I am able to change the LED output on the controller with them. The "3F" input report is pushed from the Joy-Con whenever a button is pressed or released, and I receive those. However, I am not receiving any other input reports. There should be reports in response to every output report, but I don't get those. I should also be receiving regular reports from the Joy-Con after enabling the full reporting mode, which is shown at the end of the log (the output report directly above "DR pressed" enables 60Hz reporting mode). The report formats and expected responses are all documented [here](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/bluetooth_hid_notes.md) and [here](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/bluetooth_hid_subcommands_notes.md).

Is it possible there is some incompatibility between the Joy-Con and the ESP32 that prevents receiving certain data? Perhaps more L2CAP channels should be opened (there's only one for interrupt and one for control currently)?

mringwal, 2019-06-10:
hi. All ACL packets are created equal - if you receive one, you'll receive all of them.
There are 2 channels for HID, Interrupt and Control and the example connects both. 

Do you see the response to the HID Output packets in the HCI log?

JSnowden33, 2019-06-10:
The response packets are not present in any of the logs I'm getting. I can see "LOG -- hci.c.649: hci_send_acl_packet_fragments entered" and so on for outgoing data, but no logs for incoming data other than ACL logs.

I am looking at a USB dump of a Bluetooth transmitter connected to a PC, which allows a Joy-Con to interface with some other open source Joy-Con software. It looks like every output report from this PC program includes padding to make the message 50 bytes long, so I tried doing this as well, but it doesn't change anything. Even when I follow the exact set of output reports use in this example, I should be getting a 50-byte response beginning with "A1 21 ... " in the ACL data, but nothing ever shows up.

USB capture (ACL data is in the second-to-the-right column):
[image]

ESP32 log:
```
HID Connection established
Output report packet: A2 01 00 00 00 00 00 00 00 00 00 03 3F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:04.166] EVT <= 6E 00
[00:00:04.168] EVT <= 78 02 42 00
[00:00:04.171] ACL => 80 00 36 00 32 00 42 00 A2 01 00 00 00 00 00 00 00 00 00 03 3F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:04.189] EVT <= 13 05 01 80 00 01 00
Output report packet: A2 01 01 00 00 00 00 00 00 00 00 30 81 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:04.246] EVT <= 6E 00
[00:00:04.248] EVT <= 78 02 42 00
[00:00:04.251] ACL => 80 00 36 00 32 00 42 00 A2 01 01 00 00 00 00 00 00 00 00 30 81 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:04.269] ACL <= 80 20 11 00 0D 00 42 00 A1 3F 00 00 05 00 80 00 80 00 80 00 80
HID Data: A1 3F 00 00 05 00 80 00 80 00 80 00 80
[00:00:04.281] EVT <= 13 05 01 80 00 01 00
[00:00:04.284] EVT <= 13 05 01 80 00 01 00
[00:00:04.288] ACL <= 80 20 11 00 0D 00 42 00 A1 3F 00 00 08 00 80 00 80 00 80 00 80
HID Data: A1 3F 00 00 08 00 80 00 80 00 80 00 80
Output report packet: A2 01 02 00 00 00 00 00 00 00 00 10 01 60 00 00 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:04.354] EVT <= 6E 00
[00:00:04.356] EVT <= 78 02 42 00
[00:00:04.359] ACL => 80 00 36 00 32 00 42 00 A2 01 02 00 00 00 00 00 00 00 00 10 01 60 00 00 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Output report packet: A2 01 03 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:04.431] EVT <= 6E 00
[00:00:04.433] EVT <= 78 02 42 00
[00:00:04.436] ACL => 80 00 36 00 32 00 42 00 A2 01 03 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:04.454] EVT <= 13 05 01 80 00 01 00
[00:00:04.457] EVT <= 13 05 01 80 00 01 00
[00:00:04.461] EVT <= 6E 00
[00:00:04.464] CMD => 35 0C 05 01 80 00 04 00
[00:00:04.468] EVT <= 6E 00
```

mringwal, 2019-06-11:
Hi. If you like, you can run the code on Windows using the port/windows-winusb to better be able to compare your logs - although you are already comparing ACL packets.

If you like, please post on the BTstack developer mailing list (google group), as I'm probably the only one that reads all issues.

As this is not a bug in BTstack (at least I don't see why), I'm closing this. If BTstack should happen to fail to deliver data to the application, please file a new issue reporting that.

closed 2019-06-11 by mringwal