bluekitchen/btstack #230: ESP32: l2cap_init() crash

hehopmajieh, 2019-06-13:
Hi All, 
i am trying to implement SPP Streamer in large application, as base i use spp_streamer.c example.
According to debug info crash is caused by l2cap_init();
 dump : 
`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400fd627  PS      : 0x00060030  A0      : 0x800e8f88  A1      : 0x3ffb1780  
0x400fd627: btstack_linked_list_add_tail at /home/hehopmajieh/esp/esp-idf/components/btstack/src/btstack_linked_list.c:136

A2      : 0x00000020  A3      : 0x3ffc1398  A4      : 0x3ffb15c0  A5      : 0x3ffae918  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80082388  A9      : 0x3ffb1560  
A10     : 0x3ffb2cc0  A11     : 0x00000004  A12     : 0x00000001  A13     : 0x00000080  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000020  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

ELF file SHA256: 6449cbcbcbd921869021d23df64f1c8ad150f67463b0d44143808de4e5dd0d47

Backtrace: 0x400fd627:0x3ffb1780 0x400e8f85:0x3ffb17a0 0x400e6a6f:0x3ffb17c0 0x400d31a7:0x3ffb17e0 0x400d32bf:0x3ffb1800 0x400d19ac:0x3ffb1870 0x40085755:0x3ffb1890
0x400fd627: btstack_linked_list_add_tail at /home/hehopmajieh/esp/esp-idf/components/btstack/src/btstack_linked_list.c:136

0x400e8f85: hci_add_event_handler at /home/hehopmajieh/esp/esp-idf/components/btstack/src/hci.c:4295

0x400e6a6f: l2cap_init at /home/hehopmajieh/esp/esp-idf/components/btstack/src/l2cap.c:1229

0x400d31a7: btstack_main at /home/hehopmajieh/esp/esp32_ili9341/main/../hw/bt.c:245

0x400d32bf: app_main at /home/hehopmajieh/esp/esp32_ili9341/main/main.c:130

0x400d19ac: main_task at /home/hehopmajieh/esp/esp-idf/components/esp32/cpu_start.c:529

0x40085755: vPortTaskWrapper at /home/hehopmajieh/esp/esp-idf/components/freertos/port.c:403

`

mringwal, 2019-06-13:
hi. can you run spp_streamer.c straight from the repo? I still don't understand what 'LoadProhibitied' really means.

hehopmajieh, 2019-06-13:
Hi, yes i can and it work as expected , i tried to build espressif bt example code and it runs. These seems to be some kind of heap problem /*only when bt is enabled*/, but i am not into esp32 at all.

mringwal, 2019-06-13:
Let's go slowly. If you follow the btstack/esp32/README.md and try example/spp_streamer or example/le_counter, does this work correctly?
(I don't understand what 'espressif bt example' means in the context of BTstack)

hehopmajieh, 2019-06-13:
Hi, le_counter works correctly.

mringwal, 2019-06-13:
ok. I assume spp_counter works as well then.
Now. what did you do to cause l2cap_init() to crash?

hehopmajieh, 2019-06-18:
Hi, sorry fоr the delay.
I have an application that uses litlevgl from esp-iot-solution, and i am trying to integrate spp streamer into it.
Here you can see part of code : [link](https://pastebin.com/fPZuPSpk) and [sdkconfig](https://pastebin.com/qMsDE35V)

~Best

mringwal, 2019-06-18:
please compare your app_main() to the one from components/btstack/main.c

In your app_main, BTstack's ESP32 init code is missing.

closed 2019-06-18 by mringwal

emcniece, 2021-01-16:
Does anyone have a link to `components/btstack/main.c`? I can't seem to find it in my `~/esp` directories. I'm working through the same issue, but I don't know what exactly is missing from my `app_main`.

mringwal, 2021-01-17:
Hi Eric

The main.c in the component was a problem for non-BTstack projects. It was moved into the generated projects. Please start over with ./integrate_btstack.py script in port/esp42