bluekitchen/btstack #231: How could know the sink disconnected status  immediately?

mailtoxiong, 2019-06-18:
@mringwal  
Dear Friend: 
                 Thanks for your previous helps ,thanks a lot .
                  One more question want to trouble you :
                  Normmally ,I tested ---- if i turn off the power of sink ,in A2DP mode ,the AG host released the connection ,after 18-22 seconds .
                 
                   Could I get the disconnection status   immediately?  Which API it works for this ?

Regards !
-----------------------------
Your Sincerely
Kerwin

mringwal, 2019-06-20:
If you turn off one device, there's a 'link supervision timeout'. There's no event about the disconnect itself.

If you want faster feedback, you can send hci_write_link_supervision_timeout for the HCI Connection of A2DP.

closed 2019-06-20 by mringwal

mailtoxiong, 2019-06-21:
@mringwal
if i turn off the power of sink ,in A2DP mode ,the AG host released the connection ,after 18-22 seconds .
----------------------------------------------------------------------------------------------------------------
Now :  the Release time is  almost 20 seconds , then printf the log [ A2DP Source: Signaling released.]

Which Parameters or Macro could be set ,can short the  release time ?

mringwal, 2019-06-21:
You could try to send hci_write_link_supervision_timeout to e.g. 5 seconds (that would be the value 8). You get the handle from the connected event