bluekitchen/btstack #237: [SUGGESTION] remove gap_connectable_control() from l2cap

skoperst, 2019-07-15:
https://github.com/bluekitchen/btstack/blob/dfa951b111c407e76996c3d7fcde927f8cf3c078/src/l2cap.c#L3537

I think it does not belong there, as one who turn off paging would not expect to have it changed by registering l2cap services.
I also could not find any direct relation between those in the Core spec.

mringwal, 2019-07-15:
It was added there as a convenience. It was there as a convenience: if you provide a service another device should be able to connect to us. However, you could have an L2CAP service on the 'master' device that creates the outgoing connection.

If it gets removed, the app code would need to call it explicitly, which could be confusing for existing code.

Side note: I don't expect adding/removing l2cap services at run time. do you have an example why you'd need this?

skoperst, 2019-07-18:
It will cause the app code to use the 'gap_connectable_control' explicitly, but I think that if an advanced user wants to do something different (within the standards) then he need to have a way, and controlling the gap_connectable_control, is not working as expected because of it.

About the side note, some devices exploit L2CAP services for their benefit such as some car headunits will constantly check upon SDP(I'm not sure why), when every connection on BT controller counts, and QOS is important these constant connections are less than useful. Maybe there is another way to deal with it, I'm not sure how. Declining the connection request will not cause the headunits to stop, however unregistering the l2cap channel will.

mringwal, 2021-07-02:
Hi! We've added ENABLE_EXPLICIT_CONNECTABLE_MODE_CONTROL 
If it is enabled, l2cap does not call gap_connectable_control.

closed 2021-07-02 by mringwal