bluekitchen/btstack #239: A problem with pan_lwip_http_server

AlexiLlaiho, 2019-07-25:
Good day! I'm disapointing, becase can't make a pan_lwip example.. I rad a btstack book, but I don't understand how to fix this problem:
/home/msi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/msi/esp/btstack/port/esp32/example/pan_lwip_http_server/build/main/libmain.a(pan_lwip_http_server.o):(.literal.network_setup+0x8): undefined reference to `httpd_init'
/home/msi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/msi/esp/btstack/port/esp32/example/pan_lwip_http_server/build/main/libmain.a(pan_lwip_http_server.o): in function `network_setup':
/home/msi/esp/btstack/port/esp32/example/pan_lwip_http_server/main/pan_lwip_http_server.c:246: undefined reference to `httpd_init'
collect2: error: ld returned 1 exit status

Please, explain me where I don't right.
Best regards, Alex.

AlexiLlaiho, 2019-07-25:
i won this problem by adding some files in example's folder.
[image]

closed 2019-07-25 by AlexiLlaiho

mringwal, 2019-07-25:
Glad you got it working. 

What platform do you use? There were ESP32 in the error and now you have a CCS snapshot, do anything MSP430/432?