bluekitchen/btstack #240: posix: gettimeofday is affected by discontinuous jumps in the system time

skoperst, 2019-08-08:
The use of gettimeofday here might cause very strange timeouts when NTP is used. For example 'L2CAP_RTX_TIMEOUT_MS' can fire when NTP jumps forward and deny a connection.

Maybe worth using the monolithic clock and switch to timespec?
clock_gettime(CLOCK_MONOTONIC_RAW, &ts);

```
static uint32_t btstack_run_loop_posix_get_time_ms(void){
    struct timeval tv;
    gettimeofday(&tv, NULL);
    uint32_t time_ms = (uint32_t)((tv.tv_sec  - init_tv.tv_sec) * 1000) + (tv.tv_usec / 1000);
    log_debug("btstack_run_loop_posix_get_time_ms: %u <- %u / %u", time_ms, (int) tv.tv_sec, (int) tv.tv_usec);
    return time_ms;
}
```

mringwal, 2019-08-08:
Good idea. Looks like there are defines that allow to only use it if available.
(btw. your clock should not jump by 5 seconds when using NTP... :)

skoperst, 2019-08-09:
True, in normal conditions NTP will not have such big effect. But when your initial clock is off(ex. system was shutdown and it has no clock battery), or user changes time, day light saving etc.. this can effect.

Don't get me wrong, this is very minor overall and rare condition. However its a big advantage to have 100% reliability vs 99.9%. (with 100k users, suddenly 100 people are annoyed every day and write bad reviews. start forum topics about it etc..)

skoperst, 2019-08-09:
I made a change to timespec, however it won't compile because of the CFLAG -std=c99.

What is the reason behind this requirement?

mringwal, 2019-08-12:
yes, posix is a requirement, so posix checks for monotonic clock are fine.
I don't want to use gnu99 as it seems like an additional dependency on GCC (or CLANG which supports it as well)

I've updated the code based on your pull request in 92b78dd714c7b90c9f3ef3cfc48966a44f6ef47e

closed 2019-08-12 by mringwal