bluekitchen/btstack #242: reconnection and idf version

jschoch, 2019-08-18:
I'm working on an esp32 alexa gadget here: https://github.com/jschoch/esp32-gadget

Using idf 3.2.2 I can pair but if I reboot the device it does not reconnect.  I have to unpair and repair to get it to connect again.  Using the pre release idf everything works correctly.  How might I troubleshoot this?

Here is a diff of the output:  https://gist.github.com/jschoch/e751d9cab644a43536cc6da4d688a018

I don't see any big differences other than the error code.  is there a configuration variable that might cause this?

```
< [00:00:44.336] LOG -- hci.c.2100: Connection_complete (status=34) 74:75:48:D5:32:E3

< [00:00:44.337] LOG -- hci.c.1829: Outgoing connection to 74:75:48:D5:32:E3 failed

< [00:00:44.340] LOG -- l2cap.c.905: L2CAP_EVENT_CHANNEL_OPENED status 0x22 addr 74:75:48:D5:32:E3 handle 0xffff psm 0x1 local_cid 0x40 remote_cid 0x0 local_mtu 1691, remote_mtu 672, flush_timeout 0

< [00:00:44.357] LOG -- sdp_client.c.448: SDP Client Connection failed, status 0x22.

< I (44364) main.cpp: SDP query failed 0x22

< [00:00:44.369] LOG -- l2cap.c.1100: l2cap_stop_rtx for local cid 0x40
```

mringwal, 2019-08-19:
Hi. Please enable HCI Dump in components/btstack/main.c - main() and make two new traces with 3.2.2 and master. Then convert both using btstack/tool/create_packetlog..py and post them here.

The version of esp-idf should not have any influence on BTstack's behaviour. I think the NVS implementation changed recently, but that would only explain problems reusing link keys stored with 3.2.2 on master. 

The error indicates that the physical connection could not be created, e.g. device not present, BD_ADDR not correct.

jschoch, 2019-08-31:
Here are the captures.  Nothing jumps out as to why it can't reconnect.

[capture.zip](https://github.com/bluekitchen/btstack/files/3562873/capture.zip)

mringwal, 2019-08-31:
Log b.pklg shows a single connection attempt. Could you make a log with both successful pairing and failing reconnect?

Interesting: the error is "LMP Response Timeout". If the other device is just not there, you'd get a "Page Timeout". In the a.pklg, there's a role switch before the connection is complete. Maybe this role switch is broken in 3.2.2 and was fixed later?

I found this commit: https://github.com/espressif/esp32-bt-lib/commit/d915fc93499fd925ae301306a7539b5f13bf6e65

If you like, you can use git blame to find the change in esp-idf that causes the difference.

mringwal, 2021-07-02:
Closing - no further info from submitter.

closed 2021-07-02 by mringwal