bluekitchen/btstack #243: Build fails when using headers in C++ sources

dsa-t, 2019-08-23:
I use btstack headers in С++ source code files with ESP32. 
After upgrading btstack to 28681d35be16ab9a30fe2d2b51dca13b7442dca0, the build fails to compile with messages looking like this:
```
~/esp/esp-idf/components/btstack/src/mesh/mesh_access.h:257:89: error: expected ',' or '...' before 'template'
 mesh_network_pdu_t * mesh_access_setup_unsegmented_message(const mesh_access_message_t *template, ...);
                                                                                         ^
~/esp/esp-idf/components/btstack/src/mesh/mesh_access.h:258:89: error: expected ',' or '...' before 'template'
 mesh_transport_pdu_t * mesh_access_setup_segmented_message(const mesh_access_message_t *template, ...);
                                                                                         ^
```

`template` is a keyword in C++, so the argument name should be changed.

mringwal, 2019-08-23:
> After upgrading btstack to 28681d3, the build fails to compile with messages looking like this:

thanks for reporting/catching this. I see: 'template' is not a good name here.
I'll fix it later, probably to 'message_template"

Cheers
 Matthias

mringwal, 2019-08-23:
Fixed in 39317c775

closed 2019-08-23 by mringwal