bluekitchen/btstack #245: How do you advertise a UUID128 SDP records as RFCOMM server/master?

piotrkundu, 2019-08-30:
I'm trying to figure out how to advertise UUID128 SDP record on ESP32, where the ESP32 is RFCOMM server (master) and an Android client (slave) connects to the service using createInsecureRfcommSocketToServiceRecord https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#createInsecureRfcommSocketToServiceRecord(java.util.UUID).

I've looked at spp_create_sdp_record() https://github.com/bluekitchen/btstack/blob/789c3c6f9c87f37dfab2df5d9ed2c74951d065fd/src/classic/spp_server.c#L56
where 16-bit UUID is used for SPP:
`de_add_number(attribute,  DE_UUID, DE_SIZE_16, BLUETOOTH_SERVICE_CLASS_SERIAL_PORT );`
I would like to add a custom service using uuid128. How do I do that?

piotrkundu, 2019-08-30:
Does this do the trick?
https://github.com/bluekitchen/btstack/blob/2f98d3c60ab84a4e7ec4a33c528cd323900cd4a3/test/pts/classic_test.c#L707

```
    // 0x0001 "Service Class ID List"
    de_add_number(service,  DE_UINT, DE_SIZE_16, BLUETOOTH_ATTRIBUTE_SERVICE_CLASS_ID_LIST);
    attribute = de_push_sequence(service);
    {
        de_add_uuid128(attribute, &dummy_uuid128[0] );
    }
    de_pop_sequence(service, attribute);
```

mringwal, 2019-08-30:
HI. I think you can just provide a standard SPP service and connect to it from Android.

However, good point about the ability to specify a custom UUID in general, like on Android.
You can either replace or add the de_add_number call with the de_add_uuid128 call to add your custom UUID.

piotrkundu, 2019-09-05:
That's the thing, I can't use standard SPP unfortunately. The other end expects a custom UUID. If you have a suggestion on how to add this functionality I can do a PR.
I see either extending the function **sdp_create_spp_service** with uuid128 parameter or create a new one **sdp_create_spp_custom_service**. The first option is breaking change, you'll have an optional param most people won't use (except me). The second option you need to copy all of the code from **sdp_create_spp_service** just to add the **de_add_uuid128**.

Also if you have some unit tests for this I can do them in the PR.

mringwal, 2019-09-06:
sdp_create_spp_custom_service is fine. I'll have a look about how to share code. I guess a save one would be to introduce an internal function that supports both UUIDs types and have the existing and the new custom one call it.

mringwal, 2019-09-06:
Hi Piotr. Check out develop branch - commit ec818a285cac0b16a0d9d9cbd32a15fe2cb1b33c and let me know if it works for you (I guess it's fine). Cheers.

closed 2019-09-06 by mringwal

piotrkundu, 2019-09-06:
Thanks, that's awesome. I'll test next week and let you know.

piotrkundu, 2019-09-06:
Oh, and a question. Is it possible to run a SPP server with sdp_create_spp_service and another with sdp_create_spp_custom_service, so that you can have multiple services running?
I think I've bumped into some comments about this being a limitation.

mringwal, 2019-09-07:
Sure. You need to use a different server channel [1..31] for each RFCOMM server and a different service record handle for each SDP record. If you have static memory, make sure that you have enough RFCOMM channels + services configured as well as one L2CAP channel per HCI connection.

piotrkundu, 2019-09-07:
Got it thanks. Just verified that the custom UUID128 works just fine.