bluekitchen/btstack #248: cc256x A2DP assisted mode and over hci

jesusVqz, 2019-09-06:
Hello everyone , I was working on port this library to atsamr28XXX board based , and it works but when im using the a2dp source example  the audio buffering seems underwrite , I heard the testing audio , cutted , it's like there is no enough information to sound , first I head a audio peace , then stop and then the next audio part , also I check with sine wave , and happens the same the sine hear well , but is stopped repeatly. 

Also and other project, I was looking for a way to use the assited a2dp mode and reduce the ram consumption releated to  the sco buffers and audio generation , I use the a2dp example and change some steep to send the vendor commands to activate the assited mode , all commands are replied successful but the module (CC256X) doesn't generate the I2S clock  , my hardware is fine because im using a  dev board from texas and with bluetopia works well , but I want use your library because is more flexible , Do you have any idea or some code to validate at least the  sco over hci ? thanks !

mringwal, 2019-09-09:
Hi Jesus. You can post on the developer mailing list to reach the community here: http://groups.google.com/group/btstack-dev

The A2DP Source should just work as it is - if it is not slowed down by your port. 
I suggest to compare to one of our example ports, e.g. the ESP32 or the STM32F4 Discovery and compare to that.

If it's not continous, please check:
- do you use a Bluetooth UART baudrate that's high enough? e.g. 921600/1 mbps is fine for A2DP. You cannot send audio at 115200
- is HCI packet logging disabled? if you're logging over a console UART, this might slow it down. (Binary logging via Segger RTT is possible for this)

--

We never tried Assisted A2DP Mode as it disables BLE. Good point about reduced RAM when using it.
A2DP does not use SCO connections, but encodes audio using SBC codec. 

For Assisted A2DP Mode, you need to combine the main init script with the AVRR add-on. See http://processors.wiki.ti.com/index.php/CC256x_Downloads
Did you use these? Our convert_bts_init_scripts.py should be able to combine that. Have a look at chipset/cc256x/Makefile.inc on how to use it.

For regular A2DP, SBC frames are (or other codecs) are send over HCI. SCO over HCI is only needed/used for HFP, especially HFP Wide-Band-Speech.

B-Nagendra, 2024-01-08:
hi,  I am new to btstack, and need to test A2dp on  my ARM32 platform, kindly help me how to start with BTStack ?

mringwal, 2024-01-08:
No further response.

closed 2024-01-08 by mringwal