bluekitchen/btstack #250: SSP on Cypress CYW20719

romainmoura, 2019-09-09:
Hello,
I'm trying to use SSP capability of btstack with a CYW20719 module from Cypress.
Everything works well (init, pairing ...etc...) except when I want to open the virtual serial port with PuTTY, it freezes for a few second and finally can't open the serial port.
On btstack side it seems that a link key is requested when the serial port is open but no sure if it is the problem.

Any suggestion or feedback is welcome!
Thanks for your help.

Best regards

mringwal, 2019-09-09:
Please post HCI packet log.

mringwal, 2019-09-09:
btw, how did you configure the 20719 - I did post here https://community.cypress.com/message/205568 but did not follow up with the latest suggestions...

romainmoura, 2019-09-09:
I loaded an empty application on the 20719 and use an external µC to communicate with.
Here is the log : (the log could be truncated at some point due to overflow on embedded debuger) (I tried to open the serial port @00:00:53.300)

[00:00:00.000] LOG -- rfcomm.c.2443: RFCOMM_REGISTER_SERVICE channel #1 mtu 65535 flow_control 0 credits 10
[00:00:00.000] LOG -- l2cap.c.3524: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535
[00:00:00.000] LOG -- l2cap.c.3524: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
SDP service record size: 95
[00:00:00.000] LOG -- hci.c.3042: hci_power_control: 1, current mode 0
[00:00:00.000] LOG -- hci.c.4179: BTSTACK_EVENT_STATE 1
[00:00:00.000] EVT <= 60 01 01 
[00:00:00.000] CMD => 03 0C 00 
[00:00:00.000] EVT <= 6E 00 
[00:00:00.000] EVT <= 0E 04 01 03 0C 00 
[00:00:00.000] CMD => 01 10 00 
[00:00:00.000] EVT <= 6E 00 
[00:00:00.000] LOG -- hci_transport_h4.c.278: Detected Cypress controller
[00:00:00.000] EVT <= 0E 0C 01 01 10 00 09 00 10 09 31 01 0E 42 
[00:00:00.000] LOG -- hci.c.1985: Treat Cypress as Broadcom
[00:00:00.000] LOG -- hci.c.1992: Manufacturer: 0x000f
[00:00:00.000] CMD => 14 0C 00 
[00:00:00.000] EVT <= 6E 00 
[00:00:00.100] LOG -- hci.c.1977: BR/EDR support 1, LE support 0
[00:00:00.100] CMD => 01 0C 08 FF FF FF FF FF FF FF 1F 
[00:00:00.100] EVT <= 6E 00 
[00:00:00.100] EVT <= 0E 04 01 01 0C 00 
[00:00:00.100] CMD => 56 0C 01 01 
[00:00:00.100] EVT <= 6E 00 
20 32 30 3A 37 31 3A 39 42 3A 31 39 3A 33 45 3A 34 31 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[00:00:00.200] EVT <= 6E 00 
[00:00:00.200] EVT <= 0E 04 01 13 0C 00 
[00:00:00.300] EVT <= 0E 04 01 1A 0C 00 
[00:00:00.300] LOG -- hci.c.4303: BTSTACK_EVENT_DISCOVERABLE_ENABLED 1
[00:00:00.300] EVT <= 66 01 01 
[00:00:00.300] LOG -- hci.c.1522: hci_init_done -> HCI_STATE_WORKING
[00:00:00.300] LOG -- hci.c.4179: BTSTACK_EVENT_STATE 2
[00:00:00.300] EVT <= 60 01 02 
0B 00 00 
[00:00:16.100] LOG -- l2cap.c.2236: l2cap - security level update for handle 0x000b
[00:00:16.100] LOG -- l2cap.c.2246: channel 1fff2da4, cid 0040 - state 5: actual 0 >= required 0?
[00:00:16.100] LOG -- l2cap.c.942: L2CAP_EVENT_INCOMING_CONNECTION addr 00:19:86:00:2E:02 handle 0xb psm 0x1 local_cid 0x40 remote_cid 0x40
[00:00:16.100] EVT <= 72 0E 02 2E 00 86 19 00 0B 00 01 00 40 00 40 00 
[00:00:16.100] LOG -- l2cap.c.2357: L2CAP_ACCEPT_CONNECTION local_cid 0x40
[00:00:16.100] ACL => 0B 00 10 00 0C 00 01 00 03 10 08 00 40 00 40 00 00 00 00 00 
[00:00:16.100] EVT <= 0F 04 00 01 1B 04 
[00:00:16.100] EVT <= 6E 00 
[00:00:16.100] ACL => 0B 00 10 00 0C 00 01 00 04 01 08 00 40 00 00 00 01 02 F9 03 
[00:00:16.100] LOG -- l2cap.c.1104: l2cap_stop_rtx for local cid 0x40
[00:00:16.100] LOG -- l2cap.c.1113: l2cap_start_rtx for local cid 0x40
[00:00:16.100] EVT <= 6E 00 
[00:00:16.100] EVT <= 0B 0B 00 0B 00 BF FE CF FE DB FF 7B 87 
[00:00:16.100] LOG - 0:16.200] LOG -- hci.c.3602: IO Capability Request received, stack bondable 1, io cap 1
[00:00:16.200] LOG -- hci.c.4387: gap_mitm_protection_required_for_security_level 0
[00:00:16.200] CMD => 2B 04 09 02 2E 00 86 19 00 01 00 00 
[00:00:16.200] EVT <= 6E 00 
[00:00:16.200] EVT <= 0E 0A 01 2B 04 00 02 2E 00 86 19 00 
[00:00:16.300] EVT <= 13 05 01 0B 00 01 00 
[00:00:16.400] EVT <= 33 0A 02 2E 00 86 19 00 FA 35 06 00 
SSP User Confirmation Request with numeric value '407034'
SSP User Confirmation Auto accept
[00:00:16.400] CMD => 2C 04 06 02 2E 00 86 19 00 
[00:00:16.400] EVT <= 6E 00 
[00:00:16.400] EVT <= 0E 0A 01 2C 04 00 02 2E 00 86 19 00 
279: hci_emit_security_level 3 for handle b
[00:00:18.500] EVT <= E0 03 0B 00 03 
[00:00:18.500] LOG -- l2cap.c.2236: l2cap - security level update for handle 0x000b
[00:00:18.500] LOG -- l2cap.c.2246: channel 1fff2da4, cid 0040 - state 12: actual 3 >= required 0?
[00:00:18.500] ACL <= 0B 20 18 00 14 00 40 00 06 00 00 00 0F 35 03 19 01 00 03 F1 35 05 0A 00 00 FF FF 00 
[00:00:18.500] EVT <= 78 02 40 00 
[00:00:18.500] ACL => 0B 00 6E 00 6A 00 40 00 07 00 00 00 65 00 62 36 00 5F 36 00 5C 09 00 00 0A 00 01 00 03 09 00 01 36 00 03 19 11 01 09 00 04 36 00 0E 36 00 03 19 01 00 36 00 05 19 00 03 08 01 09 00 05 36 00 03 19 10 02 09 00 06 36 00 09 09 65 6E 09 00 6A 09 01 00 09 00 09 36 00 09 36 00 06 19 11 01 09 11 02 09 01 00 25 0B 53 50 50 20 43 6F 75 6E 74 65 72 00 
[00:00:18.600] EVT <= 6E 00 
[... 19 lines cut ...]
[00:00:53.300] EVT <= 0F 04 00 01 09 04 
[00:00:53.300] EVT <= 03 0B 00 0B 00 02 2E 00 86 19 00 01 00 
[00:00:53.300] LOG -- hci.c.2133: Connection_complete (status=0) 00:19:86:00:2E:02
[00:00:53.30n outgoing buffer, only using MTU = 1017
[00:00:53.400] ACL => 0B 00 12 00 0E 00 01 00 05 14 0A 00 40 00 00 00 00 00 01 02 F9 03 
[00:00:53.400] EVT <= 13 05 01 0B 00 02 00 
[00:00:53.400] EVT <= 6E 00 
[00:00:53.400] ACL <= 0B 20 0E 00 0A 00 01 00 05 02 06 00 41 00 00 00 00 00 
[00:00:53.400] LOG -- l2cap.c.2562: L2CAP signaling handler code 5, state 11
[... cut ...] 01 00 25 0B 53 50 50 20 43 6F 75 6E 74 65 72 00 
[00:00:53.500] EVT <= 6E 00 
[00:00:53.500] EVT <= 13 05 01 0B 00 02 00 
[00:00:53.500] EVT <= 17 06 02 2E 00 86 19 00 
[00:00:53.500] LOG -- hci.c.2206: HCI_EVENT_LINK_KEY_REQUEST
[00:00:53.600] EVT <= 13 05 01 0B 00 02 00 
[00:01:24.100] EVT <= 05 04 00 0B 00 22 
[00:01:24.100] LOG -- hci.c.929: Connection closed: handle 0xb, 00:19:86:00:2E:02
[00:01:24.100] LOG -- hci.c.4248: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
[00:01:24.100] EVT <= 61 01 00

mringwal, 2019-09-09:
hard to say. the log is incomplete. It looks like the pairing goes well, but there's a disconnect soon after that.

Could you make your printf blocking, such that you're not loosing any debug output - no worries, the Bluetooth Controller can wait. 

After that, please process the textual log with tool/create_packet_log.py and send this one (as .zip as GitHub doesn't like .pklg files). I can then have another look.

romainmoura, 2019-09-10:
Hello,
Here is the log file.
[logtest.zip](https://github.com/bluekitchen/btstack/files/3594510/logtest.zip)

romainmoura, 2019-09-10:
OK, my bad, I don't implement any link key storage ! Adding the "btstack_link_key_db_memory" solve the problem.
Thanks for your help

mringwal, 2019-09-10:
Good to hear. Just remember to use Flash as backup store for link keys at the end. 
See http://bluekitchen-gmbh.com/storing-link-keys-in-flash-memory/

I still have to find some time to play with the 20719 (the 20819 and 20704 worked fine).

closed 2019-09-10 by mringwal