bluekitchen/btstack #251: Cmake and KConfig support for ESP32

piotrkundu, 2019-09-11:
Since ESP-IDF have moved towards CMake, it might be useful to add that to the ESP32 port.

Attached an example of *CMakeLists.txt* I've put in *esp-idf/components/btstack/*, but it lacks Kconfig/define flags, so when you build projects that are not for btstack you will get an error:

```esp-idf/components/btstack/main.c:385: undefined reference to `btstack_main'```

[DOWNLOAD: CMakeLists.txt ](https://github.com/bluekitchen/btstack/files/3600017/CMakeLists.txt)

mringwal, 2019-09-11:
Hi Piotr. Please have a look at the esp32-cmake branch. There's also already an open issue #196 
Thanks for the CMakeLists.txt

Do you have an idea how to automatically call the tool/compile__gatt.py from CMake? (that's the thing that prevented me from adding the CMakeLists to main branches).

piotrkundu, 2019-09-11:
I've run custom commands from CMake, but that's something I set up once every few years, so the exact knowledge is lost, but here are some hints:

execute_process()
add_custom_command()

mringwal, 2019-09-11:
sorry, forgot to mention. in port/libusb, there's already a working CMakeLIsts.txt that uses the gatt compiler, but I couldn't get it to work within the esp32 build system.

piotrkundu, 2019-09-11:
@mringwal It didn't work with port/libusb either, at least not on OSX. So I touched it up a bit.
https://github.com/bluekitchen/btstack/pull/252

A tip is to inpect if the output file was added and it wasn't (a missing space was the culprit)
`message("example SOURCE_FILES: ${SOURCE_FILES}")`

I'll give esp32 a try as well later on.

howroyd, 2019-10-08:
Not sure what I am doing wrong and if this helps?

On Win10, `esp-idf` fully setup on v4 release.  Been working fine for some months.  Cloned the BTStack repo and switched to the `esp32-cmake` branch.  Ran the python script `integrate_examples.py`.  All OK.

Then trying to build one of the examples (audio_duplex):
```
C:\Users\howro\Documents\Code\temp\audio_duplex>C:\Users\howro\.platformio\python27\python.exe C:\Users\howro\Documents\Code\esp-idf\tools\idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\howro\documents\code\temp\audio_duplex\build
Executing "ninja all"...
[1/350] Performing build step for 'bootloader'
ninja: no work to do.
[2/350] Building C object esp-idf/btstack/CMakeFiles/__idf_btstack.dir/src/btstack_crypto.c.obj
FAILED: esp-idf/btstack/CMakeFiles/__idf_btstack.dir/src/btstack_crypto.c.obj
C:\Users\howro\.espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/3rd-party/bluedroid/decoder/include -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/3rd-party/bluedroid/encoder/include -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/3rd-party/hxcmod-player -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/3rd-party/hxcmod-player/mods -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/3rd-party/md5 -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/3rd-party/yxml -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/src/classic -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/src/ble/gatt-service -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/src/ble -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/src -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/platform/embedded -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/platform/freertos -IC:/Users/howro/Documents/Code/esp-idf/components/btstack/include -IC:/Users/howro/Documents/Code/esp-idf/components/newlib/platform_include -IC:/Users/howro/Documents/Code/esp-idf/components/newlib/include -IC:/Users/howro/Documents/Code/esp-idf/components/freertos/include -IC:/Users/howro/Documents/Code/esp-idf/components/heap/include -IC:/Users/howro/Documents/Code/esp-idf/components/log/include -IC:/Users/howro/Documents/Code/esp-idf/components/soc/esp32/include -IC:/Users/howro/Documents/Code/esp-idf/components/soc/include -IC:/Users/howro/Documents/Code/esp-idf/components/esp_rom/include -IC:/Users/howro/Documents/Code/esp-idf/components/esp_common/include -IC:/Users/howro/Documents/Code/esp-idf/components/xtensa/include -IC:/Users/howro/Documents/Code/esp-idf/components/xtensa/esp32/include -IC:/Users/howro/Documents/Code/esp-idf/components/esp32/include -IC:/Users/howro/Documents/Code/esp-idf/components/driver/include -IC:/Users/howro/Documents/Code/esp-idf/components/esp_ringbuf/include -IC:/Users/howro/Documents/Code/esp-idf/components/esp_event/include -IC:/Users/howro/Documents/Code/esp-idf/components/tcpip_adapter/include -IC:/Users/howro/Documents/Code/esp-idf/components/lwip/include/apps -IC:/Users/howro/Documents/Code/esp-idf/components/lwip/include/apps/sntp -IC:/Users/howro/Documents/Code/esp-idf/components/lwip/lwip/src/include -IC:/Users/howro/Documents/Code/esp-idf/components/lwip/port/esp32/include -IC:/Users/howro/Documents/Code/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/howro/Documents/Code/esp-idf/components/vfs/include -IC:/Users/howro/Documents/Code/esp-idf/components/esp_wifi/include -IC:/Users/howro/Documents/Code/esp-idf/components/esp_wifi/esp32/include -IC:/Users/howro/Documents/Code/esp-idf/components/esp_eth/include -IC:/Users/howro/Documents/Code/esp-idf/components/efuse/include -IC:/Users/howro/Documents/Code/esp-idf/components/efuse/esp32/include -IC:/Users/howro/Documents/Code/esp-idf/components/app_trace/include -IC:/Users/howro/Documents/Code/esp-idf/components/nvs_flash/include -IC:/Users/howro/Documents/Code/esp-idf/components/spi_flash/include -IC:/Users/howro/Documents/Code/esp-idf/components/mbedtls/port/include -IC:/Users/howro/Documents/Code/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/howro/Documents/Code/esp-idf/components/bt/include -IC:/Users/howro/Documents/Code/esp-idf/components/bt/common/osi/include -IC:/Users/howro/Documents/Code/esp-idf/components/bt/host/bluedroid/api/include/api -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.1-dev-369-g4dac7c7df\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/btstack/CMakeFiles/__idf_btstack.dir/src/btstack_crypto.c.obj -MF esp-idf\btstack\CMakeFiles\__idf_btstack.dir\src\btstack_crypto.c.obj.d -o esp-idf/btstack/CMakeFiles/__idf_btstack.dir/src/btstack_crypto.c.obj   -c C:/Users/howro/Documents/Code/esp-idf/components/btstack/src/btstack_crypto.c
C:/Users/howro/Documents/Code/esp-idf/components/btstack/src/btstack_crypto.c:64:18: fatal error: uECC.h: No such file or directory
compilation terminated.
[15/350] Building C object esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cbortojson.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

matheusmbgeng, 2020-12-14:
Hello guys,

any hint on this?

> Since ESP-IDF have moved towards CMake, it might be useful to add that to the ESP32 port.
>
> Attached an example of _CMakeLists.txt_ I've put in _esp-idf/components/btstack/_, but it lacks [...]

I get this error on projects that i dont use btstack. Probably something related to btstack overwrite project app_main, i dont know. Any hint about how let btstack installed on esp-idf components and also do not screw up with non related btstack projects?

mringwal, 2020-12-14:
Please update esp-idf and btstack to current master. We've changed the integration a while ago to not provide app_main. Instead, a minimal main.c with app_main.c is copied into the generated projects.

I can compile non-BTstack projects (.e.g hello-world) using Make. It just failed for CMake, but probably due to a modified include path.

mringwal, 2021-07-08:
Just tried compiling with make,  cmake+make, make+ninja on btstack develop branch.

closed 2021-07-08 by mringwal