bluekitchen/btstack #253: esp32 how to change DISCOVERABLE IAC from GIAC to LIAC

rayPopo, 2019-09-28:
at esp-idf i can use esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_LIMITED_DISCOVERABLE); to change GIAC to LIAC
at btstack i find hci_send_cmd(&hci_write_current_iac_lap_two_iacs, 2, GAP_IAC_GENERAL_INQUIRY, GAP_IAC_LIMITED_INQUIRY); at test/pts/hid_device_test.c
but not work.

please help!
thank you!

mringwal, 2019-09-28:
hi. why do you want to set this scan mode?
before sending a hci cmd directly, you need to check if you can send.
Try similar to this: https://github.com/bluekitchen/btstack/blob/master/example/dut_mode_classic.c#L89
(set a flag on stack working, check if you can send, then send)

closed 2019-09-28 by mringwal

rayPopo, 2019-09-29:
cool! thank you answer!
I want to build a wiimote use microchip to play taiko no tatsujin wii version by my friend diy custom drum.
my friend diy a mod welding a lot of wird to wiimote board to controll wii.
it work good,but mod wiimote board too hard to me.
so i think emulator wiimote by microship maybe easy to me. 
i had try buy few microchip like arduino with H10 and nrf8001, arduino 101, leonardo with usb host shield and bluetooth dongle. at last only esp32 can do. because esp32 easy dev and support BT classic.
but esp32 not include classic hid library, but you btstack have!
btstack is cool! i must say so cool!
wii console scan mode IAC only LIAC, so i must do this.
scan mode set to LIAC just begining.
my next step is clone wiimote sdp .....