bluekitchen/btstack #255: HSP: PCM encoding instead of CVSD or mSBC

howroyd, 2019-10-09:
Would it be possible to support standard PCM encoding for HSP alongside the already implemented CVSD and mSBC, please?

mringwal, 2019-10-09:
Could you explain what you want to achieve? I don't understand your request. 

SCO channels transfer 8000 kbs. For narrow-band audio, e.g. HSP, the Bluetooth controller uses CVSD to encode 8Khz/16 bit audio. For wide-band speech, mSBC is used, in BTstack in the Host Stack, to encode 16 khz/16 bit into 8kbps and sent transparently over SCO.

Asides from CVSD, the Bluetooth controller can also use u-law, which is rarely used.

Independent from this, the data can either be sent over HCI (SCO-over-HCI), or, it can be accessible via dedicated I2S/PCM pins on the Bluetooth Controller. Do you refer to this?

mringwal, 2020-06-22:
No response for a while. Closing for now. Feel free to re-open with new information.

closed 2020-06-22 by mringwal