bluekitchen/btstack #256: [ESP32] Is it still true that CONFIG_BLUEDROID_ENABLED needs to be disabled?

piotrkundu, 2019-10-23:
I tried to disable Bluedroid by commenting out CONFIG_BLUEDROID_ENABLED in a2dp_sink_demo sdkconfig and I got this error:
```
/Users/piotrk/_git/esp-idf/components/bt/controller/bt.c:1486: undefined reference to `bredr_sco_datapath_set'
```
Just making sure, that the docs and code are in sync.

**PS. The below is NOT related to the issue, tips are welcome, but it's most likely HW or esp-idf and just an explenation to why I disabled CONFIG_BLUEDROID_ENABLED**
###Background 
I'm having issues with flashing spp_counter on ESP32 WeMos, Geekcreit and ESP32S AI-Thinker module and ESP32-CAM. All source from Banggood ( I know, I should get Adafruit stuff next time). I have a Adafruit Huzzah32 Feather that works flawlessly. Note it's spp_counter that has this problem, not other projects including a2dp_sink_demo combined with AVRCP and all.

```
 make flash monitor -j8
Toolchain path: /Users/piotrk/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin//xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /Users/piotrk/_git/esp-idf/requirements.txt are satisfied.

Project is not inside a git repository, will not use 'git describe' to determine PROJECT_VER.
App "spp_counter" version: 1
Flashing binaries to serial port /dev/cu.SLAB_USBtoUART (app at offset 0x10000)...
esptool.py v2.7-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_____....._
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: cc:50:e3:a9:9e:c8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1000000
Changed.
Configuring flash size...

A fatal error occurred: Timed out waiting for packet content
```
In another case I get:
```Serial port /dev/cu.SLAB_USBtoUART
Connecting........_____....._____....._
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: cc:50:e3:a9:9e:c8
Uploading stub...

A fatal error occurred: Invalid head of packet (0x65)
```

mringwal, 2019-10-23:
Bluedroid hasn't been needed yet. 
On ESP-IDF Master, Espressif added support for the NimBLE stack, but you cannot disable both. 
Pull request for a fix: https://github.com/espressif/esp-idf/pull/4142

What branch/version of ESP-IDF did you use? Could you try with the one from the Pull Request or e.g. release 3.3?

piotrkundu, 2019-10-24:
I'm on this 4.0-dev branch. Might be the reason. I have Bludroid enabled, but I would say connection is "shaky" and some boards are worse then others.

git describe
v4.0-dev-1443-g39f090a4f

mringwal, 2019-10-25:
So, can you run BTstack examples, or do you fail to flash them?

piotrkundu, 2019-10-30:
I fail to flash them and the reason is because I had two terminals open and second one of them got `Invalid head of packet (0x65)` after a while. A colleague had the same issue. It works in the first terminal.

piotrkundu, 2019-10-30:
But with esp-idf v4 and `CONFIG_BLUEDROID_ENABLED` not set I get `undefined reference to 'bredr_sco_datapath_set'`, so the question is if the BTstack documentation is correct for v4.

mringwal, 2019-10-31:
Please try with btstack/develop and esp-idf/master and report back - to  have (roughly) the same versions. I just did and had to fix a config define change, but it compiled after that.

mringwal, 2020-06-22:
The BTstack examples build even with CONFIG_BLUEDROID_ENABLED.
Closing for now. Feel free to re-open with new status.

closed 2020-06-22 by mringwal