bluekitchen/btstack #260: esp32 can't open two sco channels

danielsafrigin, 2019-11-05:
We have three devices A, B C, and I want to establish an SCO connection from A to B and C

A <-> establish RFCOMM connection B
A-> request SCO connection B
B -> accept SCO connection A
A <-> establish RFCOMM connection C
A-> request SCO connection C
C -> rejects SCO connection A

the error message is number 17, which by this site:
https://lisha.ufsc.br/teaching/shi/ine5346-2003-1/work/bluetooth/hci_commands.html
`17 | Unsupported Feature Or Parameter Value.`

request code: 
`hci_send_cmd(&hci_setup_synchronous_connection, rfcomm_handle, 8000, 8000, 0xFFFF, hci_get_sco_voice_setting(), 0x00, 0x03C4);`

accept code:
`hci_send_cmd(&hci_accept_synchronous_connection, event_addr, 8000, 8000, 0xFFFF, hci_get_sco_voice_setting(), 0x00, 0x03C4);`

0x03C4 is HV3 packet_type

If instead of opening two SCO connections, I try to open two eSCO connections and use packet type EV3/EV4/EV5; it does work.

also, I will note that I tried the same code with a Broadcom chip and did manage to open two SCO connections.

mringwal, 2019-11-05:
Hi. Could you enable hci_dump in esp-idf/components/btstack/main.c, and convert the console output using btstack/tool/create_packet_log.py for all three nodes and post it here?

Do you now why C rejects the SCO connection?

If the logs don't show an error on the HCI level, it would be good to post an error at the esp-idf project.

mringwal, 2020-06-22:
Did not get any feedback. Closing this for now. Feel free to re-open with new information.

closed 2020-06-22 by mringwal