bluekitchen/btstack #261: [ESP32] A2DP_SUBEVENT_STREAMING_CAN_SEND_MEDIA_PACKET_NOW packet a2dp cid

ThisUsernameIsUncreative, 2019-11-10:
While modifying the a2dp exaple I stumbled upon the problem that the A2DP_SUBEVENT_STREAMING_CAN_SEND_MEDIA_PACKET_NOW event packet does not contain the a2dp cid. upon calling:
`
// for debug reasons
printf("A2DP Timer - requesting send for cid %x, seid %x\n", context->a2dp_cid, context->local_seid);
a2dp_source_stream_endpoint_request_can_send_now(context->a2dp_cid, context->local_seid);
`
the A2DP_SUBEVENT_STREAMING_CAN_SEND_MEDIA_PACKET_NOW got triggered as expected and the serial output was:
`
A2DP Timer - requesting send for cid 57, seid 1
`
 but:
`
cid = a2dp_subevent_streaming_can_send_media_packet_now_get_a2dp_cid(packet);
local_seid = a2dp_subevent_streaming_can_send_media_packet_now_get_local_seid(packet);
i = getDeviceIndexForA2DPCid(cid);
j = getDeviceIndexForLocalSeid(local_seid);
if(i != j)
{
    printf("A2DP Stream - Device inconsistency at stream send! idx1: %d idx2 : %d cid(idx1): %x seid(idx2): %x\n", i, j, cid, local_seid);
    return;
}
if(i < 0 || j < 0)
{
    printf("A2DP Stream - Device not found at stream send! idx1: %d idx2 : %d cid(idx1): %x seid(idx2): %x\n", i, j, cid, local_seid);
    return;
}
`

yieldet:

`
A2DP Stream - Device not found at stream send! idx1: -1 idx2 : -1 cid(idx1): 42 seid(idx2): 0
`

The log showed that this was the l2cap cid (and the seid was also not what I expected). 
`
[00:00:16.730] LOG -- l2cap.c.885: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr 28:A4:82:53:50:48 handle 0x80 psm 0x19 local_cid 0x42 remote_cid 0x46 local_mtu 1691, remote_mtu 716, flush_timeout 0
[00:00:16.747] LOG -- avdtp.c.596: L2CAP_EVENT_CHANNEL_OPENED: status 0, cid 0x42 , signaling connection 0x3ffd2cf0 
[00:00:16.758] LOG -- avdtp.c.649: AVDTP_STREAM_ENDPOINT_OPENED, avdtp cid 0x57, l2cap_media_cid 0x42, local seid 1, remote seid 1

`
The second issue is that i could not get the packet hadler to recieve l2cap event packets to index by the l2cap cid. All exaples of  `l2cap_register_service()` I could find didn't work - the packets were not recieved by the packet handler.

mringwal, 2019-11-12:
Hi. Thanks for reporting. A2DP uses at least 2 L2CAP channels and it looks like BTstack is using the wrong one when sending the A2DP_SUBEVENT_STREAMING_CAN_SEND_MEDIA_PACKET_NOW event. We will have a look at this later this week.

Could you explain the second issue in more detail, and/or, create another GitHub issue with it?
What did you want to do? What did you expect? What happened?

ThisUsernameIsUncreative, 2019-11-12:
I don't know if its an issue or if i haven't understood the library enough yet. I tried to get the l2cap packets delivered to the same paket handler by calling `l2cap_register_packet_handler(&packet_handler)` - it didnt work - and I have no clue what the parameters of `l2cap_register_service()` really mean. I tried every example of the function call i found in the documentation but recieved no l2cap packets to the function handler.

But it would be a workaround anyway to index by the l2cap cid.

milamikica, 2019-11-15:
Hi, could you please check the fix and let us know if it works for you?

ThisUsernameIsUncreative, 2019-11-15:
Thanks now it works for the cid - but the seid is still 0. now it says:
```
A2DP Stream - Device inconsistency at stream send! idx1: 0 idx2 : -1 cid(idx1): 57 seid(idx2): 0
```
(output from the same snippet as in my first comment)

mringwal, 2019-11-15:
Thanks for the update. We'll have another look...

milamikica, 2019-11-18:
Hi, I've tested with the a2dp_source_demo , and I cannot reproduce it here (a2dp cid, local seid are correct). Could you try the "a2dp_source_demo" and see if the local seid is correct?

ThisUsernameIsUncreative, 2019-11-19:
It works with the demo - the wierd behavieour is probably an artefact of me trying to establish multiple streams at once

milamikica, 2019-11-19:
Hi, could you explain your setup? Do you have multiple streams to one device, or are you streaming to multiple devices?

> On 19 Nov 2019, at 19:18, ThisUsernameIsUncreative <notifications@github.com> wrote:
> 
>  the wierd behavieour is probably an artefact of me trying to establish multiple streams at once

ThisUsernameIsUncreative, 2019-11-19:
I have an esp32 and try to stream audio to multiple sinks.
So i call `a2dp_source_establish_stream(devices[i].address, media_tracker[j].local_seid, &media_tracker[j].a2dp_cid);`. for every sink (stream endpoints for each are already created at this point). 

- When I try to establish streams in a loop there are mixups at `A2DP_SUBEVENT_STREAM_ESTABLISHED`
- When I try to establish the stream after `A2DP_SUBEVENT_STREAM_ESTABLISHED` the estabilshing process gets stuck and the log says `a2dp_source.c.376: A2DP initiate set configuration locally and wait for response ...` 
- When I start the first stream ( but dont start the timer and try to establish the next stream out of the `A2DP_SUBEVENT_STREAM_STARTED` event) the log gets stuck at  `avdtp.c.693: Received L2CAP_EVENT_CHANNEL_CLOSED, cid 0x45, connection 0x3ffd4c94, stream_endpoint 0x0`
- When I accidentaly startet the timer (by not returning) a second stream got established - but the timer crashed. (unfortunatly I forgot to save the logs for that)

At this point my code im experimenting with is pretty messed up from all the try and error (and im out of ideas).

milamikica, 2019-11-20:
Hi, 

I’ve setup a new branch a2dp-multi-sink, and updated a2dp_source_demo.c to use the list of media trackers instead of a single one.

It’s not complete yet, but the connection establishment works, and [cid,seid] seems to be correct. The following is missing:
- currently the two media trackers are hardcoded (see a2dp_source_demo.c: lines 382-382)
- sample rate needs to go from global to media tracker context)
- sending audio to different sink devices is not handled

To try it out:
- checkout a2dp-multi-sink branch
- adapt Biuetooth addresses  to your setup (hardcoded in device_addr_string1, device_addr_string2)
- select first media tracker as the current one by typing ‘1’ on console 
- initiate connection to remote device by typing ‘b’ on console
- select second media tracker as the current one by typing ‘2’ on console 
- initiate connection to remote device by typing ‘b’ on console

ThisUsernameIsUncreative, 2019-11-21:
Hi,

I tried the multi sink setup and two streams got set up. But it didnt stream to the two devices simultaniously. I attached the log for this run (I removed all the dublicate lines of the CAN_SEND_NOW log per sending "block" to make it more readable).
[1b1b2b1x.txt](https://github.com/bluekitchen/btstack/files/3874651/1b1b2b1x.txt)

Log named after the commands i entered in order.

commit e85a5146 (2019-11-15): avdtp initiator: use avdtp_cid and local seid when calling a2dp_streaming_emit_can_send_media_packet_now, fixes #261 [a2381]
closed 2019-12-04 by mringwal