bluekitchen/btstack #262: hfp_ag

losper, 2019-11-12:


losper, 2019-11-12:
add the code:
``` C++
            if (hci_event_packet_get_type(event) != HCI_EVENT_HFP_META) return;
            if (event[3]
                && hci_event_hfp_meta_get_subevent_code(event) != HFP_SUBEVENT_PLACE_CALL_WITH_NUMBER
                && hci_event_hfp_meta_get_subevent_code(event) != HFP_SUBEVENT_ATTACH_NUMBER_TO_VOICE_TAG 
                && hci_event_hfp_meta_get_subevent_code(event) != HFP_SUBEVENT_TRANSMIT_DTMF_CODES){
                log_info("ERROR, status: %u\n", event[3]);
                return;
            }
```
How do I know what event[3] means?

skoperst, 2019-11-12:
Hi and Welcome.

HCI_EVENT_HFP_META  are events sent by hfp.c or hfp_ag.c.  Probably something sent you by 'hfp_emit_event' in hfp.c

Error code 106 looks like 'L2CAP_CONNECTION_BASEBAND_DISCONNECT' (0x6A) which means someone disconnected your HFP connection from a lower layer (L2CAP). I'm not sure though if its the remote side or the controller

mringwal, 2019-11-12:
FWIW, all status codes have been/are collected in src/bluetooth.h now https://github.com/bluekitchen/btstack/blob/develop/src/bluetooth.h

To understand what's going on, you can open the hci_dump.pklg file in Wireshark

closed 2019-11-12 by mringwal

losper, 2019-11-12:
thanks @skoperst @mringwal