bluekitchen/btstack #263: l2cap INTERRUPT only opened 0.5 sec

rayPopo, 2019-12-05:
hi:
i use esp32 sim nintendo switch pro controller with btstack.but INTERRUPT only opened 0.5 sec. i used https://github.com/Molorius/esp-idf/tree/feature/btclassic-hid-device to do this project same ok.

### **this is btstack log**

BTstack: setup
[00:00:00.070] LOG -- btstack_run_loop_freertos.c.301: run loop init, task 0x3ffbba3c, queue item size 8
[00:00:00.079] LOG -- main.c.271: transport_register_packet_handler
[00:00:00.101] LOG -- btstack_tlv.c.48: TLV Instance 0x3f4097b0
[00:00:00.437] LOG -- l2cap.c.3550: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
[00:00:00.438] LOG -- l2cap.c.3550: L2CAP_REGISTER_SERVICE psm 0x13 mtu 700
[00:00:00.440] LOG -- l2cap.c.3550: L2CAP_REGISTER_SERVICE psm 0x11 mtu 700
[00:00:00.447] LOG -- hci.c.3060: hci_power_control: 1, current mode 0
[00:00:00.453] LOG -- main.c.208: transport_init
[00:00:00.457] LOG -- main.c.224: transport_open
I (462) BTDM_INIT: BT controller compile version [acab313]
I (468) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (542) phy: phy_version: 4009, e7844c0, Jul 15 2019, 13:20:06, 0, 0
[00:00:00.856] LOG -- hci.c.4203: BTSTACK_EVENT_STATE 1
[00:00:00.857] EVT <= 60 01 01 
[00:00:00.857] CMD => 03 0C 00 
BTstack: execute run loop
[00:00:00.858] EVT <= 6E 00 
[00:00:00.861] EVT <= 0E 04 05 03 0C 00 
[00:00:00.865] CMD => 01 10 00 
[00:00:00.868] EVT <= 0E 0C 05 01 10 00 08 0E 03 08 60 00 0E 03 
[00:00:00.873] LOG -- hci.c.2001: Manufacturer: 0x0060
[00:00:00.878] EVT <= 6E 00 
[00:00:00.881] CMD => 14 0C 00 
[00:00:00.884] EVT <= 0E FC 05 14 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[00:00:00.952] LOG -- hci.c.1899: local name: 
[00:00:00.956] LOG -- hci.c.1658: Received local name, need baud change 0
[00:00:00.963] EVT <= 6E 00 
[00:00:00.965] CMD => 02 10 00 
[00:00:00.968] EVT <= 0E 44 05 02 10 00 BF FF FF 03 CC FF EF FF 3F FF FC 1F F2 0F E8 FE 3F F7 8F FF 1C 00 04 00 61 F7 FF FF 7F 38 C0 FF 7F FF FF 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[00:00:00.988] LOG -- hci.c.2012: Local supported commands summary 0x3f
[00:00:00.995] EVT <= 6E 00 
[00:00:00.997] CMD => 09 10 00 
[00:00:01.000] EVT <= 0E 0A 05 09 10 00 56 BE 6D 12 CF A4 
[00:00:01.005] LOG -- hci.c.1956: Local Address, Status: 0x00: Addr: A4:CF:12:6D:BE:56
[00:00:01.013] EVT <= 6E 00 
[00:00:01.016] CMD => 05 10 00 
[00:00:01.019] EVT <= 0E 0B 05 05 10 00 FD 03 FF 09 00 04 00 
[00:00:01.024] LOG -- hci.c.1916: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4
[00:00:01.035] EVT <= 6E 00 
[00:00:01.037] CMD => 03 10 00 
[00:00:01.040] EVT <= 0E 0C 05 03 10 00 BF EE CD FE DB FF 7B 87 
[00:00:01.046] LOG -- hci.c.1983: Packet types cc18, eSCO 1
[00:00:01.051] LOG -- hci.c.1986: BR/EDR support 1, LE support 0
[00:00:01.057] EVT <= 6E 00 
[00:00:01.060] CMD => 33 0C 07 00 04 3C 14 00 0A 00 
[00:00:01.064] EVT <= 0E 04 05 33 0C 00 
[00:00:01.068] EVT <= 6E 00 
[00:00:01.071] CMD => 31 0C 01 03 
[00:00:01.074] EVT <= 0E 04 05 31 0C 00 
[00:00:01.077] EVT <= 6E 00 
[00:00:01.080] CMD => 01 0C 08 FF FF FF FF FF FF FF 1F 
[00:00:01.085] EVT <= 0E 04 05 01 0C 00 
[00:00:01.088] EVT <= 6E 00 
[00:00:01.091] CMD => 56 0C 01 01 
[00:00:01.094] EVT <= 0E 04 05 56 0C 00 
[00:00:01.098] EVT <= 6E 00 
[00:00:01.100] CMD => 18 0C 02 00 60 
[00:00:01.104] EVT <= 0E 04 05 18 0C 00 
[00:00:01.107] EVT <= 6E 00 
[00:00:01.110] CMD => 0F 08 02 00 00 
[00:00:01.114] EVT <= 0E 04 05 0F 08 00 
[00:00:01.117] EVT <= 6E 00 
[00:00:01.120] CMD => 24 0C 03 08 25 00 
[00:00:01.123] EVT <= 0E 04 05 24 0C 00 
[... 401 lines cut ...]
D (3604) BT_L2CAP: L2CAP - st: CONFIG evt: 15
D (3614) BT_L2CAP: CID:0x0042 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Quota:200
D (3624) BT_L2CAP: CID:0x0043 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Quota:200
<== 04 13 05 01 80 00 01 00 
I (3634) BT_L2CAP: L2CAP - Calling Config_Rsp_Cb(), CID: 0x0042
D (3634) BT_L2CAP: L2CAP - st: CONFIG evt: 14
D (3644) BT_L2CAP: l2c_fcr_process_peer_cfg_req() CFG fcr_present:0 fcr.mode:0 CCB FCR mode:0 preferred: 0 allowed:1
D (3654) BT_L2CAP: L2CAP - Calling Config_Req_Cb(), CID: 0x0043, C-bit 0
I (3654) BT_L2CAP: L2CA_ConfigRsp()  CID: 0x0043  Result: 0 MTU present:0 Flush TO:0 FCR:0 FCS:0
D (3664) BT_L2CAP: L2CAP - st: CONFIG evt: 25
==> 02 80 20 0e 00 0a 00 01 00 05 0d 06 00 43 00 00 00 00 00 
D (3674) BT_L2CAP: L2CAP - st: CONFIG evt: 15
<== 04 13 05 01 80 00 01 00 
<== 04 38 04 80 00 80 0c 
D (3684) BT_L2CAP: CID:0x0042 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Quota:200
D (3694) BT_L2CAP: CID:0x0043 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Quota:200
I (3704) BT_L2CAP: L2CAP - Calling Config_Rsp_Cb(), CID: 0x0043
I (3714) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
W (3714) BT: HID 98:b6:e9:e7:a0:50 connected
==> 01 1a 0c 01 02 
<== 04 0e 04 05 1a 0c 00 
==> 01 1a 0c 01 00 
<== 04 0e 04 05 1a 0c 00 
<== 04 1b 03 80 00 01 
<== 04 14 06 00 80 00 02 18 00

mringwal, 2019-12-06:
Hi. Thanks for the two logs. I've converted the one from BTstack. 

[esp32-nes-hid.pklg.zip](https://github.com/bluekitchen/btstack/files/3931079/esp32-nes-hid.pklg.zip)

The log looks good so far: incoming connection, encryption setup, l2cap channels for Control and Interrupt are configured and open. Then, the remote disconnects them.

Do you know if there's a tool similar to BTstack's tool/create_packet_log.py to convert the log from Bluedroid into something Wireshark can display? I can see log messages that contain 'disconnect' as well, but then it continues somehow.

rayPopo, 2019-12-06:
thanks for you reponse,i modifid create_packet_log.py and convert bluedroid to wireshark format log
[e.pklg.zip](https://github.com/bluekitchen/btstack/files/3931916/e.pklg.zip)

rayPopo, 2019-12-06:
and i got a other source code write by bluez from my friend. emu pro controller success. this is log
[bluez.pklg.zip](https://github.com/bluekitchen/btstack/files/3931940/bluez.pklg.zip)

mringwal, 2019-12-11:
Can you confirm the setup again? I understand that you want to simulate an Nintendo Switch Pro Controller (an HID device) with the ESP32, is this correct?

Can you unpair and make new logs that include the first connection? During the first connection, the remote (the Nintendo Switch?) will do an SDP query for HID. You  can then compare the SDP responses. 

The most important part is the HID Descriptor. Did you copy the one from an original device? 

In the log from Bluedroid, I can see "Service Name: Wireless Gamepad". but don't see where hat comes from in the example you have mentioned: https://github.com/Molorius/esp-idf/tree/feature/btclassic-hid-device

rayPopo, 2019-12-13:
thanks for you reqonse.

>Can you confirm the setup again? I understand that you want to simulate an Nintendo Switch Pro Controller (an HID device) with the ESP32, is this correct?

Yes. i had simulate DS4 and wiimote by esp32 sucessed(DS4 need a real DS4 for auth report),but Pro Controller failed.

>Can you unpair and make new logs that include the first connection? During the first connection, the remote (the Nintendo Switch?) will do an SDP query for HID. You can then compare the SDP responses.

Yes. 
[log.pklg.zip](https://github.com/bluekitchen/btstack/files/3958701/log.pklg.zip)

>The most important part is the HID Descriptor. Did you copy the one from an original device?

Yes.

this my source [https://github.com/rayPopo/esp32_switch_pro](url)

>In the log from Bluedroid, I can see "Service Name: Wireless Gamepad". but don't see where hat comes from in the example you have mentioned: https://github.com/Molorius/esp-idf/tree/feature/btclassic-hid-device

i modify esp-hid/components/bt/bluedroid/stack/hid/hidd_api.cc HID_DevAddRecord function and change param subclass to emu  gamepad type. subclass == 1 is wiimote, subclass == 2 is DS4 and subclass == 3 is nintendo switch pro controller

[hidd_api.cc.zip](https://github.com/bluekitchen/btstack/files/3958749/hidd_api.cc.zip)

rayPopo, 2019-12-13:
sorry. i found bluedroid hid-device desc not pro controller sdp, i change it like pro controller sdp. but bluedroid still not disconnect. i don't known why.
this is correct sdp bluedroid log.
[Archive.zip](https://github.com/bluekitchen/btstack/files/3958920/Archive.zip)

mringwal, 2020-02-16:
hi. please try with update on develop branch. Might be similar to  https://github.com/bluekitchen/btstack/issues/268

mringwal, 2020-06-22:
Did not get more feedback. Closing it for now. Feel free to open with updated information.

closed 2020-06-22 by mringwal