bluekitchen/btstack #265: pbap_client_demo shows obex connect failed on esp32

uborfly, 2019-12-20:
I use pbap_client_demo on esp32,idf version v3.3.
I changed MAC to my microphone,then run demo on esp32.
I selected  establish PBAP connection and wait to pair, then my phone recive pair request and i permit the access to my contacts.Then the log display a error on esp32,it shows obex connect failed.Where did i wrong in this process?How can i run this demo successfully?
Here is the log:
[+] Connecting to 22:22:58:40:02:EF...
[00:00:24.518] LOG -- l2cap.c.1960: L2CAP_CREATE_CHANNEL addr 22:22:58:40:02:EF psm 0x1 mtu 1691 -> local mtu 1691
[00:00:24.521] LOG -- l2cap.c.1927: create channel 0x3ffd1290, local_cid 0x0041
[00:00:24.528] LOG -- hci.c.3931: Create_connection to 22:22:58:40:02:EF
[00:00:24.534] LOG -- hci.c.190: create_connection_for_addr 22:22:58:40:02:EF, type fd
[00:00:24.542] LOG -- hci.c.3943: conn state 0
[00:00:24.547] LOG -- hci.c.2067: command status (create connection), status 0, connection 0x3ffd1a88, addr 22:22:58:40:02:EF, type fd
[00:00:25.969] LOG -- hci.c.2159: Connection_complete (status=0) 22:22:58:40:02:EF
[00:00:25.969] LOG -- hci.c.2179: New connection: handle 128, 22:22:58:40:02:EF
[00:00:25.972] LOG -- hci.c.4305: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[00:00:25.979] LOG -- l2cap.c.1857: connection complete con_handle 0080 - for channel 0x3ffd1290 cid 0x0041
[00:00:25.989] LOG -- hci.c.2232: HCI_EVENT_READ_REMOTE_SUPPORTED_FEATURES_COMPLETE, bonding flags 6, eSCO 1
[00:00:25.998] LOG -- l2cap.c.2342: remote supported features, channel 0x3ffd1290, cid 0041 - state 4
[00:00:26.007] LOG -- l2cap.c.1885: l2cap received remote supported features, sec_level_0_allowed for psm 1 = 1
[00:00:26.018] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x41
[00:00:26.023] LOG -- l2cap.c.1225: l2cap_start_rtx for local cid 0x41
[00:00:26.029] LOG -- l2cap.c.2679: L2CAP signaling handler code 3, state 10
[00:00:26.036] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x41
[00:00:26.042] LOG -- l2cap.c.1232: l2cap_start_ertx for local cid 0x41
[00:00:26.048] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x41
[00:00:26.084] LOG -- l2cap.c.2679: L2CAP signaling handler code 3, state 10
[00:00:26.084] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x41
[00:00:26.086] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x41
[00:00:26.092] LOG -- l2cap.c.1225: l2cap_start_rtx for local cid 0x41
[00:00:26.098] LOG -- l2cap.c.2679: L2CAP signaling handler code 4, state 11
[00:00:26.105] LOG -- l2cap.c.2526: Remote MTU 672
[00:00:26.110] LOG -- l2cap.c.2679: L2CAP signaling handler code 5, state 11
[00:00:26.116] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x41
[00:00:26.123] LOG -- l2cap.c.2610: l2cap_signaling_handle_configure_response
[00:00:26.130] LOG -- l2cap.c.1032: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr 22:22:58:40:02:EF handle 0x80 psm 0x1 local_cid 0x41 remote_cid 0x4e local_mtu 1691, remote_mtu 672, flush_timeout 0
[00:00:26.155] LOG -- l2cap.c.1996: L2CAP_DISCONNECT local_cid 0x41 reason 0x0
[00:00:26.163] LOG -- l2cap.c.2679: L2CAP signaling handler code 7, state 13
[00:00:26.163] LOG -- l2cap.c.1060: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x41
[00:00:26.167] LOG -- sdp_client.c.473: SDP Client disconnected.
[00:00:26.173] LOG -- goep_client.c.342: Remote GOEP RFCOMM Server Channel: 19
[00:00:26.180] LOG -- rfcomm.c.2402: RFCOMM_CREATE_CHANNEL addr 22:22:58:40:02:EF channel #19 init credits 10
[00:00:26.189] LOG -- rfcomm.c.380: rfcomm_max_frame_size_for_l2cap_mtu:  1691 -> 1686
[00:00:26.197] LOG -- rfcomm.c.498: rfcomm_channel_create for service 0x0, channel 19 --- list of channels:
[00:00:26.207] LOG -- l2cap.c.1960: L2CAP_CREATE_CHANNEL addr 22:22:58:40:02:EF psm 0x3 mtu 1691 -> local mtu 1691
[00:00:26.217] LOG -- l2cap.c.1927: create channel 0x3ffd136c, local_cid 0x0042
[00:00:26.224] LOG -- l2cap.c.1982: l2cap_create_channel, hci connection 0x0080 already exists
[00:00:26.232] LOG -- l2cap.c.1857: connection complete con_handle 0080 - for channel 0x3ffd136c cid 0x0042
[00:00:26.241] LOG -- l2cap.c.1885: l2cap received remote supported features, sec_level_0_allowed for psm 3 = 0
[00:00:26.251] LOG -- hci.c.4469: gap_request_security_level requested level 2, planned level 0, current level 0
[00:00:26.261] LOG -- l2cap.c.1933: free channel 0x3ffd1290, local_cid 0x0041
[00:00:26.268] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x41
[00:00:26.275] LOG -- hci.c.2239: HCI_EVENT_LINK_KEY_REQUEST
[00:00:26.280] LOG -- hci.c.3631: responding to link key request
[00:00:26.285] LOG -- btstack_tlv_esp32.c.54: tag 42544c00 -> 42544C00
[00:00:26.292] LOG -- btstack_tlv_esp32.c.68: read tag 42544C00
[00:00:26.297] LOG -- btstack_tlv_esp32.c.54: tag 42544c01 -> 42544C01
[00:00:26.303] LOG -- btstack_tlv_esp32.c.68: read tag 42544C01
[00:00:26.309] LOG -- btstack_tlv_esp32.c.54: tag 42544c02 -> 42544C02
[00:00:26.315] LOG -- btstack_tlv_esp32.c.68: read tag 42544C02
[00:00:26.321] LOG -- btstack_tlv_esp32.c.54: tag 42544c03 -> 42544C03
[00:00:26.327] LOG -- btstack_tlv_esp32.c.68: read tag 42544C03
[00:00:26.333] LOG -- btstack_tlv_esp32.c.54: tag 42544c04 -> 42544C04
[00:00:26.339] LOG -- btstack_tlv_esp32.c.68: read tag 42544C04
[00:00:26.345] LOG -- btstack_tlv_esp32.c.54: tag 42544c05 -> 42544C05
[00:00:26.351] LOG -- btstack_tlv_esp32.c.68: read tag 42544C05
[00:00:26.357] LOG -- btstack_tlv_esp32.c.54: tag 42544c06 -> 42544C06
[00:00:26.363] LOG -- btstack_tlv_esp32.c.68: read tag 42544C06
[00:00:26.369] LOG -- btstack_tlv_esp32.c.54: tag 42544c07 -> 42544C07
[00:00:26.375] LOG -- btstack_tlv_esp32.c.68: read tag 42544C07
[00:00:26.380] LOG -- btstack_tlv_esp32.c.54: tag 42544c08 -> 42544C08
[00:00:26.387] LOG -- btstack_tlv_esp32.c.68: read tag 42544C08
[... 72 lines cut ...]
[00:00:30.423] LOG -- rfcomm.c.380: rfcomm_max_frame_size_for_l2cap_mtu:  1691 -> 1686
[00:00:30.430] LOG -- rfcomm.c.1145: L2CAP_EVENT_CHANNEL_OPENED: outgoing connection
[00:00:30.438] LOG -- rfcomm.c.1347: Sending SABM #0 - (multi 0x0x3ffd12fc)
[00:00:30.493] LOG -- rfcomm.c.1221: Received UA #0
[00:00:30.493] LOG -- rfcomm.c.930: Multiplexer up and running
[00:00:30.493] LOG -- rfcomm.c.1999: Muliplexer opened, sending UIH PN next
[00:00:30.499] LOG -- rfcomm.c.2012: Sending UIH Parameter Negotiation Command for #38 (channel 0x0x3ffd1338) mtu 1686
[00:00:30.534] LOG -- rfcomm.c.1639: Received UIH Parameter Negotiation Response max frame 990, credits 7
[00:00:30.534] LOG -- rfcomm.c.2040: Sending SABM #38
[00:00:30.621] LOG -- rfcomm.c.1600: Received UA #38
[00:00:30.622] LOG -- rfcomm.c.2073: Sending MSC CMD for #38
[00:00:30.622] LOG -- rfcomm.c.1757: rfcomm_channel_ready_for_open state 8, flags needed 0020, current 3000, rf credits 7
[00:00:30.631] LOG -- rfcomm.c.2080: Providing credits for #38
[00:00:30.637] LOG -- rfcomm.c.1757: rfcomm_channel_ready_for_open state 8, flags needed 0020, current a000, rf credits 7
[00:00:30.648] LOG -- rfcomm.c.1654: Received MSC RSP for #38
[00:00:30.653] LOG -- rfcomm.c.1757: rfcomm_channel_ready_for_open state 8, flags needed 0020, current a020, rf credits 7
[00:00:30.663] LOG -- rfcomm.c.1397: rfcomm_channel_opened!
[00:00:30.669] LOG -- rfcomm.c.266: RFCOMM_EVENT_CHANNEL_OPENED status 0x0 addr 22:22:58:40:02:EF handle 0x80 channel #19 cid 0x01 mtu 990
[00:00:30.681] LOG -- goep_client.c.172: goep_client: connection opened. cid 1, max frame size 990
[00:00:30.690] LOG -- pbap_client.c.669: pbap: connection established
[00:00:30.696] LOG -- rfcomm.c.1647: Received MSC CMD for #38,
[00:00:30.702] LOG -- rfcomm.c.1908: Sending MSC RSP for #38
[00:00:32.881] LOG -- rfcomm.c.1434: RFCOMM data UIH_PF, new credits channel 0x01: 4, now 10
[00:00:32.882] LOG -- pbap_client.c.719: pbap: obex connect failed, result 0xc6
[!] Connection failed, status 0xb1

mringwal, 2019-12-20:
Hi. Please enable HCI packet log, by uncommenting hci_dump_open in esp-idf/components/btstack/main.c and post a new log. You can use btstack/tool/create_packet_log.py to convert it to .pklg and view in Wireshark

uborfly, 2019-12-23:
@mringwal Thanks for your reply! Here is the log i enabled HCI packet log.
a - establish PBAP connection to 22:22:58:40:02:EF
b - select SIM1
r - set path to '/telecom'
R - set path to '/SIM1/telecom'
u - set path to 'telecom/pb'
v - set vCardSelector to N and TEL
V - set vCardSelectorOperator to AND
e - select phonebook 'pb'
f - select phonebook 'fav'
i - select phonebook 'ich'
o - select phonebook 'och'
m - select phonebook 'mch'
c - select phonebook 'cch'
s - select phonebook 'spd'
d - get size of    'telecom/pb.vcf'
g - pull phonebook 'telecom/pb.vcf'
h - pull vCard listing 'telecom/pb'
l - get vCard 0.vcf
L - get vCard X-BT-UID::1234567890ABCDEF1234567890000001
j - Lookup contact with number '13991827789'
t - disconnnect
p - authenticate using password '0000'
r - set path to 'telecom'
x - abort operation

[00:00:01.415] EVT <= 6E 00
[+] Connecting to 22:22:58:40:02:EF...
[00:06:41.861] LOG -- l2cap.c.1960: L2CAP_CREATE_CHANNEL addr 22:22:58:40:02:EF psm 0x1 mtu 1691 -> local mtu 1691
[00:06:41.863] LOG -- l2cap.c.1927: create channel 0x3ffd1290, local_cid 0x0041
[00:06:41.870] LOG -- hci.c.3931: Create_connection to 22:22:58:40:02:EF
[00:06:41.876] LOG -- hci.c.190: create_connection_for_addr 22:22:58:40:02:EF, type fd
[00:06:41.884] LOG -- hci.c.3943: conn state 0
[00:06:41.888] CMD => 05 04 0D EF 02 40 58 22 22 18 CC 00 00 00 00 01
[00:06:41.895] EVT <= 6E 00
[00:06:41.897] EVT <= 0F 04 00 05 05 04
[00:06:41.901] LOG -- hci.c.2067: command status (create connection), status 0, connection 0x3ffd1a88, addr 22:22:58:40:02:EF, type fd
[00:06:42.248] EVT <= 03 0B 00 80 00 EF 02 40 58 22 22 01 00
[00:06:42.248] LOG -- hci.c.2159: Connection_complete (status=0) 22:22:58:40:02:EF
[00:06:42.250] LOG -- hci.c.2179: New connection: handle 128, 22:22:58:40:02:EF
[00:06:42.257] LOG -- hci.c.4305: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[00:06:42.263] EVT <= 61 01 01
[00:06:42.266] LOG -- l2cap.c.1857: connection complete con_handle 0080 - for channel 0x3ffd1290 cid 0x0041
[00:06:42.275] CMD => 1B 04 02 80 00
[00:06:42.279] EVT <= 1B 03 80 00 05
[00:06:42.282] ACL <= 80 20 0A 00 06 00 01 00 0A 02 02 00 02 00
[00:06:42.288] EVT <= 0F 04 00 05 1B 04
[00:06:42.292] EVT <= 0B 0B 00 80 00 BF FE CF FE DB FF 7B 87
[00:06:42.297] LOG -- hci.c.2232: HCI_EVENT_READ_REMOTE_SUPPORTED_FEATURES_COMPLETE, bonding flags 6, eSCO 1
[00:06:42.307] LOG -- l2cap.c.2342: remote supported features, channel 0x3ffd1290, cid 0041 - state 4
[00:06:42.315] LOG -- l2cap.c.1885: l2cap received remote supported features, sec_level_0_allowed for psm 1 = 1
[00:06:42.325] EVT <= 6E 00
[00:06:42.328] ACL => 80 00 10 00 0C 00 01 00 0B 02 08 00 02 00 00 00 80 02 00 00
[00:06:42.336] EVT <= 6E 00
[00:06:42.338] ACL => 80 00 0C 00 08 00 01 00 02 01 04 00 01 00 41 00
[00:06:42.344] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x41
[00:06:42.350] LOG -- l2cap.c.1225: l2cap_start_rtx for local cid 0x41
[00:06:42.357] EVT <= 13 05 01 80 00 01 00
[00:06:42.360] ACL <= 80 20 0A 00 06 00 01 00 0A 03 02 00 03 00
[00:06:42.366] EVT <= 13 05 01 80 00 01 00
[00:06:42.370] ACL <= 80 20 10 00 0C 00 01 00 03 01 08 00 40 00 41 00 01 00 00 00
[00:06:42.377] LOG -- l2cap.c.2679: L2CAP signaling handler code 3, state 10
[00:06:42.384] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x41
[00:06:42.390] LOG -- l2cap.c.1232: l2cap_start_ertx for local cid 0x41
[00:06:42.397] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x41
[00:06:42.403] EVT <= 6E 00
[00:06:42.405] ACL => 80 00 14 00 10 00 01 00 0B 03 0C 00 03 00 00 00 06 00 00 00 00 00 00 00
[00:06:42.414] EVT <= 6E 00
[00:06:42.416] CMD => 35 0C 05 01 80 00 03 00
[00:06:42.421] EVT <= 13 05 01 80 00 01 00
[... 197 lines cut ...]
[00:06:43.656] EVT <= 13 05 01 80 00 01 00
[00:06:43.660] EVT <= 6E 00
[00:06:43.662] EVT <= 78 02 42 00
[00:06:43.666] CMD => 35 0C 05 01 80 00 03 00
[00:06:43.670] EVT <= 6E 00
[00:06:43.683] ACL <= 80 20 15 00 11 00 42 00 99 FF 19 04 C6 00 0C 10 00 FF FE CB 00 00 00 01 E1
[00:06:43.683] LOG -- rfcomm.c.1434: RFCOMM data UIH_PF, new credits channel 0x01: 4, now 10
[00:06:43.689] LOG -- pbap_client.c.719: pbap: obex connect failed, result 0xc6
[!] Connection failed, status 0xb1
[00:06:43.699] CMD => 35 0C 05 01 80 00 01 00
[00:06:43.703] EVT <= 6E 00
[00:07:18.052] EVT <= 05 04 00 80 00 13
[00:07:18.053] LOG -- l2cap.c.1060: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x42
[00:07:18.053] EVT <= 71 02 42 00
[00:07:18.055] LOG -- rfcomm.c.1168: L2CAP_EVENT_CHANNEL_CLOSED cid 0x42, mult 0x3ffd12fc
[00:07:18.063] LOG -- rfcomm.c.1170: L2CAP_EVENT_CHANNEL_CLOSED state 7
[00:07:18.069] LOG -- rfcomm.c.290: RFCOMM_EVENT_CHANNEL_CLOSED cid 0x01
[00:07:18.076] EVT <= 81 02 01 00
[+] Operation complete
[+] Connection closed
[00:07:18.083] LOG -- l2cap.c.1933: free channel 0x3ffd136c, local_cid 0x0042
[00:07:18.090] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x42
[00:07:18.096] LOG -- hci.c.934: Connection closed: handle 0x80, 22:22:58:40:02:EF
[00:07:18.103] LOG -- hci.c.4305: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
[00:07:18.110] EVT <= 61 01 00

mringwal, 2019-12-23:
It's easier to look at the HCI log in Wireshark :) (GitHub requires to e.g. zip it before uploading)
Anyway, BTstack sends a Connect request and gets the response 0xc6 == OBEX_RESP_NOT_ACCEPTABLE - defined in btstack/src/classic/obex.h

Anything special about your cellphone? Could you try with a different phone / an iPhone?

Hm... I just tried against my iOS 12 device and I got a different error: 0xd3. 
Don't know what it means. I'm sure it worked before..

uborfly, 2019-12-24:
@mringwal I tried iOS 12.0 and I got error 0xd3 too.And another OnePlus7T with Android 10.0 and SANSUNG NOTE 7 and NOTE 8 with Android 8.0 and Android 9.0 got error 0xc6. Is this the error in lib?

mringwal, 2019-12-24:
The error codes is returned from the remote device and is defined in the OBEX specification. A few codes are defined in btstack/src/classic/obex.h. 

I'm quite sure I did test PBAP with an Android 8 device before as well.

welle108, 2020-05-07:
Has there been any progress on this? I am getting the same error whenever I attempt to connect to my Pixel 3a XL running Android 10. 

```
[00:01:50.775] LOG -- rfcomm.c.266: RFCOMM_EVENT_CHANNEL_OPENED status 0x0 addr 58:CB:52:29:47:E2 handle 0x80 channel #5 cid 0x03 mtu 990
[00:01:50.787] LOG -- goep_client.c.172: goep_client: connection opened. cid 3, max frame size 990
[00:01:50.796] LOG -- pbap_client.c.660: pbap: connection established
[00:01:50.833] LOG -- rfcomm.c.1434: RFCOMM data UIH_PF, new credits channel 0x03: 4, now 10
[00:01:50.834] LOG -- pbap_client.c.714: pbap: obex connect failed, result 0xc6
[!] Connection failed, status 0xb1
```
I wasn't entirely sure if 0xb1 and 0xc6 are both OBEX responses, so I looked up both of their meanings in the OBEX specification. 0xb1 is 'Moved Permanentely' and 0xc6 is 'Not Acceptable'. Could the issue maybe be in goep_client.c?

mringwal, 2020-05-07:
Hi. Sorry, we didn't work on this. The first error, 0x0c6 is the OBEX result, so you get 'Not Acceptable'. The 0xb1 is OBEX_CONNECT_FAILED from bluetooth.h

Wireshark displays it as 
271	-31621996.317	22:22:58:40:02:ef ()			OBEX	Rcvd Not Acceptable						

Looking at your previous log, this is the response to OBEX Connect with the fixed PBAP UUID. (The path to the phonebook is another issue, but it comes later)

Without knowing what's wrong, here are some ideas:
- when I pair with my iPhone, it as me if I allow access to contact. If I say no, I would not be surprised to get an error when trying to connect via PBAP. 
- the OBEX connect has a few fields, e.g. OBEX Version number, max packet length. Maybe it doesn't like them. You try different older/newer values in the pbap_client.c, PBAP_W2_SEND_CONNECT_REQUEST case.

Please let me know if you make some progress.

welle108, 2020-05-07:
I think I may have found the answer. It looks like both MAP and PBAP were removed from Android 10:

https://developer.android.com/reference/android/bluetooth/BluetoothProfile.html

mringwal, 2020-05-08:
hi. I don't think PBAP was removed - after all, BTstack found the SDP record. Googling Android PBAP, gets us to the source cod of the OBEX Server: https://android.googlesource.com/platform/packages/apps/Bluetooth/+/refs/heads/master/src/com/android/bluetooth/pbap/BluetoothPbapObexServer.java

ResponseCodes.OBEX_HTTP_NOT_ACCEPTABLE is sent if the Target UUID is wrong, but we use the one from the PBAP spec, which is the same that is hard-coded there.

Can you try to figure out how to enable debug output? There are quite a few log.d() statements. :)

mringwal, 2020-05-08:
wow... very weird. It looks like BTstack sends 18 bytes of UUID, which is one of the reasons why Android would return 'not acceptable'. Could you try and use 16 in pbap_client.c, line 284:
- goep_client_header_add_target(pbap_client->goep_cid, pbap_uuid, 18);
+ goep_client_header_add_target(pbap_client->goep_cid, pbap_uuid, 16);

[pbap_patch.txt](https://github.com/bluekitchen/btstack/files/4597456/pbap_patch.txt)

mringwal, 2020-05-08:
.. or update your develop branch. It's an obvious error - once you look at the right spot.

welle108, 2020-05-08:
Hi again. Changing line 284 had no effect to the output. Installed adb and got [this](https://drive.google.com/file/d/19hojz9D_fTS0MxB8DZNO7DrPMkXtccS7/view?usp=sharing) when I ran 

`adb bugreport [PATH]`

Apologies for not being able help more. I'm pretty new to bluetooth, but looking through your code has helped a ton on a research project I'm doing

mringwal, 2020-05-08:
wow. that's a log of debug output. Could you post the current HCI packet log as well please?

welle108, 2020-05-17:
Hi Matthias. Apologies for the delay; had to move due to covid. I enabled the hci log, converted to a .pklg file, then did another bugreport with adb. You can find that [here](https://drive.google.com/drive/folders/1nGo4XyCRCaXKFteklNJ7rVfmXFBdCYkA?usp=sharing).

My goal is to get the pbap code working, then attempt to implement MAP with it. Their dependencies are basically identical, and just by looking at the way you implement the lower layers of the stack/ how they seem to work just fine aside from the actual pbap code at the top, I'm hoping it won't be too much of a nightmare. Do you have any recommendations for good resources for learning bt? I've got the official specifications and a few books, but still have a long way to go

mringwal, 2020-05-17:
Hi Scott. Hope you're well at your new place. 

In the log, the UUID for the OBEX Connect is still 18 bytes (although it's fixed now)
Please switch to the develop branch and do a git pull. Then, re-run the ./integrate_btstack.py and then rebuild and re-run the pbap_client_demo.

Our pbab_client sits on top of the goep_client, for a MAP Client, it would be the same.
Why do you want to implement MAP resp. what do you want to use MAP for?

In Bluetooth Classic, they tried to re-use existing protocols, e.g. OBEX, which didn't make life easier. For MAP, you need to understand OBEX, which was designed to look a bit like HTTP, but for IrDA, many years before Bluetooth.

welle108, 2020-05-18:
That did it. Thanks Matthias! 

To answer your question, I self-designed a research project about bluetooth to get out of a chem requirement at school, and found out that I'm really interested in the technology. Since my new grad offer got postponed until September, I thought implementing MAP would be a good way to improve my skills and learn more about being a better programmer. Once again, thank you so much for always responding so promptly. In a world of gatekeep-y assholes, it's really nice knowing of at least one place on the internet where I can ask an intelligent person a question and not get destroyed in the comments section lol

mringwal, 2020-05-18:
Hi Scott. Glad it worked after properly updating. I'm closing this one. Feel free to post on the BTstack Mailing list, it's well moderated :)

As for map, Mila started implementing a MAP Client on the 'map' branch in the test/map_client. We don't know exactly what's working resp. what's missing as had to switch priorities. After this spoiler, there's certainly no MAP Server :) - and you may peek at Android's Java implementation for hints...

closed 2020-05-18 by mringwal