bluekitchen/btstack #268: hid_host_demo does not receive data on L2CAP channels

Jascha-N, 2020-01-12:
Hi,

I've been trying to pair my ESP32 with a classic bluetooth game controller using slightly modified code from the hid_host_demo example. SDP discovery works fine and the HID Interrupt and Control channels are opened. However, no packets are received after this. Instead I get a constant stream of warnings from the ESP-IDF bluetooth library, which may or may not be related to the issue. I've tried pairing the game controller with my Windows PC and capturing the packets with Wireshark and USBPcap. The controller works fine on Windows and the packet flow looks very similar to one from the ESP. The only difference I can see is that my PC _does_ receive the HID input reports after opening the L2CAP channels while the ESP with btstack does not. I'm fairly inexperienced with bluetooth and I hope you can help me out here. I've added the log from the ESP32 including the HCI packet dump. Thanks in advance.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5400
load:0x40078000,len:11604
load:0x40080400,len:6360
entry 0x40080720
I (175) cpu_start: Pro cpu up.
I (175) cpu_start: Application information:
I (175) cpu_start: Project name:     hid_host_demo
I (178) cpu_start: App version:      1
I (182) cpu_start: Compile time:     Jan 12 2020 15:15:29
I (189) cpu_start: ELF file SHA256:  a23154820bb639c9...
I (195) cpu_start: ESP-IDF:          v3.3.1-dirty
I (200) cpu_start: Single core mode
I (204) heap_init: Initializing. RAM available for dynamic allocation:
I (211) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (217) heap_init: At 3FFB7468 len 00000B98 (2 KiB): DRAM
I (223) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (230) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (236) heap_init: At 3FFC82D0 len 00017D30 (95 KiB): DRAM
I (242) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (248) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (254) heap_init: At 40090DD4 len 0000F22C (60 KiB): IRAM
I (261) cpu_start: Pro cpu start user code
I (55) esp_core_dump_common: Init core dump to UART
E (55) esp_core_dump_common: No core dump partition found!
I (56) cpu_start: Starting scheduler on PRO CPU.
BTstack: setup
[00:00:00.064] LOG -- btstack_run_loop_freertos.c.299: run loop init, task 0x3ffbc06c, queue item size 8
[00:00:00.073] LOG -- main.c.271: transport_register_packet_handler
[00:00:00.084] LOG -- btstack_tlv.c.48: TLV Instance 0x3f4052bc
[00:00:00.084] LOG -- btstack_tlv_esp32.c.54: tag 42544400 -> 42544400
[00:00:00.091] LOG -- btstack_tlv_esp32.c.68: read tag 42544400
[00:00:00.096] LOG -- btstack_tlv_esp32.c.54: tag 42544401 -> 42544401
[00:00:00.103] LOG -- btstack_tlv_esp32.c.68: read tag 42544401
[00:00:00.108] LOG -- btstack_tlv_esp32.c.54: tag 42544402 -> 42544402
[00:00:00.114] LOG -- btstack_tlv_esp32.c.68: read tag 42544402
[00:00:00.120] LOG -- btstack_tlv_esp32.c.54: tag 42544403 -> 42544403
[00:00:00.126] LOG -- btstack_tlv_esp32.c.68: read tag 42544403
[00:00:00.132] LOG -- btstack_tlv_esp32.c.54: tag 42544404 -> 42544404
[00:00:00.138] LOG -- btstack_tlv_esp32.c.68: read tag 42544404
[00:00:00.144] LOG -- btstack_tlv_esp32.c.54: tag 42544405 -> 42544405
[00:00:00.150] LOG -- btstack_tlv_esp32.c.68: read tag 42544405
[00:00:00.156] LOG -- btstack_tlv_esp32.c.54: tag 42544406 -> 42544406
[00:00:00.162] LOG -- btstack_tlv_esp32.c.68: read tag 42544406
[00:00:00.168] LOG -- btstack_tlv_esp32.c.54: tag 42544407 -> 42544407
[00:00:00.174] LOG -- btstack_tlv_esp32.c.68: read tag 42544407
[00:00:00.180] LOG -- btstack_tlv_esp32.c.54: tag 42544408 -> 42544408
[00:00:00.186] LOG -- btstack_tlv_esp32.c.68: read tag 42544408
[00:00:00.191] LOG -- btstack_tlv_esp32.c.54: tag 42544409 -> 42544409
[00:00:00.198] LOG -- btstack_tlv_esp32.c.68: read tag 42544409
[00:00:00.203] LOG -- btstack_tlv_esp32.c.54: tag 4254440a -> 4254440A
[00:00:00.210] LOG -- btstack_tlv_esp32.c.68: read tag 4254440A
[00:00:00.215] LOG -- btstack_tlv_esp32.c.54: tag 4254440b -> 4254440B
[00:00:00.221] LOG -- btstack_tlv_esp32.c.68: read tag 4254440B
[00:00:00.227] LOG -- btstack_tlv_esp32.c.54: tag 4254440c -> 4254440C
[00:00:00.233] LOG -- btstack_tlv_esp32.c.68: read tag 4254440C
[00:00:00.239] LOG -- btstack_tlv_esp32.c.54: tag 4254440d -> 4254440D
[00:00:00.245] LOG -- btstack_tlv_esp32.c.68: read tag 4254440D
[00:00:00.251] LOG -- btstack_tlv_esp32.c.54: tag 4254440e -> 4254440E
[00:00:00.257] LOG -- btstack_tlv_esp32.c.68: read tag 4254440E
[00:00:00.263] LOG -- btstack_tlv_esp32.c.54: tag 4254440f -> 4254440F
[... 334 lines cut ...]
[00:00:03.226] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x43
[00:00:03.232] ACL => 80 00 10 00 0C 00 01 00 04 07 08 00 43 00 00 00 01 02 30 00
[00:00:03.239] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x43
[00:00:03.245] LOG -- l2cap.c.1225: l2cap_start_rtx for local cid 0x43
[00:00:03.252] ACL <= 80 20 10 00 0C 00 01 00 04 03 08 00 43 00 00 00 01 02 80 00
[00:00:03.259] LOG -- l2cap.c.2679: L2CAP signaling handler code 4, state 11
[00:00:03.265] LOG -- l2cap.c.2526: Remote MTU 128
[00:00:03.270] EVT <= 13 05 01 80 00 01 00
[00:00:03.274] ACL <= 80 20 0E 00 0A 00 01 00 05 07 06 00 43 00 00 00 00 00
[00:00:03.281] LOG -- l2cap.c.2679: L2CAP signaling handler code 5, state 11
[00:00:03.287] LOG -- l2cap.c.1216: l2cap_stop_rtx for local cid 0x43
[00:00:03.294] LOG -- l2cap.c.2610: l2cap_signaling_handle_configure_response
[00:00:03.301] EVT <= 6E 00
[00:00:03.303] ACL => 80 00 12 00 0E 00 01 00 05 03 0A 00 43 00 00 00 00 00 01 02 30 00
[00:00:03.311] LOG -- l2cap.c.1032: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr DF:90:78:01:5B:9B handle 0x80 psm 0x13 local_cid 0x43 remote_cid 0x43 local_mtu 48, remote_mtu 128, flush_timeout 0
[00:00:03.328] EVT <= 70 18 00 9B 5B 01 78 90 DF 80 00 13 00 43 00 43 00 30 00 80 00 00 00 00 00 00
HID Connection established
[00:00:03.339] EVT <= 13 05 01 80 00 01 00
[00:00:03.343] EVT <= 6E 00
[00:00:03.346] CMD => 35 0C 05 01 80 00 03 00
[00:00:03.350] EVT <= 6E 00
ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1
 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_ta
sk.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054
```

mringwal, 2020-01-13:
Hi Jascha. The log looks correct until the 'ASSERT_WARN' show up. An HCI connection was created and both HID L2CAP Channels fully established. The warnings are from the lower layers of the Bluetooth implemented in the ESP32, as it's closed source, I don't know what it means, but it should not be there.

I would suggest to:
a) post your log on the esp-idf github issue tracker and ask about the warning at the end
b) try your BTstack code on a different system, e.g. on your Windows machine with an USB Bluetooth Dongle - see btstack/port/windows-winusb

Jascha-N, 2020-02-11:
I've tried running it on my Windows machine, but the results are the same. It connects but does not receive any inputs. I feel like the warning message was just a red herring.

Console output:
```
BTstack/windows-winusb booting up
Packet Log: hci_dump.pklg
Done 0
BTstack up and running at 00:15:83:EA:56:1B
Start SDP HID query for remote HID Device.
HID Control PSM: 0x0011
HID Interrupt PSM: 0x0013
HID Descriptor:
05 01 09 02 A1 01 85 02 09 01 A1 00 05 09 19 01 29 05 15 00 25 01 95 05 75 01 81
 02 95 01 75 03 81 01 05 01 09 30 09 31 09 38 15 81 25 7F 75 08 95 03 81 06 C0 C
0 05 01 09 06 A1 01 85 01 05 07 15 00 26 FF 00 75 08 95 06 19 00 29 FF 81 00 C0
05 0C 09 01 A1 01 85 03 15 00 26 80 03 19 00 2A 80 03 75 10 95 01 81 00 C0 05 01
 09 05 A1 01 85 04 05 01 26 FF 00 46 FF 00 09 30 09 31 09 32 09 35 75 08 95 04 8
1 02 05 01 25 07 46 3B 01 75 04 95 01 65 14 09 39 81 42 15 00 25 01 35 00 45 01
75 01 95 04 05 09 09 01 09 02 09 04 09 05 81 02 15 00 25 01 35 00 45 01 75 01 95
 08 05 09 09 07 09 08 09 0B 09 0C 09 0E 09 0F 09 09 09 0A 81 02 05 02 15 00 26 F
F 00 09 C4 09 C5 75 08 95 02 81 02 C0
Setup HID
HID Connection established
CTRL-C - SIGINT received, shutting down..
```

Packet log: [hci_dump.zip](https://github.com/bluekitchen/btstack/files/4187071/hci_dump.zip)

mringwal, 2020-02-11:
There's nothing special in the log. It just doesn't send much.
What game controller do you use? Could you post the log from Windows? Is there a special command that is sent by Windows that makes it start sending reports?

Jascha-N, 2020-02-11:
> What game controller do you use?

It's a cheap Chinese BT3.0 controller, VR Shinecon SC-B03.

> Could you post the log from Windows?

[windows-usbpcap.zip](https://github.com/bluekitchen/btstack/files/4187973/windows-usbpcap.zip)

> Is there a special command that is sent by Windows that makes it start sending reports?

I don't see anything out of the ordinary.

Jascha-N, 2020-02-12:
One difference I can tell is that Windows seems to switch to sniff mode before it starts listening for HID reports. My understanding is not sufficient to know if this could make a difference.

Jascha-N, 2020-02-13:
I got it to work! I call `gap_set_default_link_policy_settings(LM_LINK_POLICY_ENABLE_SNIFF_MODE)` at the start of the program. I don't even have to call `gap_sniff_mode_enter`. It seems to automatically switch to sniff mode after opening the channels. Do you know why this works?

```
HCI_EVENT_MODE_CHANGE, handle 0x0080, mode 2
```

mringwal, 2020-02-16:
Great work! Where did you see the 'set link policy' hci command in the windows pcap file? I only see two QoS setup commands.

Going through the HID 1.1 specification again, I found "Bluetooth HID Hosts should always accept slave-initiated requests for Sniff mode", so your call to `gap_set_default_link_policy_settings` allows for that, which makes your game controller happy. I'll update the hid_host_example.

closed 2020-02-16 by mringwal

Jascha-N, 2020-02-16:
```
Frame 7173: 35 bytes on wire (280 bits), 35 bytes captured (280 bits) on interface wireshark_extcap1636, id 0
USB URB
Bluetooth
Bluetooth HCI USB Transport
Bluetooth HCI Event - Mode Change
    Event Code: Mode Change (0x14)
    Parameter Total Length: 6
    Status: Success (0x00)
    Connection Handle: 0x0049
    Current Mode: Sniff Mode (0x02)
    Interval: 18 Baseband slots (11,250000 msec)
```

mringwal, 2020-02-16:
ah thanks. that's the sniff mode event. I was looking for an HCI Command... of course, the sniff mode change requires setting the link policy first.

Jascha-N, 2020-02-16:
I didn't see anything else unfortunately. But the sniff mode event got me looking in the right direction I guess.

commit 6e82c1f8 (2020-02-16): hid_host_demo: allow for sniff mode, fixes #268