bluekitchen/btstack #270: Raspberry pi 3b(old version) won't get past Phase 1

nathanRamaNoodles, 2020-02-04:
**Describe the bug**
Raspberry pi 3 b running any example only gets to phase 1 and then exits program.
**To Reproduce**
Steps to reproduce the behavior:
1. Run any example in raspi's port

**Expected behavior**
example code should run without exiting and should get past phase 1.

**Additional context**
Here is a log:
```
Packet Log: /tmp/hci_dump.pklg
H5, 921600, BT_REG_EN at GPIO 128
Power Cycle Controller
Phase 1: Download firmware
```

mringwal, 2020-02-04:
Please upload /tmp/hci_dump.pklg (as .zip).
Does it start at least once after a power cycle of the Raspi?

nathanRamaNoodles, 2020-02-05:
[btstack_log.zip](https://github.com/bluekitchen/btstack/files/4162042/btstack_log.zip)
It doesn't start even after rebooting the pi.

mringwal, 2020-02-06:
Thanks for the log. The Raspi 3 B has the Bluetooth Controller connected without Hardware Flowcontrol (RTS/CTS), so the H4 protocol is used to download the PatchRAM and then H5 is used. 

In your log, BTstack sends the first HCI Reset and gets not answer. This really should work and I did test this setup before (I have a Raspberry Pi 3, Model B, 1 GB RAM) here..

hm.. could you try to find another Pi 3B and check there to see if only your particular device works? Alternatively, does it work with the provided bluez stack?

nathanRamaNoodles, 2020-02-11:
I figured out the problem; turns out my btstack was out of date by 3 months 0.0.  Time flies when you're debugging.
However, the raspberry pi 3B seems slower for Bluetooth than the 3B+.  But that's good enough.  I might use libusb...

closed 2020-02-11 by nathanRamaNoodles

mringwal, 2020-02-11:
For some reason, every second Raspi version changes the way the Bluetooth/Wifi controller is connected to the main CPU. Since 3B+ a full hardware UART is used again. libusb is safe, too..