bluekitchen/btstack #272: Unfiltered BLE characteristic_value_updates (notify/indicate)

jimmo, 2020-03-03:
**Is your feature request related to a problem? Please describe.**

I'm working on the MicroPython binding for BTstack (currently BLE-only). cc @dpgeorge

We have an existing Python API which is currently implemented on top of NimBLE -- http://docs.micropython.org/en/latest/library/ubluetooth.html

The way the API for GATT clients currently behaves for notifications is that the client (in a Python callback) gets an `_IRQ_GATTC_NOTIFY` or `_IRQ_GATTC_INDICATE` event for any notifications received. They don't need to explicitly listen for them (although they might have configured the CCCB of course).

In BTstack, there doesn't appear to be a way to achieve this directly. The two closest options are:
 - Call `gatt_client_listen_for_characteristic_value_updates()` for each connection handle and value handle pair.
 - Handle the `ATT_HANDLE_VALUE_NOTIFICATION` directly and duplicate some of the code from gattc_client.c to parse the packet. (Not really a sensible option)

The first option requires that we know the handles, which requires the user to have done a discovery. Additionally, we'd need to manage the `gatt_client_notification_t` nodes in order to clean up on disconnection via `gatt_client_stop_listening_for_characteristic_value_updates`. Our bindings don't currently need to track any state for connections. And this would be a behavioural change to our API (although realistically the user should always be doing a discovery).

**Describe the solution you'd like**
One simple option would be a way to do a single call to `gatt_client_listen_for_characteristic_value_updates()` with some sort of sentinel "ALL" value for the two handles that would match any incoming notification/indication.

Other options seem more complicated or more code size, e.g. adding more event types, etc. But open to suggestions!

**Describe alternatives you've considered**
We could add an additional method to our API to allow the user to explicitly listen for notifications for a connection handle and value handle pair. We'd still need the extra state management to stop listening on disconnection. The other disadvantage (to us) would be that we'd have to then implement this for the NimBLE bindings.

Thanks!

jimmo, 2020-03-03:
Quick followup -- we'd also prefer not to have to do any allocations outside of the main initialisation sequence. e.g. on discovery of a characteristic/descriptor, it would have to allocate a `gatt_client_notification_t`. Aside from the obvious reason for not wanting to do this, our heap is locked during the BTstack callbacks. We can mitigate this somewhat by deferring until when we dispatch the event to Python, but we'd still prefer not to allocate at all.

mringwal, 2020-03-03:
Hi Jim. Thanks for the detailed feature request. The motivation for the explicit listener registration was to allow to implement GATT Clients for particular GATT Services in a modular way.

I fully understand your problem when trying to provide MicroPython bindings: I did run into a similar situation implementing the btpclient for auto-pts. There, I did a quick hack to allow to register a listener for 'any value handle' (https://github.com/bluekitchen/btstack/commit/fea53d87370b9d8f24af2eb1669c0a155e0035eb).  I was already wondering if this should be made public...

Ok, we'll add something to allow for wildcards for value and connection handles in the next days.

dpgeorge, 2020-03-03:
> The motivation for the explicit listener registration was to allow to implement GATT Clients for particular GATT Services in a modular way.

Yes this makes sense, that you want to have separate callbacks to have independent code handling independent services.

But in MicroPython we are deferring this handling to the Python code, so at the C level we need to know about all notifications/indications so they can be passed through to the Python code.

> I did run into a similar situation implementing the btpclient for auto-pts. There, I did a quick hack to allow to register a listener for 'any value handle' ([fea53d8](https://github.com/bluekitchen/btstack/commit/fea53d87370b9d8f24af2eb1669c0a155e0035eb)). I was already wondering if this should be made public...

Of course, just because MicroPython's user-facing API for BLE needs this feature doesn't mean it's the right thing to add to BTstack!  But it's nice to know you also hit this, and if BTstack can support wildcards that would make it easier/more efficient for us.

mringwal, 2020-03-03:
> Of course, just because MicroPython's user-facing API for BLE needs this feature doesn't mean it's the right thing to add to BTstack! But it's nice to know you also hit this, and if BTstack can support wildcards that would make it easier/more efficient for us.

No worry here, the wildcard here is easy to implement and has no obvious drawbacks, so that 's fine.

mringwal, 2020-03-03:
you can now use NULL for any characteristic and GATT_CLIENT_ANY_CONNECTION as con handle 

see: https://github.com/bluekitchen/btstack/commit/b3f03c843f6a029757848cf4db52890477de181b

closed 2020-03-03 by mringwal

dpgeorge, 2020-03-04:
@mringwal just wondering what timeline we can expect for this enhancement to go into master?  Or is it sensible for us to work off the btstack develop branch for now?

mringwal, 2020-03-04:
We try to merge develop onto master about monthly-ish. However, it would make sense for MP to use develop branch for now - there might be other similar changes. develop should be fully functional, if anything stopped working, please let us now immediately.

dpgeorge, 2020-03-04:
> However, it would make sense for MP to use develop branch for now - there might be other similar changes

Yes, good point.  So we'll work off develop.