bluekitchen/btstack #273: how to set BTstack toolchain and libraries with Eclipse

Darram007, 2020-03-04:
Hello,
Is it possible to implement BTstack functionality with Eclipse(Eclipse IDE for C/C++ Developers
Version: 2019-12 (4.14.0)) to access ESP32 bluetooth modul functionality?
I am wondering that it can be intigrated with ESP-IDF frimware and toolchain which already set to the eclipse environment? if it is possible way to add then please add you comments.
Thank you

mringwal, 2020-03-04:
If you follow the readme.md, BTstack is copied into the esp-idf/components folder. The script also creates individual projects, with a CMakeFiles.txt and a Makefile. Maybe you can setup an Eclipse project based on these?

Hm. How do you compile the other esp-idf examples in Eclipse for C/C++? if you can do that, everything should be ready after btstack is copied into components.

Darram007, 2020-03-04:
Thank you for the fast reply. I followed steps as mentioned and also able to build and flash through Eclipse. ( ESP-IDF stable version and ESP32-WROOM-32D)

Thank you.

closed 2020-03-04 by Darram007