bluekitchen/btstack #275: LE_Set_Scan_Parameters sometimes happen after LE_Set_Scan_Enable

dpgeorge, 2020-03-10:
**Describe the bug**
To start scanning our code uses:
```c
    gap_set_scan_parameters(0, 10000 / 625, 10000 / 625);
    gap_start_scan();
```
But sometimes the scan parameters are not set (the controller returns an error setting them) because BTstack sends out the LE_Set_Scan_Parameters command *after* LE_Set_Scan_Enable.  From inspecting `src/hci.c` it seems that this occurs because `hci_run()` checks and manages changes to LE scanning before checking for changes to LE scan parameters.  This combined with the fact that `hci_run()` called by `gap_set_scan_parameters()` may exit before scan parameters are set (eg because other events come in from the controller) means that scanning may be enabled before parameters are set.

(Note that our code is enabling BTstack right before starting scanning, with no pause, which may be part of the issue.)

**To Reproduce**
We are using MicroPython + BTstack + CYW4343.  Running the following code:
```python
import bluetooth
ble = bluetooth.BLE()
ble.active(1)  # enable BTstack
ble.gap_scan(2000, 10000, 10000)  # set parameters then start scan for 2000ms
```

**Expected behavior**
LE scan parameters to be set before LE scanning starts.

**HCI Packet Logs**
See attached.

**Environment:**
 - Current BTstack branch: develop at a6121b51b229b883431c4a1600b0f46db31d32dc
 - Bluetooth Controller CYW4343
 - Remote device: N/A

[btstack_log2.zip](https://github.com/bluekitchen/btstack/files/4310195/btstack_log2.zip)

mringwal, 2020-03-11:
Thanks for the bug report and the initial analysis. I guess it needs three blocks (disable if needed, set params if needed, enable if needed), or at least a check that skips the scan enable if there are parameters to change. I'll have a go at it in the next days.

dpgeorge, 2020-03-11:
> I guess it needs three blocks (disable if needed, set params if needed, enable if needed), or at least a check that skips the scan enable if there are parameters to change.

Yes, that's what I would have guessed.  Probably the latter option has a smaller code footprint.  Another option is to put the parameter-change code first, but skip it if scanning is enabled.

mringwal, 2020-03-11:
> Another option is to put the parameter-change code first, but skip it if scanning is enabled.

This is a tiny bit cleaner than the other one. Thanks!

dpgeorge, 2020-03-11:
Actually, what about the following situation, which aims to change the scan parameters after scanning for 10s:
```c
gap_set_scan_parameters(0, 10000 / 625, 10000 / 625);
gap_start_scan();
sleep(10);
gap_stop_scan();
gap_set_scan_parameters(1, 10000 / 625, 10000 / 625);
gap_start_scan();
sleep(10);
 ```
Maybe these GAP scan functions should block until they complete, or provide a function for the user to check if they are complete?  Eg:
```c
void gap_set_scan_parameters(uint8_t scan_type, uint16_t scan_interval, uint16_t scan_window){
    hci_stack->le_scan_type     = scan_type;
    hci_stack->le_scan_interval = scan_interval;
    hci_stack->le_scan_window   = scan_window;
    while (hci_stack->le_scan_type != 0xff){
        hci_run();
    }
}
```

mringwal, 2020-03-12:
BTstack is asynchronous and if you call sleep(..) nothing will happen. To do a 10 second scanning, you'd start scanning and use a timer to stop it 10 seconds later.

Polling hci_run() won't work, e.g. because the UART is handled from the main run loop. So, you'd need to call bstack_run_loop_execute_once() or such. I've seen a Bluetooth stack with that and my feeling is that it will just explode...

I guess the question would be if stop / set params / start works, although in that case you're supposed to only call set params and have BTstack handle the stop / set / start action. If that doesn't work, that needs to be fixed. For advertisements, there are separate flags for stop / set param / set adv data / set scan response / start, so I think all cases are covered.

dpgeorge, 2020-03-13:
Ok, so to stay focused on the original issue I guess my main question is: can one expect the following code to work:
```c
    gap_set_scan_parameters(0, 10000 / 625, 10000 / 625);
    gap_start_scan();
```
ie to set the parameters then start the scan.  Or, since everything is asynchronous, is the code supposed to wait (what that means is a separate question) after calling `gap_set_scan_parameters()` for it to take effect and before calling `gap_start_scan()`?

mringwal, 2020-03-13:
Hi. The (undocumented) idea is that you should be able to call one or more gap functions to configure your desired state and the stack should fulfill that.

With this, your snippet to set params and start scanning should work, if not, it's a bug that needs fixing.

dpgeorge, 2020-03-13:
> With this, your snippet to set params and start scanning should work, if not, it's a bug that needs fixing.

Ok, so at least that needs to be fixed.

mringwal, 2020-03-13:
Please let me know if you find another corner case.

closed 2020-03-13 by mringwal