bluekitchen/btstack #279: Correct Configuration for A2DP sink example

welle108, 2020-04-13:
After cloning BTStack library into ESP-IDF directory I am unable to flash examples to Adafruit HUZZAH32 ESP32 module. 

**Command used**
'''
make flash monitor
'''

**Expected behavior**
A2DP example will flash to ESP32

**Output**

Initially I got this warning
'''
esp-idf/components/bt/controller/bt.c:1486: undefined reference to `bredr_sco_datapath_set
'''
similar to [this](/Users/piotrk/_git/esp-idf/components/bt/controller/bt.c:1486: undefined reference to `bredr_sco_datapath_set') issue, so I ran menuconfig and put the controller in Dual Mode, which got rid of the bredr warning, but now I'm getting a ton of errors involving `osi_mem_dbg' which look like this:

'''
/home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:68: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:78: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:88: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:93: undefined reference to `osi_mem_dbg_record'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:96: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:101: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:121: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:140: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:142: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/btstack/port/esp32/example/a2dp_sink_demo/build/bt/libbt.a(avct_lcb_act.o):/home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:159: more undefined references to `osi_mem_dbg_clean' follow
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/btstack/port/esp32/example/a2dp_sink_demo/build/bt/libbt.a(avct_lcb_act.o): in function `avct_lcb_send_msg':
/home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:535: undefined reference to `osi_mem_dbg_record'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:538: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/btstack/port/esp32/example/a2dp_sink_demo/build/bt/libbt.a(avct_lcb_act.o): in function `avct_lcb_free_msg_ind':
/home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:612: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/btstack/port/esp32/example/a2dp_sink_demo/build/bt/libbt.a(avct_lcb_act.o): in function `avct_lcb_msg_ind':
/home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:653: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:667: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_lcb_act.c:673: undefined reference to `osi_mem_dbg_record'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/btstack/port/esp32/example/a2dp_sink_demo/build/bt/libbt.a(avct_l2c.o): in function `avct_l2c_data_ind_cback':
/home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avct/avct_l2c.c:402: undefined reference to `osi_mem_dbg_clean'
/home/link/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/link/esp/esp-idf/btstack/port/esp32/example/a2dp_sink_demo/build/bt/libbt.a(avrc_opt.o): in function `avrc_vendor_msg':
/home/link/esp/esp-idf/components/bt/host/bluedroid/stack/avrc/avrc_opt.c:58: undefined reference to `osi_mem_dbg_record'
collect2: error: ld returned 1 exit status
make: *** [/home/link/esp/esp-idf/make/project.mk:564: /home/link/esp/esp-idf/btstack/port/esp32/example/a2dp_sink_demo/build/a2dp_sink_demo.elf] Error 1
'''

I've tried toggling multiple settings in the component config menu of menuconfig, but nothing seems to affect it. I've verified that I can flash normal ESP IDF examples to my ESP fine, so the issue is definitely how I integrated it. Any help is appreciated

welle108, 2020-04-14:
In case anyone else has this happen, don't forget to set default serial port in port/esp32/template. This fixed all of my problems and allowed me to flash the example to my board. Currently having a hard time pairing the device to my android, but that is another issue for another day

closed 2020-04-14 by welle108

mringwal, 2020-04-14:
Hi. Did you get the A2DP Sink demo to work on your ESP32?
The compile errors related tp `osi_mem_dbg` seem to originate from the original Bluedroid stack - which is not needed if you use BTstack. Unfortunately, it cannot be skipped from the built.

welle108, 2020-04-14:
Hi. I was able to ultimately get the examples flashed and running more or less with some issues. Last night the ESP32 was fairly responsive, and I could see every event on the monitor (Volume change, Stream paused, etc...) but I was getting 

``` error storing samples in SBC ring buffer!!! ```

errors. When trying to recreate the behavior today, the ESP32 is able to connect, but now I just get a stream started message and the monitor freezes. Not telling you this because I necessarily expect you to fix it for me; more just so you're aware the behavior exists. I'm working on this, and the PBAP example (Getting the same OBEX errors as [this](https://github.com/bluekitchen/btstack/issues/265) guy) with my professor for a school project, so I'll let you know if we make any headway on either

mringwal, 2020-04-15:
Hi Scott. If you get an error storing samples, you're receiving audio faster than it's played back. See our blog post: http://bluekitchen-gmbh.com/a2dp-sink-and-source-on-stm32-f4-discovery-board/

Anyway, you can increase the max compensation for resampling by increasing the compensation value here (I'd suggest to double it): https://github.com/bluekitchen/btstack/blob/master/example/a2dp_sink_demo.c#L490