bluekitchen/btstack #281: support RTL8761B

zhaojh329, 2020-04-19:
support RTL8761B

mringwal, 2020-04-19:
Hi. Looks like this is a USB Bluetooth Dongle. 

A quick search indicates a firmware has to be loaded before it can be used.
https://www.spinics.net/lists/linux-bluetooth/msg84456.html

Do you have  datasheets for it?, which also explains the firmware download? 
Any special need for this particular Bluetooth Controller?

I suggest to buy a normal CSR8510, or BCM20702 USB dongle.

zhaojh329, 2020-04-19:
RTL8761B support Bluetooth 5.0.

mringwal, 2020-04-19:
Which Bluetooth 5.0 features in detail? Every Bluetooth 4.0 Controller can be listed as 5.0 - with  identical features.
- LE Date Length Extension?
- 2 Mb-Phy?
- Long range?

closed 2020-04-20 by zhaojh329