bluekitchen/btstack #284: Adding new command packets

V11che, 2020-05-22:
Hi,

So I'm really trying to figure out how to modify the stack to support a particular chipset. I have added my own command in hci_cmd.c and when it wants to run, it creates the correct packet and all is happy. 

However, it often doesnt send the packet and just annoyingly continues on it's merry way...

Due to this check:

`int hci_send_cmd_va_arg(const hci_cmd_t *cmd, va_list argptr){
    if (!hci_can_send_command_packet_now()){ 
        log_error("hci_send_cmd called but cannot send packet now");
        return 0;
    }
...
}`
 Specifically the "hci_stack->num_cmd_packets" is 0 and I don't see a way of accessing this from outside of hci.c

Any help would be great.

Cheers

V

mringwal, 2020-05-22:
HI. While you cannot access hci_stack->nu_cmd_packets, you can call hci_can_send_command_packet_now in your code.

See http://bluekitchen-gmbh.com/btstack/protocols/#hci-host-controller-interface

If you consider a pull request for this chipset (which one? :), please use the vendor name as part of the command name, or, only define them it in the code that uses them.

closed 2020-05-22 by mringwal

V11che, 2020-05-22:
Hi, its the Realtek RTL8761BTV

I'll add it if I get it working. I've already seen the page you linked, but "hci_can_send_command_packet_now" just returns 0 because i think it expects I've got no more command packets to send and I can't say I have more.

```
int hci_can_send_command_packet_now(void){
    if (hci_can_send_comand_packet_transport() == 0) return 0;
    return hci_stack->num_cmd_packets > 0; <----- This is where I get 0 so the check is failing.
}
```

for example this in my code just steps over the "if" so doesnt call the hci_send_cmd : 

```

static void rtlbt_change_baudrate(uint32_t baudrate)
{
		if (hci_can_send_command_packet_now())
		{
			hci_send_cmd(&hci_rtl8761_vendor_change_baudrate, &baudrate, sizeof(baudrate));
		}
}
 ```
command in hci_cmd.c looks like : 
```
const hci_cmd_t hci_rtl8761_vendor_change_baudrate = {
    OPCODE(OGF_VENDOR, HCI_VENDOR_CHANGE_BAUD), "4"
};
```
What am I missing? hci_stack->num_cmd_packets never changes no matter how long I wait, so I'm at a bit of a loss without a bodge.

cheers

mringwal, 2020-05-22:
hi. after an HCI Command is sent, you cannot send another one until the Bluetooth Controller has sent a HCI Command Status or Complete Event.  

The easiest way to deal with this is to  put the code to send an HCI command into a hci packet handler, and check if you can send every time it gets called, like here: https://github.com/bluekitchen/btstack/blob/master/example/dut_mode_classic.c#L90

To change the baudrate, however, it would be better to implement the btstack_chipset.h interface, see cc256x or bcm versions. To change it, first the command has to be sent, then a HCI Command Complete Event has to be received, and then the UART baud rate is changed. If you implement the chipset interface, the stack will (should) take care of that.

V11che, 2020-05-23:
Thanks, I'll take a look at that. I just used the baudrate change as one of 3 different things I need to add as an example :)

Thanks for you're help.