bluekitchen/btstack #285: How to listen for HID output reports as a device from the host?

nathanRamaNoodles, 2020-05-23:
**Is your feature request related to a problem? Please describe.**
How can I listen for `a2 reports`: HID_MESSAGE_TYPE_DATC_DEPRECATED reports in our own custom packet_handler when they are getting passed into hid_device's own packet_handler.

**Describe the solution you'd like**
Give a way to listen on hid output reports

Basically, I'm trying to emulate a ps4 controller, but when the host sends an `a2` HID output report, I am not able to process it even though I see it coming to me in Wireshark.  Maybe there's already a way to do this and I can't see it?

Using the existing hid_gamepad_example from [here](https://github.com/bluekitchen/btstack/pull/209); I am trying to see the HID output report in `L2CAP_DATA_PACKET`, but I get nothing.

mringwal, 2020-05-25:
Hi Nathan. Could you upload the .pklg file here please?

nathanRamaNoodles, 2020-05-25:
I figured it out. I never registered the report_callback
```
hid_device_register_report_data_callback
```

closed 2020-05-25 by nathanRamaNoodles