bluekitchen/btstack #286: Cannot connect to bluetooth gamepad using hid_host_demo.c example

esxbr, 2020-05-23:
**Describe the bug**
The connection drops after L2CAP_EVENT_CHANNEL_OPENED

Bluetooth Gamepad:
Device: IPEGA PG-9021
Type: Classic BR/EDR
Profile Supported: HID

[image]

**HCI Packet Logs**
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:5624
load:0x40078000,len:13168
ho 0 tail 12 room 4
load:0x40080400,len:3724
0x40080400: _init at ??:?

entry 0x40080644
I (203) cpu_start: Pro cpu up.
I (203) cpu_start: Application information:
I (203) cpu_start: Project name:     hid_host_demo
I (206) cpu_start: App version:      1
I (210) cpu_start: Compile time:     May 22 2020 23:08:56
I (216) cpu_start: ELF file SHA256:  7899b74e8dc91b8e...
I (222) cpu_start: ESP-IDF:          v4.2-dev-1575-ga3520970f
I (228) cpu_start: Single core mode
I (233) heap_init: Initializing. RAM available for dynamic allocation:
I (240) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (246) heap_init: At 3FFB7468 len 00000B98 (2 KiB): DRAM
I (252) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (258) heap_init: At 3FFC9E08 len 000161F8 (88 KiB): DRAM
I (264) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (271) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (277) heap_init: At 40093258 len 0000CDA8 (51 KiB): IRAM
I (283) cpu_start: Pro cpu start user code
I (301) spi_flash: detected chip: generic
I (301) spi_flash: flash io: dio
I (301) esp_core_dump_uart: Init core dump to UART
I (304) cpu_start: Starting scheduler on PRO CPU.
[00:00:00.310] LOG -- btstack_run_loop_freertos.c.309: run loop init, task 0x3ffb7b84, queue item size 8
[00:00:00.318] LOG -- main.c.272: transport_register_packet_handler
[00:00:00.328] LOG -- btstack_tlv.c.48: TLV Instance 0x3f40686c
[00:00:00.330] LOG -- btstack_tlv_esp32.c.54: tag 42544400 -> 42544400
[00:00:00.336] LOG -- btstack_tlv_esp32.c.68: read tag 42544400
[00:00:00.342] LOG -- btstack_tlv_esp32.c.54: tag 42544401 -> 42544401
[00:00:00.348] LOG -- btstack_tlv_esp32.c.68: read tag 42544401
[00:00:00.354] LOG -- btstack_tlv_esp32.c.54: tag 42544402 -> 42544402
[00:00:00.360] LOG -- btstack_tlv_esp32.c.68: read tag 42544402
[00:00:00.366] LOG -- btstack_tlv_esp32.c.54: tag 42544403 -> 42544403
[00:00:00.372] LOG -- btstack_tlv_esp32.c.68: read tag 42544403
[00:00:00.378] LOG -- btstack_tlv_esp32.c.54: tag 42544404 -> 42544404
[00:00:00.384] LOG -- btstack_tlv_esp32.c.68: read tag 42544404
[00:00:00.389] LOG -- btstack_tlv_esp32.c.54: tag 42544405 -> 42544405
[00:00:00.396] LOG -- btstack_tlv_esp32.c.68: read tag 42544405
[00:00:00.401] LOG -- btstack_tlv_esp32.c.54: tag 42544406 -> 42544406
[00:00:00.408] LOG -- btstack_tlv_esp32.c.68: read tag 42544406
[00:00:00.413] LOG -- btstack_tlv_esp32.c.54: tag 42544407 -> 42544407
[00:00:00.419] LOG -- btstack_tlv_esp32.c.68: read tag 42544407
[00:00:00.425] LOG -- btstack_tlv_esp32.c.54: tag 42544408 -> 42544408
[00:00:00.431] LOG -- btstack_tlv_esp32.c.68: read tag 42544408
[00:00:00.437] LOG -- btstack_tlv_esp32.c.54: tag 42544409 -> 42544409
[00:00:00.443] LOG -- btstack_tlv_esp32.c.68: read tag 42544409
[00:00:00.449] LOG -- btstack_tlv_esp32.c.54: tag 4254440a -> 4254440A
[00:00:00.455] LOG -- btstack_tlv_esp32.c.68: read tag 4254440A
[00:00:00.461] LOG -- btstack_tlv_esp32.c.54: tag 4254440b -> 4254440B
[... 170 lines cut ...]
[00:00:08.179] EVT <= 6E 00 
[00:00:08.182] CMD => 35 0C 05 01 80 00 03 00 
[00:00:08.186] EVT <= 13 05 01 80 00 01 00 
[00:00:08.190] EVT <= 6E 00 
[00:00:08.210] ACL <= 80 20 1A 00 16 00 41 00 07 00 01 00 11 00 0E 36 00 0B 36 00 08 09 00 01 35 03 19 11 24 00 
[00:00:08.212] LOG -- l2cap.c.2035: L2CAP_DISCONNECT local_cid 0x41 reason 0x0
[00:00:08.216] ACL => 80 00 0C 00 08 00 01 00 06 03 04 00 40 00 41 00 
[00:00:08.222] EVT <= 6E 00 
[00:00:08.225] CMD => 35 0C 05 01 80 00 01 00 
[00:00:08.229] EVT <= 13 05 01 80 00 01 00 
[00:00:08.233] EVT <= 6E 00 
[00:00:08.252] ACL <= 80 20 0C 00 08 00 01 00 07 03 04 00 40 00 41 00 
[00:00:08.253] LOG -- l2cap.c.2720: L2CAP signaling handler code 7, state 13
[00:00:08.254] LOG -- l2cap.c.1061: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x41
[00:00:08.261] EVT <= 71 02 41 00 
[00:00:08.264] LOG -- sdp_client.c.473: SDP Client disconnected.
HID Control PSM missing
[00:00:08.272] LOG -- l2cap.c.1972: free channel 0x3ffd1888, local_cid 0x0041
[00:00:08.279] LOG -- l2cap.c.1217: l2cap_stop_rtx for local cid 0x41
[00:00:08.285] CMD => 35 0C 05 01 80 00 01 00 
[00:00:08.289] EVT <= 6E 00

**Environment:  (please complete the following information):**
 - Current BTstack branch: master
 - Bluetooth Controller ESP32 DOIT DEVKIT 1.0

mringwal, 2020-05-25:
Hi there.

During the SDP query, the example ask the game controller for info on its HID Device service. The game controller responds, but it only reports that it support HID Device. It does not list the HID Control or HID Interrupt PSMs. These are mainly constant, so we could use the defaults (0x17 and 0x19). 

However, there's also no HID Descriptor in the SDP response. I'm a bit irritated here. Without the HID Descriptor, the format of the received HID Reports is unknown.

If you like, you can comment the checks for HID Interrupt / Control PSM and let the example open the L2CP Channels, which should allow you to receive HID Reports.

mringwal, 2020-05-25:
Well, you need to set the values, so replace

        case SDP_EVENT_QUERY_COMPLETE:
            if (!hid_control_psm) {
                printf("HID Control PSM missing\n");
                break;
            }
            if (!hid_interrupt_psm) {
                printf("HID Interrupt PSM missing\n");
                break;
            }

with

        case SDP_EVENT_QUERY_COMPLETE:
            if (!hid_control_psm) {
                printf("HID Control PSM missing\n");
                hid_control_psm = BLUETOOTH_PSM_HID_CONTROL;
            }
            if (!hid_interrupt_psm) {
                printf("HID Interrupt PSM missing\n");
                hid_interrupt_psm  = BLUETOOTH_PSM_HID_INTERRUPT;
            }

esxbr, 2020-05-25:
I implemented the whole thing (pairing, reconnection...)... it's missing only the keys code. I'll finish it tomorrow.
I'm not sure if it's 100% implemented correctly .. but it's working fine here ... 

/*
 * Copyright (C) 2017 BlueKitchen GmbH
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 *
 * 1. Redistributions of source code must retain the above copyright
 *    notice, this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *    notice, this list of conditions and the following disclaimer in the
 *    documentation and/or other materials provided with the distribution.
 * 3. Neither the name of the copyright holders nor the names of
 *    contributors may be used to endorse or promote products derived
 *    from this software without specific prior written permission.
 * 4. Any redistribution, use, or modification is done solely for
 *    personal benefit and not for any commercial purpose or for
 *    monetary gain.
 *
 * THIS SOFTWARE IS PROVIDED BY BLUEKITCHEN GMBH AND CONTRIBUTORS
 * ``AS IS'' AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 * LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
 * FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL MATTHIAS
 * RINGWALD OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
 * INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING,
 * BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS
 * OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED
 * AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
 * OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF
 * THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF
 * SUCH DAMAGE.
 *
 * Please inquire about commercial licensing options at 
 * contact@bluekitchen-gmbh.com
 *
 */

#define BTSTACK_FILE__ "hid_host_demo.c"

/*
 * hid_host_demo.c
 */

/* EXAMPLE_START(hid_host_demo): HID Host Demo
 *
 * @text This example implements an HID Host. For now, it connnects to a fixed device, queries the HID SDP
 * record and opens the HID Control + Interrupt channels
 */

#include <inttypes.h>
#include <stdio.h>

#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "btstack.h"
#include "btstack_config.h"

#include "btstack_port_esp32.h"
#include "btstack_run_loop.h"
#include "hci_dump.h"

#define L2CAP_CHANNEL_MTU 128         // PS4 requires a 79-byte packet
#define MAX_ATTRIBUTE_VALUE_SIZE 300

#define ENABLE_LOG_DEBUG
[... 480 lines cut ...]
    //hci_set_master_slave_policy(HCI_ROLE_MASTER);
    hci_power_control(HCI_POWER_ON);
    return 0;
}

/* EXAMPLE_END */

int app_main(void){

    // optional: enable packet logger
    //hci_dump_open(NULL, HCI_DUMP_STDOUT);

    // Configure BTstack for ESP32 VHCI Controller
    btstack_init();

    // Setup example
    btstack_main(0, NULL);

    // Enter run loop (forever)
    btstack_run_loop_execute();

    return 0;
}

/* EXAMPLE_END */

mringwal, 2020-05-26:
Hi. I've added the fallback to defaults on the develop branch. Good point about re-connect / incoming connections.

mringwal, 2020-06-22:
The hid_host_demo was updated to also handle reconnects. There's at least one bug in it bu I'll close this ticket as you've got your setup to work.

closed 2020-06-22 by mringwal