bluekitchen/btstack #291: gap_delete_all_link_keys() does nothing

maxdos64, 2020-06-08:
**Description**
gap_delete_all_link_keys(void) is supposed to delete the existing link database for a device.
Calling it does not delete any link keys though.

**Reproduce** 
The following code was used to test the functionality
```
int main(int argc, const char * argv[]){

	/// GET STARTED with BTstack ///
	btstack_memory_init();
	btstack_run_loop_init(btstack_run_loop_posix_get_instance());

	// use logger: format HCI_DUMP_PACKETLOGGER, HCI_DUMP_BLUEZ or HCI_DUMP_STDOUT
	const char * pklg_path = "/tmp/hci_dump.pklg";
	hci_dump_open(pklg_path, HCI_DUMP_PACKETLOGGER);
	printf("Packet Log: %s\n", pklg_path);

	config.device_name = "/dev/ttyUSB0"; // CSR8811 breakout board

	// accept path from command line
	if (argc >= 3 && strcmp(argv[1], "-u") == 0){
		config.device_name = argv[2];
		argc -= 2;
		memmove(&argv[1], &argv[3], (argc-1) * sizeof(char *));
	}
	printf("H4 device: %s\n", config.device_name);

	// init HCI
	const btstack_uart_block_t * uart_driver = btstack_uart_block_posix_instance();
	const hci_transport_t * transport = hci_transport_h4_instance(uart_driver);
	hci_init(transport, (void*) &config);

	gap_delete_all_link_keys();

	// set BD_ADDR for CSR without Flash/unique address
	// bd_addr_t own_address = { 0x11, 0x22, 0x33, 0x44, 0x55, 0x66};
	// btstack_chipset_csr_set_bd_addr(own_address);

	// inform about BTstack state
	hci_event_callback_registration.callback = &packet_handler;
	hci_add_event_handler(&hci_event_callback_registration);

	// handle CTRL-c
	signal(SIGINT, sigint_handler);

	// setup app
	btstack_main(argc, argv);

	// go
	btstack_run_loop_execute();

	return 0;
}
```

**Environment**
 - Current BTstack branch: master
 - Bluetooth Controller: CYW92081

**Log**

[gap_delete_not_working.pklg.txt](https://github.com/bluekitchen/btstack/files/4744203/gap_delete_not_working.pklg.txt)

mringwal, 2020-06-08:
The link keys are stored per Bluetooth Controller. In main, the stack doesn't know which Bluetooth Controller is used and hence, cannot delete the link keys.
Please move the call to ``gap_delete_all_link_keys`` into the packet handler where 'btstack state == working' is handled and try again.

maxdos64, 2020-06-09:
Thank you for the quick response !
This way it works :)
(maybe worth a note in the manual)

Thank you!

closed 2020-06-09 by maxdos64

mringwal, 2020-06-09:
https://github.com/bluekitchen/btstack/commit/c771037830eb42bd089a431fd6ebe192c4dcae06