bluekitchen/btstack #292: Message Access Profile例子 Iphone无法正常工作

yywOwen, 2020-06-15:
查看Message Access Profile Specification 57页，如下图
[image]

mringwal, 2020-06-15:
Hi. Sorry we mainly speak english (and german, and serbian..). MAP isn't implemented yet, but I see you've found our work-in-progress code on the 'map' branch. Our development on MAP is currently stalled, what you do need it for? Feel free to send Pull Requests for the MAP branch.

yywOwen, 2020-06-15:
China is at the forefront of the Bluetooth market, and hopefully it will support Chinese in the future, with more and more Chinese engineers involved

mringwal, 2020-06-15:
HI. We don't exclude any language on purpose. We just stopped working on MAP due to other priorities.

Could you explain the issue in english please? You've highlighted Message Listing Format v1.1. Is it related to support for the Chinese market?

vvhh2002, 2020-06-20:
This young guy just wants to implement MAP. Please forgive his rudeness. I think this issue could be closed.

mringwal, 2020-06-22:
@vvhh2002 Thanks. No problem from my side. I still don't understand what @yywOwen wanted to report/request in this issue.

mringwal, 2020-07-15:
Closing this as I don't understand the problem being reported.

closed 2020-07-15 by mringwal